liquidctl/liquidctl #151: Tracking issue for release 1.4.0

jonasmalacofilho, 2020-07-13:
During the next two weeks I'll be finalizing a new liquidctl release, version 1.4.0.  If you can, please start to test the code in the main branch; you can use this issue to report any problems.

The most notable user-facing changes in this version will be the addition of drivers for fourth-generation Kraken X coolers (with limited support for Kraken Z coolers as well), Corsair Platinum and PRO XT coolers, and some Gigabyte motherboards with RGB Fusion 2.0 controllers.

I'll be updating the [changelog](https://github.com/jonasmalacofilho/liquidctl/blob/master/CHANGELOG.md) during the next few days with all other significant changes, and there are some improvements I still hope to finish in time for the release.

Linux package maintainers may also want to follow any updates I make to the [`liquidctl-git`<sup>AUR</sup>](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=liquidctl-git) PKGBUILD for ArchLinux.

Thank you Tom, Casey, Sean, and all other contributors.  Thank you for contributing drivers or fixes, capturing or helping analyze USB traffic, reporting bugs, or simply getting in touch.

---

If you're working on a new feature or bug fix, please feel free to open pull requests normally during this two-week window.  I'll evaluate on a case-by-case basis whether a particular change can safely get into this release...

And from now on I intend to do _periodic_ releases every three months.  The changes accepted into the main branch at the time will determinate whether it's a patch or minor release.

haringsrob, 2020-07-23:
```
✘ rob@  ~  liquidctl list
Device ID 0: Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
 rob@  ~  liquidctl initialize all
ERROR: Unexpected error with Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
Traceback (most recent call last):
AssertionError
```

I receive this error on arch linux, not running as root. I have the Corsair Hydro Series H115i Pro RGB. Not sure if there is support for it.

jonasmalacofilho, 2020-07-23:
@haringsrob 

Can you post the output of `liquidctl initialize all --debug`?

haringsrob, 2020-07-23:
Sure thing
```
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (EndeavourOS; liquidctl-git 1.3.3.r279.f44281d-1)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CommonSmartDeviceDriver, CoolitPlatinumDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, RGBFusion2Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device feed:1307
[DEBUG] liquidctl.driver.usb: probing drivers for device feed:1307
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:407f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:4937
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:085c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:4937
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 0a12:0001
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c13
[DEBUG] liquidctl.driver.usb: probing drivers for device feed:1307
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.driver.usb: sending feature report 0xcc with 63 bytes: 60:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: got feature report 0x00 with 63 bytes: 01:00:01:01:00:06:00:00:00:00:00:49:54:38:32:39:37:42:58:2d:47:42:58:35:37:30:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02:00:01:00:02:00:01:00:00:01:02:00:00:01:97:82:00:00:00:00
[ERROR] liquidctl.cli: Unexpected error with Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/liquidctl/cli.py", line 301, in main
    _print_dev_status(dev, dev.initialize(**opts))
  File "/usr/lib/python3.8/site-packages/liquidctl/driver/rgb_fusion2.py", line 185, in initialize
    assert data[0] == _REPORT_ID and data[1] == 0x01
AssertionError
```

jonasmalacofilho, 2020-07-23:
@haringsrob,

Does Endeavour use vanilla Arch Linux packages (for core and community, not AUR), or does it have its own repositories?  Also, what `python-hidapi`, `hidapi` and `libusb` versions do you have?

Finally, can you please show me the output of the following command?  It should (indirectly) tell me which backend is being used by `python-hidapi`.

```
$ find /usr/lib/python3.8 -iname 'hid*.so' -print -exec ldd {} \;
```

---

@CaseySJ,

Can you please take a look a this, since you're more familiar than me with this driver?

At first glance it seems like an invalid response (report ID should not be 0 if the device supports reports) from an older firmware version, but maybe you've seen this problem before.

On another front, can you please test the latest code in the main branch (as whole, not just this specific driver) and make sure everything still works for you?

haringsrob, 2020-07-23:
Hey @jonasmalacofilho, afaik endeavour uses vanilla, but how can I check? I did install this using `yay liquidctl-git`.

As for the version checking, not quite sure how to do that.
 
Output of the command: 
```
rob@  /  find /usr/lib/python3.8 -iname 'hid*.so' -print -exec ldd {} \;
/usr/lib/python3.8/site-packages/hid.cpython-38-x86_64-linux-gnu.so
	linux-vdso.so.1 (0x00007ffe2b54a000)
	libhidapi-hidraw.so.0 => /usr/lib/libhidapi-hidraw.so.0 (0x00007f5965010000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f5964fee000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f5964e27000)
	libudev.so.1 => /usr/lib/libudev.so.1 (0x00007f5964e01000)
	/usr/lib64/ld-linux-x86-64.so.2 (0x00007f596505d000)
```

jonasmalacofilho, 2020-07-23:
@haringsrob,

> Hey @jonasmalacofilho, afaik endeavour uses vanilla, but how can I check?

I found a page in Endeavour's docs that appears to say that it's indeed vanilla/from Arch's repositories.  But I honestly don't know how to check this either.  I guess you could take a look at your `/etc/pacman.d/mirrorlist`, if it's indeed vanilla there should only be ArchLinux mirrors there.

> As for the version checking, not quite sure how to do that.

```
$ pacman -Q python-hidapi hidapi libusb
```

Yay has the same option, if you use it for all your package management.

> Output of the command:

Thanks!

CaseySJ, 2020-07-23:
@jonasmalacofilho,

Just a heads-up...I'll boot into Kubuntu Linux shortly and check the latest code. My system has the 0x5702 controller, but I can still test Linux compatibility of the latest code.

On macOS, however, the latest code (main branch) works correctly. It was downloaded and tested a couple of hours ago.

> @CaseySJ,
>
> Can you please take a look a this, since you're more familiar than me with this driver?
>
> At first glance it seems like an invalid response (report ID should not be 0 if the device [...]

CaseySJ, 2020-07-23:
** Update **

I am not seeing any problems running the latest master branch on Kubuntu Linux 20.04 on my Gigabyte Z490 Vision D. This board has the ITE Tech 0x5702. @sgtsixpack successfully tested an earlier Windows version on Gigabyte X570 with the 0x8297 controller.

```bash
casey@Z490-VISION-D:~/.local/lib/python3.8/site-packages/liquidctl/driver$ liquidctl --verbose initialize
Gigabyte RGB Fusion 2.0 5702 Controller (experimental)
├── Hardware name       IT5702-GIGABYTE V1.0.10.0  
├── Firmware version                     1.0.10.0  
└── LED channnels                               7  

casey@Z490-VISION-D:~/.local/lib/python3.8/site-packages/liquidctl/driver$ liquidctl --verbose list
Device ID 0: Gigabyte RGB Fusion 2.0 5702 Controller (experimental)
├── Vendor ID: 0x048d
├── Product ID: 0x5702
├── Release number: 0x0001
├── Bus: hid
├── Address: /dev/hidraw3
└── Driver: RGBFusion2Driver using module hidraw

casey@Z490-VISION-D:~/.local/lib/python3.8/site-packages/liquidctl/driver$ liquidctl -g initialize
[DEBUG] liquidctl.cli: running liquidctl v1.3.3
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CoolitPlatinumDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, RGBFusion2Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 5702 Controller (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4051
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:405b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:0411
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:082d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:5411
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0026
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 05ac:828d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0a5c:4500
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: Gigabyte RGB Fusion 2.0 5702 Controller (experimental)
[DEBUG] liquidctl.driver.usb: sending feature report 0xcc with 63 bytes: 60:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: got feature report 0xcc with 63 bytes: 01:00:07:01:00:0a:00:00:00:00:00:49:54:35:37:30:32:2d:47:49:47:41:42:59:54:45:20:56:31:2e:30:2e:31:30:2e:30:00:00:00:00:01:02:00:02:00:01:00:02:00:01:00:00:01:02:00:00:01:02:57:00:00:00:00
Gigabyte RGB Fusion 2.0 5702 Controller (experimental)
├── Hardware name       IT5702-GIGABYTE V1.0.10.0  
├── Firmware version                     1.0.10.0  
└── LED channnels                               7  
```

sgtsixpack, 2020-07-23:
Is this a new version?  Want me to test?  Have u changed anything which could affect things on my end?

jonasmalacofilho, 2020-07-23:
@sgtsixpack, I'm not sure what version you are running.  Just in case, can you update to [the build for `f44281d659aa`](https://ci.appveyor.com/api/buildjobs/xa7kb9vf0yqu8ack/artifacts/liquidctl-f44281d659aa-bin-windows-x86_64.zip) and test?

@CaseySJ, you had a earlier firmware version before?  Do you remember whether it was before or after 1.0.6.0?

sgtsixpack, 2020-07-23:
Some error here:

liquidctl initialize all
ERROR: Unexpected error with Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
Traceback (most recent call last):
AssertionError

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl --version
liquidctl v1.3.3 (AppVeyor; liquidctl-f44281d659aa-bin-windows-x86_64)

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>TIMEOUT 2

Waiting for 0 seconds, press a key to continue ...

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl -m Kraken set fan speed 23 40  24 49  25 55  26 58  27 60  28 66  29 82  30 92  31 98  32 100

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl -m Kraken set pump speed 25 90  26 100

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl -m Kraken set ring color fading FF2608 0B03FF F35B10 --speed fastest

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl -m Kraken set logo color fading F35B10 0B03FF FF2608 --speed fastest

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl -m Gigabyte set sync color static 0052FF

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl status
NZXT Kraken X (X42, X52, X62 or X72)
├── Liquid temperature     28.1  °C
├── Fan speed              1473  rpm
├── Pump speed             2703  rpm
└── Firmware version      6.0.2

CaseySJ, 2020-07-23:
@jonasmalacofilho, 

My Vision D has always has 1.0.10.0. As you can see from https://github.com/jonasmalacofilho/liquidctl/issues/127#issue-630419203, the original USB captures had 1.0.10.0. Perhaps @sgtsixpack reported 1.0.6.0 on the 0x8297?

> @CaseySJ, you had a earlier firmware version before? Do you remember whether it was before or after 1.0.6.0?

jonasmalacofilho, 2020-07-23:
@sgtsixpack can you also post the output of `liquidctl initialize --debug`?

sgtsixpack, 2020-07-23:
liquidctl initialize --debug
[DEBUG] __main__: running liquidctl v1.3.3 (AppVeyor; liquidctl-f44281d659aa-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CommonSmartDeviceDriver, CoolitPlatinumDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, RGBFusion2Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c232
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c231
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c227
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c227
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c227
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 09a6:8001
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c223
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 152d:2338
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
Too many devices, filter or select one (see: liquidctl --help)

jonasmalacofilho, 2020-07-23:
Sorry, my bad: `liquidctl initialize -m fusion --debug`.

sgtsixpack, 2020-07-23:
liquidctl initialize -m fusion --debug
[DEBUG] __main__: running liquidctl v1.3.3 (AppVeyor; liquidctl-f44281d659aa-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CommonSmartDeviceDriver, CoolitPlatinumDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, RGBFusion2Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c232
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c231
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c227
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c227
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c227
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 09a6:8001
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c223
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 152d:2338
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] __main__: device: Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.driver.usb: sending feature report 0xcc with 63 bytes: 60:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: got feature report 0x00 with 64 bytes: 01:00:01:01:00:06:00:00:00:00:00:49:54:38:32:39:37:42:58:2d:47:42:58:35:37:30:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02:00:01:00:02:00:01:00:00:01:02:00:00:01:97:82:00:00:00:00:61
[ERROR] __main__: Unexpected error with Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
Traceback (most recent call last):
  File "liquidctl\cli.py", line 301, in main
  File "liquidctl\driver\rgb_fusion2.py", line 185, in initialize
AssertionError

CaseySJ, 2020-07-23:
I'll check my installation of Windows with the 0x5702.

CaseySJ, 2020-07-23:
No errors on Windows 10 with 0x5702 device:
```
C:\Users\Casey\Downloads\liquidctl-f44281d659aa-bin-windows-x86_64\liquidctl-f44281d659aa-bin-windows-x86_64>.\liquidctl.exe --verbose initialize
Gigabyte RGB Fusion 2.0 5702 Controller (experimental)
├── Hardware name       IT5702-GIGABYTE V1.0.10.0
├── Firmware version                     1.0.10.0
└── LED channnels                               7

C:\Users\Casey\Downloads\liquidctl-f44281d659aa-bin-windows-x86_64\liquidctl-f44281d659aa-bin-windows-x86_64>.\liquidctl.exe -g initialize
[DEBUG] __main__: running liquidctl v1.3.3 (AppVeyor; liquidctl-f44281d659aa-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CommonSmartDeviceDriver, CoolitPlatinumDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, RGBFusion2Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 05ac:820a
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 5702 Controller (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 05ac:820b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 8086:15ec
[DEBUG] liquidctl.driver.usb: probing drivers for device 8086:15ec
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:0411
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 05ac:820a
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 05ac:828d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:5411
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0a5c:4500
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:082d
[DEBUG] liquidctl.driver.usb: probing drivers for device 05ac:820b
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0026
[DEBUG] liquidctl.driver.usb: probing drivers for device 8086:06ed
[DEBUG] __main__: device: Gigabyte RGB Fusion 2.0 5702 Controller (experimental)
[DEBUG] liquidctl.driver.usb: sending feature report 0xcc with 63 bytes: 60:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: got feature report 0xcc with 64 bytes: 01:00:07:01:00:0a:00:00:00:00:00:49:54:35:37:30:32:2d:47:49:47:41:42:59:54:45:20:56:31:2e:30:2e:31:30:2e:30:00:00:00:00:01:02:00:02:00:01:00:02:00:01:00:00:01:02:00:00:01:02:57:00:00:00:00:00
Gigabyte RGB Fusion 2.0 5702 Controller (experimental)
├── Hardware name       IT5702-GIGABYTE V1.0.10.0
├── Firmware version                     1.0.10.0
└── LED channnels                               7
```

sgtsixpack, 2020-07-23:
The previous version liquidctl-7c6afd0b01a2-bin-windows-x86_64  had no errors for me.

This one is the one I just tested:
liquidctl-f44281d659aa-bin-windows-x86_64

CaseySJ, 2020-07-23:
> The previous version liquidctl-7c6afd0b01a2-bin-windows-x86_64 had no errors for me.
> 
> This one is the one I just tested:
> liquidctl-f44281d659aa-bin-windows-x86_64

The 0x8297 device does not seem to be placing REPORT_ID `0xcc` into the first byte of the reply. The latest code sends REPORT_ID `0xcc` and checks the first byte of the reply for `0xcc`, otherwise it raises an assertion that you're seeing. 

The 0x5702 device returns REPORT_ID `0xcc` so no assertion is raised.

sgtsixpack, 2020-07-23:
U think a restart could fix it?  Maybe you could write a bit of code to save what it replies with?  Would that help?

jonasmalacofilho, 2020-07-23:
@sgtsixpack 

> The previous version liquidctl-7c6afd0b01a2-bin-windows-x86_64 had no errors for me.

That's unexpected, as I can't sse any functional changes to that the assertion or the lower level code that could explain this difference.

Regardless, please test [8cc5c788307d](https://ci.appveyor.com/api/buildjobs/1jm405xgxovkffq5/artifacts/liquidctl-8cc5c788307d-bin-windows-x86_64.zip).

---

@haringsrob 

Please test `8cc5c788307d` too.

I would also like to see the HID descriptor, can you please post the output of `sudo usbhid-dump -d 048d:8297`?

sgtsixpack, 2020-07-23:
So I tested the one I thought was working and got the same output:
C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl initialize all
ERROR: Unexpected error with Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
Traceback (most recent call last):
AssertionError

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl --version
liquidctl v1.3.3 (AppVeyor; liquidctl-7c6afd0b01a2-bin-windows-x86_64)

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>TIMEOUT 2

Waiting for 0 seconds, press a key to continue ...

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl -m Kraken set fan speed 23 40  24 49  25 55  26 58  27 60  28 66  29 82  30 92  31 98  32 100

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl -m Kraken set pump speed 25 90  26 100

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl -m Kraken set ring color fading FF2608 0B03FF F35B10 --speed fastest

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl -m Kraken set logo color fading F35B10 0B03FF FF2608 --speed fastest

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl -m Gigabyte set sync color static 0052FF

C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp>liquidctl status
NZXT Kraken X (X42, X52, X62 or X72)
├── Liquid temperature     28.4  °C
├── Fan speed              1434  rpm
├── Pump speed             2703  rpm
└── Firmware version      6.0.2

sgtsixpack, 2020-07-23:
Latest version (8cc5c788307d):

 liquidctl initialize -m fusion --debug
[DEBUG] __main__: running liquidctl v1.3.3 (AppVeyor; liquidctl-8cc5c788307d-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CommonSmartDeviceDriver, CoolitPlatinumDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, RGBFusion2Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c232
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c231
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c227
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c227
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c227
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 09a6:8001
[DEBUG] liquidctl.driver.usb: probing drivers for device 1241:1503
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c223
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1058:25a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 152d:2338
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c226
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0084
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] __main__: device: Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.driver.usb: sending feature report 0xcc with 63 bytes: 60:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: got feature report 0x00 with 64 bytes: 01:00:01:01:00:06:00:00:00:00:00:49:54:38:32:39:37:42:58:2d:47:42:58:35:37:30:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02:00:01:00:02:00:01:00:00:01:02:00:00:01:97:82:00:00:00:00:30
Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
├── Hardware name       IT8297BX-GBX570
├── Firmware version            1.0.6.0
└── LED channels                      1

jonasmalacofilho, 2020-07-23:
Does this number of channels (1) makes sense?

sgtsixpack, 2020-07-23:
Manual says:

8) LED_C1/LED_C2 (RGB LED strip Headers)
12v/G/R/B

9) D_LED1/D_LED2 (Addressable LED Strip Headers)
5v/D/nc/G

10) LED_CPU (CPU Cooler LED Strip/RGB LED Strip Header)
12v/G/R/B

but 8 & 9 are duplicated on different corners of the board.

https://www.gigabyte.com/Motherboard/X570-AORUS-ELITE-rev-10/support#support-manual

So my board has at three channels for plugable strips (5 plugables in total).  It has lighting on the motherboard itself.  I believe on the IO port and near the onboard audio.  So that is five channels.  I need to check though, and I do not have RGB fusion installed.

sgtsixpack, 2020-07-23:
I need to test the channels individually, I've been using sync and I can't find the commands in the manual.  Maybe its in the repository.

CaseySJ, 2020-07-23:
Your board certainly has more than 1 LED channel. I just took a glance at the OpenRGB project on GitHub, in which it seems that the same assumption is made that the 4th byte (3rd after REPORT_ID) is "total LEDs". (This is the first time I'm looking at that code.)
https://github.com/CalcProgrammer1/OpenRGB/blob/9a33a78574e8dfc54b0408288907cbe586bc0635/Controllers/RGBFusion2USBController/RGBFusion2USBController.h#L138-L153

jonasmalacofilho, 2020-07-23:
@CaseySJ agreed.

sgtsixpack, 2020-07-23:
Just incase u missed it:

"LED1: IOports
LED2: no effect/ not plugged in
LED3: no effect/ not plugged in
LED4: HD audio
LED5: no effect/ not plugged in
LED6: My O11 dynamic case strip is plugged into this (exact location on board needs to be checked)
LED7: no effect/ not plugged in"

My case strip is plugged in at the bottom either 8 or 9.  
https://imgur.com/a/LUOK7rF

CaseySJ, 2020-07-23:
Okay, we can try enabling LED8. We would need to change [this code](https://github.com/jonasmalacofilho/liquidctl/blob/2c24b020134989c2ffec0028586aa7bb48f4ccd4/liquidctl/driver/rgb_fusion2.py#L84-L92) to this:
```Python
_COLOR_CHANNELS = {
    'led1': (0x20, 0x01),
    'led2': (0x21, 0x02),
    'led3': (0x22, 0x04),
    'led4': (0x23, 0x08),
    'led5': (0x24, 0x10),
    'led6': (0x25, 0x20),
    'led7': (0x26, 0x40),
    'led8': (0x27, 0x80),
}
```

> I used to have RGB ram, I'm wondering is this one of the LED channels? I still have it but I have Patriot Viper 4400 which even @ 3600 c16 is better than Corsair Vengence Pro RGB M2Z3600C18.

jonasmalacofilho, 2020-07-24:
Notice: this release may be affected by #157.  I'm also expecting a similar issue to be opened for Fedora, since it too has updated to the broken `hidapi` modules.

haringsrob, 2020-07-24:
Hey, seems like all is good now.

```
 rob@harings  /  liquidctl list
Device ID 0: Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
 rob@harings  /  liquidctl initialize all --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (EndeavourOS; liquidctl-git 1.3.3.r284.2c24b02-1)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CommonSmartDeviceDriver, CoolitPlatinumDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, RGBFusion2Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device feed:1307
[DEBUG] liquidctl.driver.usb: probing drivers for device feed:1307
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:407f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:4937
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:085c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:4937
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c539
[DEBUG] liquidctl.driver.usb: probing drivers for device 0a12:0001
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c13
[DEBUG] liquidctl.driver.usb: probing drivers for device feed:1307
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.driver.usb: sending feature report 0xcc with 63 bytes: 60:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: got feature report 0x00 with 63 bytes: 01:00:01:01:00:06:00:00:00:00:00:49:54:38:32:39:37:42:58:2d:47:42:58:35:37:30:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02:00:01:00:02:00:01:00:00:01:02:00:00:01:97:82:00:00:00:00
Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
├── Hardware name       IT8297BX-GBX570
└── Firmware version            1.0.6.0
```

jonasmalacofilho, 2020-07-24:
@haringsrob,

This controller doesn't support fan control or report anything any (variable) status.  So those are expected (although I agree that the error message could be improved).

But can you show me the HID descriptor of this device?  You can get it by installing `usbutils` and running `sudo usbhid-dump -d 048d:8297`.

haringsrob, 2020-07-24:
Sure: 
```
003:002:000:DESCRIPTOR         1595619527.478157
 06 89 FF 09 10 A1 01 85 5A 09 01 15 00 26 FF 00
 75 08 95 10 B1 00 C0 06 89 FF 09 CC A1 01 85 CC
 09 01 15 00 26 FF 00 75 08 95 3F B1 00 C0
```

jonasmalacofilho, 2020-07-24:
@haringsrob Thank you!

---

@CaseySJ, 

Decoding this (using a [convenient online tool](https://eleccelerator.com/usbdescreqparser/)) we get...

```
0x06, 0x89, 0xFF,  // Usage Page (Vendor Defined 0xFF89)
0x09, 0x10,        // Usage (0x10)
0xA1, 0x01,        // Collection (Application)
0x85, 0x5A,        //   Report ID (90)
0x09, 0x01,        //   Usage (0x01)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x10,        //   Report Count (16)
0xB1, 0x00,        //   Feature (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection
0x06, 0x89, 0xFF,  // Usage Page (Vendor Defined 0xFF89)
0x09, 0xCC,        // Usage (0xCC)
0xA1, 0x01,        // Collection (Application)
0x85, 0xCC,        //   Report ID (-52)
0x09, 0x01,        //   Usage (0x01)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x3F,        //   Report Count (63)
0xB1, 0x00,        //   Feature (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection

// 46 bytes
```

This all pretty much in line with our driver, so the the device should have responded with with the report ID.

I'll add some notes in the code indicating that this has been checked and that it is indeed a bug in the device, not our code.

CaseySJ, 2020-07-24:
@jonasmalacofilho,

Let's also add `led8` per [this comment](https://github.com/jonasmalacofilho/liquidctl/issues/151#issuecomment-663287869) above.

mburgholte, 2020-07-28:
Hi,
I've a problem under Linux Mint 19 and kernel 5.4.0-42-generic. Initialization throws an error. Everything else works like a charm. By the way, I don't need RGB Fusion. Is there a way to ignore this device?
```
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (4028c26e1b05)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RGBFusion2Driver, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:1711
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Grid+ V3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2002
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT HUE 2 Ambient
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1610
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1610
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1610
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1610
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1610
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1726
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1726
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1726
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1726
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:274c
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:5734
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:274c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1726
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1610
[DEBUG] liquidctl.driver.usb: probing drivers for device 0424:2734
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2002
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2001
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:1711
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2006
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.driver.usb: sending feature report 0xcc with 63 bytes: 60:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
ioctl (GFEATURE): Value too large for defined data type
[ERROR] liquidctl.cli: Unexpected error with Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/liquidctl-1.3.3-py3.6.egg/liquidctl/cli.py", line 300, in main
    _print_dev_status(dev, dev.initialize(**opts))
  File "/usr/local/lib/python3.6/dist-packages/liquidctl-1.3.3-py3.6.egg/liquidctl/driver/rgb_fusion2.py", line 184, in initialize
    data = self._get_feature_report(_REPORT_ID)
  File "/usr/local/lib/python3.6/dist-packages/liquidctl-1.3.3-py3.6.egg/liquidctl/driver/rgb_fusion2.py", line 288, in _get_feature_report
    return self.device.get_feature_report(report_id, _REPORT_BYTE_LENGTH)
  File "/usr/local/lib/python3.6/dist-packages/liquidctl-1.3.3-py3.6.egg/liquidctl/driver/usb.py", line 433, in get_feature_report
    LOGGER.debug('got feature report 0x%02x with %d bytes: %r', data[0],
IndexError: list index out of range

jonasmalacofilho, 2020-07-28:
@mburgholte Can you show me the HID descriptor for your RGB Fusion? You can get it by installing usbutils and running sudo usbhid-dump -d 048d:8297.

mburgholte, 2020-07-28:
Hi @jonasmalacofilho,
for sure. Please find below the requested informations.
```
usbhid-dump -d 048d:8297
001:002:000:DESCRIPTOR         1595957044.011398
 06 89 FF 09 10 A1 01 85 5A 09 01 15 00 26 FF 00
 75 08 95 10 B1 00 C0 06 89 FF 09 CC A1 01 85 CC
 09 01 15 00 26 FF 00 75 08 95 3F B1 00 C0

lsusb -v -d 048d:8297

Bus 001 Device 002: ID 048d:8297 Integrated Technology Express, Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x048d Integrated Technology Express, Inc.
  idProduct          0x8297 
  bcdDevice            0.03
  iManufacturer           1 ITE Tech. Inc.
  iProduct                2 ITE Device(8595)
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      46
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0001
  Self Powered

jonasmalacofilho, 2020-07-28:
@haringsrob,

Due to this new issue with the 8297 controllers, can I also ask you for the output of `lsusb -v -d 048d:8297`?

jonasmalacofilho, 2020-07-28:
@mburgholte 

In the meantime, you can also use the filter options (of which `-m <case insensitive substring of the description>` is generally the most convenient) to restrict which devices you want `status` and `initialize` to work on.

It's an inclusive filter, and I honestly don't plan to add an exclusive one.  But, luckily, all of the other devices will match `-m nzxt`, so you can use that to "ignore" the problematic motherboard controller.

---

When you said that "Everything else works like a charm", did you mean the other devices or other commands on the same RGB Fusion controller?

sgtsixpack, 2020-07-28:
Can you troubleshoot my path again plz?  I just reinstalled with a 56GB Raid 10 (4 x 14TB £170 WD140EMFZ, R/N US7SAP140).

Powershell told me to use the command .\liquidctl and that worked but I don't know why my path isn't.  Under environmental variables/ User variables for SgtSixpack Path  %USERPROFILE%\AppData\Local\Microsoft\WindowsApps; %USERPROFILE%\SgtSixpack\Documents

Going to reboot 1 more time, and see if I fixed it.

PS C:\users\sgtsixpack\documents> liquidctl -h
liquidctl : The term 'liquidctl' is not recognized as the name of a cmdlet, function, script file, or operable
program. Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ liquidctl -h
+ ~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (liquidctl:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

Suggestion [3,General]: The command liquidctl was not found, but does exist in the current location. Windows PowerShell does not load commands from the current location by default. If you trust this command, instead type: **".\liquidctl"**. See "get-help about_Command_Precedence" for more details.
PS C:\users\sgtsixpack\documents> .\liquidctl -h
liquidctl – monitor and control liquid coolers and other devices.

Usage:
  liquidctl [options] list
  liquidctl [options] initialize [all]
  liquidctl [options] status
  liquidctl [options] set <channel> speed (<temperature> <percentage>) ...
  liquidctl [options] set <channel> speed <percentage>
  liquidctl [options] set <channel> color <mode> [<color>] ...
  liquidctl --help
  liquidctl --version

Device selection options (see: list -v):
  -m, --match <substring>     Filter devices by description substring
  -n, --pick <number>         Pick among many results for a given filter
  --vendor <id>               Filter devices by vendor id
  --product <id>              Filter devices by product id
  --release <number>          Filter devices by release number
  --serial <number>           Filter devices by serial number
  --bus <bus>                 Filter devices by bus
  --address <address>         Filter devices by address in bus
  --usb-port <port>           Filter devices by USB port in bus
  -d, --device <id>           Select device by listing id

Animation options (devices/modes can support zero or more):
  --speed <value>             Abstract animation speed (device/mode specific)
  --time-per-color <value>    Time to wait on each color (seconds)
  --time-off <value>          Time to wait with the LED turned off (seconds)
  --alert-threshold <number>  Threshold temperature for a visual alert (°C)
  --alert-color <color>       Color used by the visual high temperature alert

Other device options:
  --single-12v-ocp            Enable single rail +12V OCP
  --pump-mode <mode>          Set the pump mode (certain Corsair coolers)
  --legacy-690lc              Use Asetek 690LC in legacy mode (old Krakens)
  --unsafe <features>         Comma-separated bleeding-edge features to enable

Other interface options:
  -v, --verbose               Output additional information
  -g, --debug                 Show debug information on stderr
  --version                   Display the version number
  --help                      Show this message

Deprecated:
  --hid <ignored>             Deprecated

Examples:
  liquidctl list --verbose
  liquidctl initialize all
  liquidctl --match kraken set pump speed 90
  liquidctl --product 0x170e set led color fading 350017 ff2608
  liquidctl status

Copyright (C) 2018–2020  Jonas Malaco, CaseySJ, Tom Frey and contributors

liquidctl incorporates work by leaty, Ksenija Stanojevic, Alexander Tong, Jens
Neumaier, Kristóf Jakab, Sean Nelson, Chris Griffith, notaz, realies and Thomas
Pircher.

SPDX-License-Identifier: GPL-3.0-or-later

This program is distributed in the hope that it will be useful, but WITHOUT ANY
WARRANTY; without even the implied warranty of MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.  See the GNU General Public License for more details.
PS C:\users\sgtsixpack\documents>

jonasmalacofilho, 2020-07-29:
@sgtsixpack,

This was off-topic; in the future, please try reach me through email, twitter or discord for general help.  That said, I think your problem is that you meant to use `%USERPROFILE%\Documents`, since `%USERPROFILE%` should already point to `C:\Users\SgtSixpack\`.

haringsrob, 2020-07-29:
Hey @jonasmalacofilho here it is:

```
❯ lsusb -v -d 048d:8297

Bus 003 Device 003: ID 048d:8297 Integrated Technology Express, Inc. ITE Device(8595)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x048d Integrated Technology Express, Inc.
  idProduct          0x8297
  bcdDevice            0.03
  iManufacturer           1 ITE Tech. Inc.
  iProduct                2 ITE Device(8595)
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      46
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

mburgholte, 2020-07-29:
> @mburgholte
> 
> In the meantime, you can also use the filter options (of which `-m <case insensitive substring of the description>` is generally the most convenient) to restrict which devices you want `status` and `initialize` to work on.
> 
> It's an inclusive filter, and I honestly don't plan to add an exclusive one. But, luckily, all of your other devices will match `-m nzxt`, so you can use that to "ignore" the problematic motherboard controller.
> 
> When you said that "Everything else works like a charm", did you mean the other devices or other commands on the same RGB Fusion controller?

Hi @jonasmalacofilho,
I don't use this rgb controller. My NZXT devices are working very well.

jonasmalacofilho, 2020-07-29:
@mburgholte,

Got it, thanks!

Can you test if you're again able to `initialize all` devices with 0c8f1bdbcd4a ("rgb_fusion2: Account for the report ID in get_feature_report buffer size")?

mburgholte, 2020-07-31:
> @mburgholte,
> 
> Got it, thanks!
> 
> Can you test if you're again able to `initialize all` devices with [0c8f1bd](https://github.com/jonasmalacofilho/liquidctl/commit/0c8f1bdbcd4acee8e6a99c61e12247c43d472e99) or later?

Hi @jonasmalacofilho,
it works! Thanks a lot.

jonasmalacofilho, 2020-07-31:
@mburgholte, great!  And once again, thanks!

jonasmalacofilho, 2020-07-31:
Done, liquidctl v1.4.0 is out!  : )

closed 2020-07-31 by jonasmalacofilho

CaseySJ, 2020-07-31:
@sgtsixpack,

We added `led8` because it seems RGB Fusion controller on your X570 might use it. Have you had a chance to test it?

sgtsixpack, 2020-07-31:
I'm quite busy testing my raid 10 install, 
[/Offtopic]
everything is on hold till I can get my data on my new disks.  Every hurdle has been against me.  Like being forced to install windows on an array with no write cache (apparently even the onboard cache of the disks =500MB is disabled & u need write caching but I have no UPS).  Currently testing out freeware to clone my install to a USB (so far this is via a clone of the array to an old 1TB with USB 2.0 to Sata interface, as clonezilla does not see the array only disks).  Unsure how to clone the 16MB system partition which is hidden is disk management but shows in minitool partition wizard ( have tried clonezilla partition to partition (but it needs the filesystem/partition setup first & minitool partition wizard; both fail to produce a bootable USB).  Now I'm using imageusb 👍 https://www.iceflatline.com/2017/12/how-to-clone-your-windows-10-installation-usb-drive-to-another-usb-drive/
to test if it makes my usb clone bootable.  At least then I have a ~130MB read speed to clone back to the array after changing RAID parameters in the bios.
[/Ontopic]
I was wondering if anyone else had tested [led8] before putting it in 1.4.  Also I was going to post that there are a few typos in the docs (sorry I've lost track where to find but I found 2), 
[/Offtopic]
but I must cook dinner first (I cook ~ once a week).

CaseySJ, 2020-07-31:
@sgtsixpack 

Understood; no rush. If `led8` does not work (although it should be okay) simply open **new** issue.

P.S. I used to cook once a week before the pandemic, but working from home changes many routines. Cooking is enjoyable, especially if you grow many of your own vegetables!

jonasmalacofilho, 2020-07-31:
@sgtsixpack thanks for letting me know about the typos.

_(Something confused the markdown syntax highlighter, and that made the spell checker completely ignore all normal text...)_

---

@CaseySJ,

But have you been cooking more, or less?  I usually find it very enjoyable, lately not so much...

sgtsixpack, 2020-07-31:
Oh btw I made a flat gradient (except for between 37 & 38'c water temp) of 4.61% fan speed between 25 & 38 degrees water temperature as my 3 x 3k Noctua industrial fans were idling @tooloud%:
@echo
liquidctl initialize all
liquidctl --version
TIMEOUT 2
@liquidctl -m Kraken set fan speed 25 40  26 49  27 55  28 58  29 60  30 66  31 82  32 92  33 98  34 100
liquidctl -m Kraken set fan speed 25 44  26 48  27 53  28 57  29 62  30 67  31 71  32 76  33 81  34 85  35 90  36 94  37 99  38 100
liquidctl -m Kraken set pump speed 25 90  26 100
liquidctl -m Kraken set ring color fading FF2608 0B03FF F35B10 --speed fastest
liquidctl -m Kraken set logo color fading F35B10 0B03FF FF2608 --speed fastest
@liquidctl -m Gigabyte set sync color fixed ff2000
liquidctl -m Gigabyte set sync color fixed 0052FF
liquidctl status
TIMEOUT 30
Adjust temps for your climate.

CaseySJ, 2020-07-31:
Been cooking a lot more and trying new recipes. It's wonderful when you can walk out into the garden (100% organic of course) and pick ripe vegetables for consumption the same day. Just-in-time delivery never tasted so good.

> @CaseySJ,
> 
> But have you been cooking more, or less? I usually find it very enjoyable, lately not so much...

jonasmalacofilho, 2020-08-04:
@sgtsixpack,

Cool (pun intended).  But, in the future, liquidctl can do the math for you.

```
liquidctl -m kraken set fan speed 25 44 37 99 38 100
```

Should be equivalent to your manual profile, ± rounding errors.  You can check this by adding `--verbose` to the command.

---

@CaseySJ,

That's sounds super nice.  Unfortunately I live in an apartment, so I'm limited to very small plants.  But I should look into it!  Do you have any suggestions besides the usual herbs and spices?

CaseySJ, 2020-08-04:
@jonasmalacofilho,

When the pandemic began it gave me the opportunity to put more time and effort into the backyard, which is something I had neglected for too long. I started by building six raised planter boxes, each measuring 0.61m (2ft) x 1.52m (5ft) and then watching a number of YouTube gardening channels, among them:

- Huw Richards
- CaliKim29
- James Prigioni
- Epic Gardening

Good lessons on organic soil, starting from seed, soil amendment, plant nutrition at different phrases, organic pest control methods (cold pressed neem oil with soap), etc.

I've added some photos to my [Flickr album](https://www.flickr.com/photos/188809372@N03/albums/72157715358613348). This began in April and now in August it's astonishing how much the plants have grown and how much food they're producing. 

This being my first ever attempt at gardening, I decided to grow:

- Heirloom and cherry tomatoes
- Okra
- Cantaloupe
- Artichoke
- Garlic
- Swiss Chard
- Kale
- Zucchini
- Jalapeno and Thai hot peppers
- Green and Red bell peppers
- Japanese eggplant

I should have done this sooner! 

> @CaseySJ,
> 
> That's sounds super nice. Unfortunately I live in an apartment, so I'm limited to very small plants. But I should look into it! Do you have any suggestions besides the usual herbs and spices?