liquidctl/liquidctl #152: Fan/Pump speed not tracking correctly.

RAGEdemon, 2020-07-18:
Hardware:
Max fan speed = 1700 rpm (custom PWM fans)
Max pump speed = 2700 rpm (Kraken x62)

Set commands:
liquidctl set fan speed  34 0  36 50  37 80  38 100
liquidctl set pump speed  35 60  36 80  37 100

Problem Example:
Expected behaviour @ 37.7 degrees: 
fan Speed = 80% (~1400 rpm)
pump speed = 100% (~2700 rpm)

Actual behaviour:
fan speed = 50% (~750 rpm)
pump speed = 80% (~2300 rpm)

Reference Image: https://i.imgur.com/2Ngbass.jpg

This behaviour could line up with the 36 degrees command setting, i.e. there is 1 degree difference between command setting and actual output - but it is fickle - sometimes it randomly works, other times it doesn't - there doesn't seem to be any logic. if it was a constant known error, it could be compensated for.

Suggestions/advice would be appreciated. Thank you.

RAGEdemon, 2020-07-18:
-- verbose command revealed program only works in increments of 2 degrees - in process of further testing.

Recommend changing program behaviour to 0.5 degree increments - the difference in coolant temperature between idle and fully loaded is only a few degrees - unfortunately 2 degrees increments for a few degrees variation is too low a resolution.

thank you.

closed 2020-07-18 by RAGEdemon

jonasmalacofilho, 2020-07-18:
That was not a recommendation.

Your _feature request_ for 0.5°C increments has been noted, but is not feasible.

That said, I was been able to push the our implementation of the protocol further still (notice that 2°C increments were already a substantial improvement over the stock 5°C increments), and the code in the main branch is currently capable of 1°C steps.