liquidctl/liquidctl #153: OSError: open failed  on Ubuntu 16.04 for NZXT x72

doonny, 2020-07-20:
I am using liquidctl on Ubuntu 16.04.6 with NZXT x72, after successfully compiled the src, I could list the device as follow:

liquidctl list
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)

But when I try to init the pump, the following error shows up:
**OSError: open failed**

I have added the rules file according the instructions shown here : https://github.com/jonasmalacofilho/liquidctl/blob/master/extra/linux/71-liquidctl.rules

Has anyone faced the same issue ?

More information on the system:

Python 3.6.11 (default, Jun 29 2020, 05:15:03)
[GCC 5.4.0 20160609] on linux

Linux fpga-z390 4.15.0-107-generic #108~16.04.1-Ubuntu SMP Fri Jun 12 02:57:13 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

jonasmalacofilho, 2020-07-20:
The udev rules may simply have not yet had an effect.  Sometimes a reboot is all that is necessary[1], but given the relatively old version of Ubuntu, it could also be that the installed udev doesn't yet support `TAG+="uaccess"`.

You may want to check `journalctl -b` if rebooting doesn't improve things.

The header of the rules file mentions an alternative approach to uaccess tagging, but it won't necessarily apply your system either (for instance, it assumes that a `pludev` group exists, which was common back then, but probably not ubiquitous).

> ```
> # These rules assume a system with modern versions of systemd/udev, that
> # support the `uaccess` tag; on older systems the rules can be changed to set
> # GROUP="plugdev" and MODE="0660" instead.  The currently deprecated 'plugdev'
> # group is not used by default to avoid generating warnings on systems that
> # have already removed it.
> ```

It's probably better to first check whether this is just an udev problem, or something more complicated.  Running the same liquidctl command with root privileges/`sudo` should succeed if it's the former.

doonny, 2020-07-20:
Hi thanks for the reply.  It is an udev problem, I could run liquidctl with sudo privileges.  

Problem fixed after I used the following rules here:

SUBSYSTEMS=="usb", ATTRS{idVendor}=="1e71", ATTRS{idProduct}=="170e", GROUP="myGroup", MODE="0666"

Thx again, this is an awesome tool !

jonasmalacofilho, 2020-07-21:
Happy to hear!

And thanks!

closed 2020-07-21 by jonasmalacofilho