liquidctl/liquidctl #154: Corsair Lighting Node Core Support

sam-brett, 2020-07-20:
Hello, 

I've just started using this and I was wondering if there would be any support for the Corsair Lighting Node? I've attached the system profiler data below. I believe this would be an extension of the implementation from the corsair hydro series. 

``` 
CORSAIR Lighting Node CORE:
              Product ID: 0x0c1a
              Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
              Version: 0.09
              Serial Number: 36C0020<snip>C00F5
              Speed: Up to 12 Mb/s
              Manufacturer: Corsair
              Location ID: 0x14a40000 / 8
              Current Available (mA): 500
              Current Required (mA): 50
              Extra Operating Current (mA): 0
```

sam-brett, 2020-07-20:
I’m a java dev, but I’m happy to do some work, I just might need a bit of spoon feeding.

jonasmalacofilho, 2020-07-20:
Sure, I'll be happy to help.

The first step is to capture some USB traffic between iCue and the device.  [This relatively long comment](https://github.com/jonasmalacofilho/liquidctl/issues/142#issuecomment-650568291) goes over what we're generally looking for.

And [this other comment](https://github.com/jonasmalacofilho/liquidctl/issues/129#issuecomment-640258429), while focused on porting driver from OCL (which probably wont happen here), delves a bit into how the liquidctl code is organized.

Keep me updated and let me know if you have any questions along the way.  Once you have some initial pcap files we can start to try finding similarities between protocols already implemented here or by OpenCorsairLink.

wmchris, 2020-07-27:
dunno if https://github.com/phillip-smith/CorsairCommanderProControl will help. Should use the same protocol.

sam-brett, 2020-07-30:
I've done some captures here. There is one where I let it run on the rainbow setting, the other I cycled through the "Instant Lighting" defaults. 

[Archive.zip](https://github.com/jonasmalacofilho/liquidctl/files/5001961/Archive.zip)

jonasmalacofilho, 2020-08-04:
@sam-brett,

I was only able to quickly look at the captures, but at first glance the protocol does appear to be the same to the Commander Pro.

See if you agree with me, OpenCorsairLink's implementation is (mostly) in [`OpenCorsairLink/protocol/commanderpro`](https://github.com/audiohacked/OpenCorsairLink/tree/testing/protocol/commanderpro).  You'll probably also want to use the following Wireshark filter:

```
usb.darwin.device_address == 22 && usb.capdata
```

_Note: the `usb.capdata` part of this filter seems ok so far, but can sometimes be a bit too aggressive._

And add a new column with type Custom and expression `usb.capdata`.

---

The previously linked docs on [porting drivers from OpenCorsairLink](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/developer/porting-drivers-from-opencorsairlink.md) is a good starting point, including understanding the code structure of both projects.

MarshallAsch, 2020-12-29:
@sam-brett Hey I have completed support for the Commander Pro and the Lighting Node Pro. I believe that the lighting node core should use the same protocol as those two devices. Would you be able to test [this](https://github.com/MarshallAsch/liquidctl/tree/commander-pro) version and see if it works for your device. 

you will have to add this to line `120` in the `liquidctl/driver/commander_pro.py` file.

```
(0x1B1C, 0x0C1A, None, 'Corsair Lighting Node Core (experimental)',
            {'fan_count': 0, 'temp_probs': 0, 'led_channels': 2}),
```

jonasmalacofilho, 2021-01-13:
Ping @sam-brett.

kevinlester, 2021-01-17:
@MarshallAsch Thank you for creating the patch to support the Lightning Node Core (LNC).  I gave it a try on my setup and it was recognized successfully, but I couldn't get some of the lighting effects to work.  My setup is:
- Lightning Node Core plugged into the Commander Pro
- Commander Pro plugged into the motherboard
- H100i Pro XT with fan power plugged into the AIO, and RGBs plugged into the LNC.
- 6 QL120s (inclusive of the ones in the H100i) hooked up directly to the LNC.

Output of relevant commands:
```
> liquidctl initialize all
Corsair Lighting Node Core (experimental)
├── Firmware version      0.9.16  
└── Bootloader version       3.0  

Corsair H100i PRO XT (experimental)
└── Firmware version    1.1.22  

Corsair Commander Pro (experimental)
├── Firmware version                0.9.212  
├── Bootloader version                  0.5  
├── Temp sensor 1             Not Connected  
├── Temp sensor 2             Not Connected  
├── Temp sensor 3             Not Connected  
├── Temp sensor 4             Not Connected  
├── Fan 1 Mode                          PWM  
├── Fan 2 Mode                          PWM  
├── Fan 3 Mode                          PWM  
├── Fan 4 Mode            Auto/Disconnected  
├── Fan 5 Mode            Auto/Disconnected  
└── Fan 6 Mode                          PWM  
```
The above looks good.

```
> liquidctl status
WARNING: DDR4 DIMM2 (experimental): nothing to return, requires unsafe features 'smbus,ddr4_temperature'
WARNING: DDR4 DIMM4 (experimental): nothing to return, requires unsafe features 'smbus,ddr4_temperature'
Corsair H100i PRO XT (experimental)
├── Liquid temperature    24.0  °C
├── Fan 1 speed           1538  rpm
├── Fan 2 speed           1534  rpm
└── Pump speed            2290  rpm

Corsair Commander Pro (experimental)
├── 12 volt rail     11.93  V
├── 5 volt rail       5.00  V
├── 3.3 volt rail     3.33  V
├── Temp sensor 1      0.0  °C
├── Temp sensor 2      0.0  °C
├── Temp sensor 3      0.0  °C
├── Temp sensor 4      0.0  °C
├── Fan 1 speed        958  rpm
├── Fan 2 speed        944  rpm
├── Fan 3 speed        952  rpm
├── Fan 4 speed          0  rpm
├── Fan 5 speed          0  rpm
└── Fan 6 speed        895  rpm
```
The LNC doesn't report a status, but I think that's expected.  Not sure why my DIMMS show up.

```
liquidctl --match 'Lighting Node Core' set led1 color fixed 00ff00 --start-led 1
```
Works!  The first LED changes to green 

```
liquidctl --match 'Lighting Node Core' set led1 color fixed 00ff00 --start-led 1 --num-leds 48
Usage:
  liquidctl [options] list
  liquidctl [options] initialize [all]
  liquidctl [options] status
  liquidctl [options] set <channel> speed (<temperature> <percentage>) ...
  liquidctl [options] set <channel> speed <percentage>
  liquidctl [options] set <channel> color <mode> [<color>] ...
  liquidctl --help
  liquidctl --version
```
Doesn't work; it doesn't seem to like the --num-leds option for some reason, and I tried a few different ways of specifying it.  Without being able to impact multiple LEDs I couldn't test a few of the modes, but rainbow, rainbow2, color_pulse, blink all seemed to work.  Also note, that I could only change the modes a few times before it stopped responding :(. 

Question:
1. Is there a way to un-initialize liquidctl so that when the mode gets stuck I can give it a kick?
2. I was surprised to see that the mode changes persisted across a reboot.  Is there a way I can reset the fans back to their default rainbow mode when I'm not running liquidctl?

Cheers!

kevinlester, 2021-01-17:
> Also note, that I could only change the modes a few times before it stopped responding :(.
[](url)

Below is a debug log for a failed liquidctl command.  I executed several successful commands prior to this one (though this one was not the first failed command in the sequence).  Example failed command:  `liquidctl --match 'Lighting Node Core' set led1 color fixed 000000 --start-led 1 --debug 2> out.txt`. 

Debug log details:
<details>
<summary>Details</summary>
[DEBUG] liquidctl.cli: running liquidctl v1.4.2 (a3c37784e05a-dirty)

[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-10
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-8
[DEBUG] liquidctl.driver.smbus: ignoring 0-0037, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-13
[DEBUG] liquidctl.driver.smbus: ignoring 0-0053, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-11
[DEBUG] liquidctl.driver.smbus: ignoring 0-0051, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-9
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-7
[DEBUG] liquidctl.driver.smbus: ignoring 0-0036, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-12
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 0b05:18f3
[DEBUG] liquidctl.driver.usb: found HID device 275d:0ba6
[DEBUG] liquidctl.driver.usb: found HID device 275d:0ba6
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c1a
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair Lighting Node Core (experimental)
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c20
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c10
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0220
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0220
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 05ac:0220
[DEBUG] liquidctl.driver.usb: found USB device 05ac:1006
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c10
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c20
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c1a
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0610
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0610
[DEBUG] liquidctl.driver.usb: found USB device 275d:0ba6
[DEBUG] liquidctl.driver.usb: found USB device 0b05:18f3
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair Lighting Node Core (experimental)
[DEBUG] liquidctl.keyval: data in /var/run/liquidctl/vid1b1c_pid0c1a/loc2
[DEBUG] liquidctl.keyval: loaded saved_effects=[{'channel': 0, 'start_led': 0, 'num_leds': 1, 'mode': 8, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [0, 255, 0]}, {'channel': 0, 'start_led': 0, 'num_leds': 1, 'mode': 8, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [0, 0, 255]}, {'channel': 0, 'start_led': 0, 'num_leds': 1, 'mode': 8, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 0]}, {'channel': 0, 'start_led': 1, 'num_leds': 1, 'mode': 8, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 0]}, {'channel': 0, 'start_led': 1, 'num_leds': 1, 'mode': 8, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 255]}, {'channel': 0, 'start_led': 3, 'num_leds': 1, 'mode': 8, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 255]}, {'channel': 0, 'start_led': 3, 'num_leds': 1, 'mode': 2, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 255]}, {'channel': 0, 'start_led': 3, 'num_leds': 1, 'mode': 2, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 255]}, {'channel': 0, 'start_led': 1, 'num_leds': 1, 'mode': 8, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 255]}, {'channel': 0, 'start_led': 9, 'num_leds': 1, 'mode': 8, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 255]}, {'channel': 0, 'start_led': 19, 'num_leds': 1, 'mode': 8, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 255]}, {'channel': 0, 'start_led': 19, 'num_leds': 1, 'mode': 8, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 255]}, {'channel': 0, 'start_led': 0, 'num_leds': 1, 'mode': 4, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 0]}, {'channel': 0, 'start_led': 0, 'num_leds': 1, 'mode': 4, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [0, 0, 0]}, {'channel': 0, 'start_led': 0, 'num_leds': 1, 'mode': 4, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [0, 0, 0]}] (from /var/run/liquidctl/vid1b1c_pid0c1a/loc2/saved_effects)
[... cut ...]
[DEBUG] liquidctl.driver.usb: read 16 bytes: 12:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 33:ff:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 16 bytes: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
</details>

jonasmalacofilho, 2021-01-17:
> The LNC doesn't report a status, but I think that's expected.

As it only has LED channels, I think this makes sense.  But maybe it can still report on its voltage rails, @MarshallAsch?

> Not sure why my DIMMS show up.

There's now support in liquidctl for reading temperatures from DDR4 modules that use standard EEPROM+sensor chips.  There's more information about this on the [DDR4 DIMMs guide](https://github.com/liquidctl/liquidctl/blob/master/docs/ddr4-guide.md).

> Doesn't work; it doesn't seem to like the --num-leds option for some reason, a

The option was renamed to `--maximum-leds`.  Apparently we forgot to update the docs, sorry about that.

>> Also note, that I could only change the modes a few times before it stopped responding.
> 
> Below is a debug log for a failed liquidctl command.
> [...] though this one was not the first failed command in the sequence [...]

I don't know what's going on there.  Can you remember which command (exactly) caused the device to stop responding?  And (with the cooler working) try it again with `--debug`?

> Is there a way to un-initialize liquidctl so that when the mode gets stuck I can give it a kick?

Not at the moment.

> I was surprised to see that the mode changes persisted across a reboot. Is there a way I can reset the fans back to their default rainbow mode when I'm not running liquidctl?

Have you tried the `clear` pseudo-mode?

---

P.S. Can you switch to the "master" branch and apply the patch to the _SUPPORTED_DEVICES variable there?  I don't expect this to magically change the behaviors you described, but it will certainly help avoid future confusion since there have been quite a few commits since the merge commit you're working on.

kevinlester, 2021-01-18:
> Can you switch to the "master" branch

Switched.

> The option was renamed to --maximum-leds

Cool, using that option works as expected.  I ran through all the lighting options and they all seemed to function correctly.

> And (with the cooler working) try it again with --debug

Below is a "clean" debug log after a reboot.  Liquidctl was initialized, then I ran through a few different modes until the LEDs no longer changed from my commands.  The log is from the first command that had no visible effect.  

<details>
<summary>Details</summary>

[DEBUG] liquidctl.cli: running liquidctl v1.4.2 (dc21ff10600e-dirty)
[DEBUG] liquidctl.driver.smbus: searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-3
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-10
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-1
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-8
[DEBUG] liquidctl.driver.smbus: ignoring 0-0037, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-6
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-13
[DEBUG] liquidctl.driver.smbus: ignoring 0-0053, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-4
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-11
[DEBUG] liquidctl.driver.smbus: ignoring 0-0051, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-2
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-0
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-9
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-7
[DEBUG] liquidctl.driver.smbus: ignoring 0-0036, cannot infer bus number
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-5
[DEBUG] liquidctl.driver.smbus: found I²C bus i2c-12
[DEBUG] liquidctl.driver.usb: searching HidapiBus (CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
[DEBUG] liquidctl.driver.usb: found HID device 0b05:18f3
[DEBUG] liquidctl.driver.usb: found HID device 275d:0ba6
[DEBUG] liquidctl.driver.usb: found HID device 275d:0ba6
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c1a
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair Lighting Node Core (experimental)
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c20
[DEBUG] liquidctl.driver.usb: found HID device 1b1c:0c10
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0220
[DEBUG] liquidctl.driver.usb: found HID device 05ac:0220
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver)
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0003
[DEBUG] liquidctl.driver.usb: found USB device 05ac:0220
[DEBUG] liquidctl.driver.usb: found USB device 05ac:1006
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c10
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c20
[DEBUG] liquidctl.driver.usb: found USB device 1b1c:0c1a
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0610
[DEBUG] liquidctl.driver.usb: found USB device 05e3:0610
[DEBUG] liquidctl.driver.usb: found USB device 275d:0ba6
[DEBUG] liquidctl.driver.usb: found USB device 0b05:18f3
[DEBUG] liquidctl.driver.usb: found USB device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair Lighting Node Core (experimental)
[DEBUG] liquidctl.keyval: data in /var/run/liquidctl/vid1b1c_pid0c1a/loc2
[DEBUG] liquidctl.util: clamped 96 to interval [1, 95]
[DEBUG] liquidctl.keyval: loaded saved_effects=[{'channel': 1, 'start_led': 0, 'num_leds': 95, 'mode': 0, 'speed': 1, 'direction': 1, 'random_colors': 1, 'colors': []}, {'channel': 0, 'start_led': 0, 'num_leds': 95, 'mode': 9, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 0]}, {'channel': 0, 'start_led': 0, 'num_leds': 95, 'mode': 10, 'speed': 1, 'direction': 1, 'random_colors': 1, 'colors': []}, {'channel': 0, 'start_led': 0, 'num_leds': 95, 'mode': 0, 'speed': 1, 'direction': 1, 'random_colors': 1, 'colors': []}, {'channel': 1, 'start_led': 0, 'num_leds': 95, 'mode': 0, 'speed': 1, 'direction': 1, 'random_colors': 1, 'colors': []}, {'channel': 0, 'start_led': 0, 'num_leds': 95, 'mode': 3, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 255, 0, 255, 0]}, {'channel': 0, 'start_led': 0, 'num_leds': 95, 'mode': 1, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 255, 0, 255, 0]}, {'channel': 0, 'start_led': 0, 'num_leds': 95, 'mode': 6, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 255, 0, 255, 0]}, {'channel': 0, 'start_led': 0, 'num_leds': 95, 'mode': 8, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 255, 0, 255, 0]}, {'channel': 0, 'start_led': 0, 'num_leds': 95, 'mode': 0, 'speed': 1, 'direction': 1, 'random_colors': 1, 'colors': []}] (from /var/run/liquidctl/vid1b1c_pid0c1a/loc2/saved_effects)
[DEBUG] liquidctl.keyval: stored saved_effects=[{'channel': 1, 'start_led': 0, 'num_leds': 95, 'mode': 0, 'speed': 1, 'direction': 1, 'random_colors': 1, 'colors': []}, {'channel': 0, 'start_led': 0, 'num_leds': 95, 'mode': 9, 'speed': 1, 'direction': 1, 'random_colors': 0, 'colors': [255, 0, 0]}, {'channel': 0, 'start_led': 0, 'num_leds': 95, 'mode': 10, 'speed': 1, 'direction': 1, 'random_colors': 1, 'colors': []}, {'channel': 0, 'start_led': 0, 'num_leds': 95, 'mode': 0, 'speed': 1, 'direction': 1, 'random_colors': 1, 'colors': []}, {'channel': 1, 'start_led': 0, 'num_leds': 95, 'mode': 0, 'speed': 1, 'direction': 1, 'random_colors': 1, [... cut ...]
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 33:ff:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 16 bytes: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00

</details>

> Have you tried the clear pseudo-mode?

The `clear` mode had no visible impact on the LEDs.  It did serve the purpose of enabling me to give liquidctl a "kick" when it stopped responding to commands, but has no other impact that I could tell.  Is this the expected functionality?

Cheers!

jonasmalacofilho, 2021-01-18:
> So I'm wondering if there is a 10 command or so limit for saved_effects, and after that number is reached future commands are ignored.

It appears to be just that, but the limit is in the device side.  Can you figure out by experimentation what that exact limit is on the LNC?

@MarshallAsch can you do the same on the Commander Pro?

> The QL120s have 34 LEDs per fan, for a grand total of 102 LEDs per channel.

That makes sense.

While we check whether it applies to the Commander Pro as well (I imagine it supports as many QL120s in a channel as the LNC), you can temporarily lift the 96-LED limit by changing these lines:

https://github.com/liquidctl/liquidctl/blob/dc21ff10600e97837638955dd924c0e4ac5b5448/liquidctl/driver/commander_pro.py#L447-L448

> I do not seem to be able to address the second channel of the LNC.

Looking at the code, that's rather odd...  Did you make sure your werent in the "too many saved effects" state?

Can you provide a traffic of using the second channel from iCue?

> The clear mode had no visible impact on the LEDs. It did serve the purpose of enabling me to give liquidctl a "kick" when it stopped responding to commands, but has no other impact that I could tell. *Is this the expected functionality?*

I'll have to delegate this one to @MarshallAsch : )

kevinlester, 2021-01-18:
> Can you figure out by experimentation what that exact limit is on the LNC?

The correct answer appears to be 8.  The 9th command consistently fails to make a change.

> you can temporarily lift the 96-LED limit by changing these lines:

Nice, my last 6 LEDs are now happily liquidctl-ing

> Did you make sure your werent in the "too many saved effects" state?

Yup, definitely not working on the second channel.

> Can you provide a traffic of using the second channel from iCue?

I'll have to set up a windows VM to capture the traffic.  Will report back once I get that set up.

MarshallAsch, 2021-01-20:
Hey sorry I didn't get back to this sooner. 

> the 96 led limit

I was looking at some comments on the corsair forums and I saw that number there that they impose on the number of LED's. It could have been an old value.  Maybe the limit is 136? which is 4 of those fans.

> As it only has LED channels, I think this makes sense. But maybe it can still report on its voltage rails

From my tests with @BangerTech on the `Lighting Node PRO` did not have support for getting the voltage. but if the core does that is simple to add.  @sam-brett does it have the voltage shown in icue?

> Can you figure out by experimentation what that exact limit is on the LNC?

I can check the limit for me on the comander pro. And add support to limit the number of commands that can be set

kevinlester, 2021-01-21:
Sorry for the delay.  I couldn't get iCue to detect my Corsair devices devices in VirtualBox, but got it working using virt-manager.  In iCue I now see my AIO, Commander Pro, and LNC.

> does it have the voltage shown in icue?

It does not show voltages in icue.

> Can you provide a traffic of using the second channel from iCue?

Here's traffic captured on my Linux host with iCue running in my windows VM.  I couldn't get the display filter to work as described in your development docs, but hopefully these contain what you need.  Order of operations were:
Disabled Front Lights
Enabled Front Lights
Disabled Rear Lights
Enabled Rear Lights
Switched effect to Color Wave
Switched effect to Visor
[LNC-cap.pcapng.gz](https://github.com/liquidctl/liquidctl/files/5846441/LNC-cap.pcapng.gz)

I'm not entirely sure how to correlate the channels as shown in iCue (front and back lights), vs the channels on the device (left side fans, right side fans).  If there is a better way for me to capture packets that would be helpful, please let me know.  I'm also curious as to why a display filter of `usb.device_address == 0xc1a` wasn't filtering as expected.  Any ideas on that?

kevinlester, 2021-01-23:
If it's helpful, here is a packet capture from iCue where I only changed the color of the 3 fans that I cannot control in liquidctl (specified using led2).  In the capture I had all the lights off, then turned all the lights on the back 3 fans to white, then all the lights on the back 3 fans to blue.
[channel2_LNC_capture.pcapng.zip](https://github.com/liquidctl/liquidctl/files/5860396/channel2_LNC_capture.pcapng.zip)

MarshallAsch, 2021-01-25:
@kevinlester is that capture using software mode or hardware mode? The implementation is only for hardware control at the moment because software control requires sending a message to the device every 20 seconds or so and `liquidctl` does not run as a service.

MarshallAsch, 2021-01-25:
Also looking at the device again there is one lighting channel not 2. My bad about that. 

Aside from that, you have 6 QL120s fans so the first 34 LED's will be for the first fan, the next 34 will be for fan 2, etc...

Can you give it another shot on the latest version of the [liquidctl/154-node-core](/liquidctl/154-node-core) branch to see if that fixed the rest of your issues @kevinlester?

kevinlester, 2021-01-25:
>is that capture using software mode or hardware mode

Software mode.  I'll grab a capture changing all the fans in hardware mode in the morning.  I just did a quick pull of that branch and don't see an immediate difference at the moment.  If I understood your last comment correctly, the following command should turn my first 4 fans to rainbow, correct?  So far only my first 3 are responding.
```
$ git status
On branch 154-node-core
Your branch is up to date with 'origin/154-node-core'.
$ liquidctl --match 'Lighting Node Core' set led1 color rainbow --start-led 1 --maximum-leds 136
```
But all the LEDs on my first 3 are responding (no longer capped at 96 LEDs), so that's an improvement.  Let me know what else I should test and I'll try it out tomorrow.  
Cheers!

kevinlester, 2021-01-26:
Success!  The latest version of the 154-node-core branch works perfectly based on my testing.  I am able to address all 204 LEDs and am prevented from issuing commands that have no response from the LNC.  I have a feeling I forgot to `pip install .` last night.  In any case, after correctly using the branch all seems to be working as expected.  Huzzah!

MarshallAsch, 2021-01-26:
Excellent! glad it worked. It will be included in release 1.6.0.

closed 2021-01-26 by jonasmalacofilho

kevinlester, 2021-02-06:
@MarshallAsch - it took a bit of tinkering to capture the traffic between a guest VM and my device, so I took some time and expanded the developer docs to (hopefully) make it easier for other people to get up and running when they would like to capture traffic.  I have a local branch created with my changes, but don't have write access so can't submit a PR.  May I please have temporary write access to submit the PR, or is there another way you would prefer I submit my changes for review?  Cheers!

MarshallAsch, 2021-02-06:
@kevinlester You can make a fork of the repository (there should be a fork button at the top of the github page) and then make a PR from your version onto this repository. And then any data captures that you have can be submitted to the [liquidctl/collected-device-data](https://github.com/liquidctl/collected-device-data) repo.

kevinlester, 2021-02-06:
👍 [PR](https://github.com/liquidctl/liquidctl/pull/283)

PR #266 (2021-01-26): Lighting Node Core Support
Closes #154.

This enables support for the Corsair Lighting Node Core.

- [x] update man pages with new device
- [x] update readme with new device
- [x] added device id to the commander pro driver
- [x] added device usage to the commander pro driver docs
- [x] create tests for the new device
- [x] create tests for the new limited number of effect commands
- [x] figure out why led channel 2 does not seem to be working