liquidctl/liquidctl #155: Corsair Commander Pro support

hgrahamcs, 2020-07-20:
Hello, I was looking to attempt to port the commander pro code from [audiohacked/OpenCorsairLink ](https://github.com/audiohacked/OpenCorsairLink) and would like to know where I should start. So far I have been attempting to understand how the driver works in OpenCorsairLink but haven't had much luck yet. When I do eventually figure it out what steps would I have to take to make a new driver for it in this project. 

Thanks.

jonasmalacofilho, 2020-07-20:
Hi Henry,

Great to have you here!

I wrote [this longer than usual comment](https://github.com/jonasmalacofilho/liquidctl/issues/129#issuecomment-640258429) about porting drivers from OCL.  Even though it's for a different device, the general steps are the same and it should get you started.  Let me know how it goes and if you have any questions...

Also, if you decide you need to do some additional reverse engineering of the protocol, [this other comment](https://github.com/jonasmalacofilho/liquidctl/issues/142#issuecomment-650568291) may also help.

hgrahamcs, 2020-07-20:
Thanks for the quick response, 

Those both look like great resources. Thanks for the help.

jonasmalacofilho, 2020-08-04:
Note: the Corsair Lighting Node Core (issue #154) appears to share the same protocol.

hol430, 2020-08-19:
Hi all, just wondering if there has been any work done on this issue? I'm interested in helping out with this, but don't want to duplicate any existing efforts. I have my own quick and dirty (certainly not fully-featured) driver which was based on some of the work in open corsair link, but figure it will be helpful to more people if it lives in this library.

hol430, 2020-08-21:
Ok, so I have made a start on this and have a question about how we want `set_fixed_speed()` to work. The commander pro supports setting speed as fixed RPM or as a percentage. I assume we want to encapsulate both of these inside `set_fixed_speed()`, and if so how would we ideally want to expose this to the CLI? Would we want something like `liquidctl --match "Commander pro" set <channel> speed 1250 --rpm`?

I'm still not super familiar with the rest of the project infrastructure so apologies if I've missed a common pattern here.

hol430, 2020-08-23:
Ah ok, thanks for the detailed explanation, it's good info for me as I'm not really much of a python programmer. Another option which I also don't really like might be to automatically interpret the speed as percent if it's between 0..100, or RPM if it's >100. It doesn't seem very intuitive though and it also doesn't address the issue of the parameter being named `duty`, which doesn't really describe what it does in this instance.

Out of curiosity, how do the other devices work for set_fixed_speed - ie do they operate on a percentage basis? If so, maybe set_fixed_speed could always assume a percentage and we add a new command for rpm control - e.g. `liquidctl -mcommander set fan2 speed_rpm 1250`?

For now I just have `set_fixed_speed_percent()` and `set_fixed_speed_rpm()`, I'll have a bit more of a think about this as I work on some of the other features...

jonasmalacofilho, 2020-08-28:
> Out of curiosity, how do the other devices work for set_fixed_speed - ie do they operate on a percentage basis?

Yeah, they operate solely on duty/percentage.

> [...]
> For now I just have set_fixed_speed_percent() and set_fixed_speed_rpm(), I'll have a bit more of a think about this as I work on some of the other features...

The CLI would work well with either `set fan2 speed 500 --rpm` (flag) or `set fan2 rpm speed 500` (command).  But let's implement `set_fixed_speed_rpm` without actually connect it to the CLI just yet.  For now I would like to only have the percent/duty mode functional (in `set_fixed_speed`, for uniformity with the other drivers).

While a second `_rpm` API is our best solution so far, let's give ourselves a bit more.  The important thing is that both APIs work and have tests; this way exposing them to the CLI or even renaming/refactoring them later will be trivial changes.

Thanks again, btw!

hol430, 2020-08-30:
Ah ok, thanks for that. Yeah, leaving RPM control disconnected from CLI seems like the best bet for now. As a bit of a progress update, I've implemented most of the features in OpenCorsairLink. There are a few commands which still require some additional research, but other than that it seems to be working well. The biggest problem at the moment is that there are a few features which aren't accessible via the CLI, particularly in the LED control, which is a rather complicated feature.

Basically, there are a few options, such as alternating vs random colours, animation speed, colour direction, etc, which I have hardcoded for now. Do we want to make these accessible via the CLI? If so, it would probably mean adding 3-4 extra command line options. Let me know if that's ok, if it is I'll make it happen.

If anyone else has a commander pro and wants to test it, the code is [here](https://github.com/hol430/liquidctl/tree/commanderpro).

MarshallAsch, 2020-11-08:
@hol430 is there any update with your commander pro support. I got a commander pro the other day and didn't realize that you had all ready implemented a port. Is there anything I can do for you or @jonasmalacofilho to help get this into the main project. (I personally don't want to keep track of multiple versions of a project on my system to control multiple devices)

hol430, 2020-11-08:
Hi @MarshallAsch, thanks for bumping this, I had actually forgotten about this project. Yes, it's something I'd like to get back to and finish off this commander pro driver. I'll double check this over the next few days, but I'm pretty sure I've ported and tested all of the features from OpenCorsairLink. The problem is that most of the features are not connected to the CLI at this stage, as they're (as I understand it) specific to the commander pro - such as the example of fixed speed control as discussed above.

I'm not sure how we want to approach this - some of the functionality is pretty complicated and might be a bit unwieldy to use via the cli, such as the 'graph' mode for fan speed which basically sets fan speed based on a linear interp of current temperature. Something like this might be easier to use via a simple python script.

On a side note - is anyone aware of any guis for this project? I have a simple one which I often use for my old ad-hoc version of the driver and I find it a convenient way of controlling the device. If none exist for liquidctl, I might throw something together even if it's just something simple for myself for controlling the commander pro.

MarshallAsch, 2020-11-08:
I know of [this](https://gitlab.com/CalcProgrammer1/OpenRGB/) project has some support for this device listed however I cant run it to actually check on my computer. 

> fixed speed control being a problem

The protocol seems to support setting a fixed duty percentage instead of a fixed rpm. (I may have created some of the functionality for this same issue before I noticed that you linked to your implementation, and submitted a PR with it because why not)

PR #213 (2020-12-31): Added initial support for the corsair commander pro
Closes: #155 

I just noticed now that @hol430 has already started an implementation [here](https://github.com/hol430/liquidctl/tree/commanderpro). oops. I don't really want to duplicate all the work that you did, but a PR was never submitted with the change so. You seemed to have some modes and settings in your implementation that I dont have or that I don't quite have a plan for. Another one of the major differences between my implementation and yours is that I think how I implemented it is more simular to the existing project code structur [... cut]
closed 2020-12-31 by MarshallAsch