liquidctl/liquidctl #156: Corsair H100i Pro XT model CW-9060033-WW not found

douginoz, 2020-07-23:
Corsair H100i Pro XT (model CW-9060033-WW)
idVendor 0x1b1 Corsair
idProduct 0x0c15
iProduct H100i Platinum
Ubuntu 20.04
Asus x299 Prime II motherboard
H100i usb plugged into motherboard USB port
```

root@sophie:~# lsusb -vt
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 6: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 6: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
        |__ Port 4: Dev 6, If 0, Class=Vendor Specific Class, Driver=, 12M
            ID 1b1c:0c15 Corsair 
        |__ Port 3: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 051d:0002 American Power Conversion Uninterruptible Power Supply
    |__ Port 8: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        ID 046d:c52b Logitech, Inc. Unifying Receiver
    |__ Port 8: Dev 3, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        ID 046d:c52b Logitech, Inc. Unifying Receiver
    |__ Port 8: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 046d:c52b Logitech, Inc. Unifying Receiver
    |__ Port 12: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
    |__ Port 12: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
    |__ Port 13: Dev 7, If 1, Class=Wireless, Driver=btusb, 12M
        ID 8087:0025 Intel Corp. 
    |__ Port 13: Dev 7, If 0, Class=Wireless, Driver=btusb, 12M
        ID 8087:0025 Intel Corp. 
    |__ Port 14: Dev 8, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 0b05:18b8 ASUSTek Computer, Inc. 

# liquidctl list --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hidraw, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1c4f:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1c4f:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xb05:0x18b8
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x51d:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0x1024
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0x2011
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0x4002
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x174c:0x3074
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0xc15
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x51d:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x174c:0x2074
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xb05:0x18b8
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x8087:0x25
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1c4f:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
```

jonasmalacofilho, 2020-07-23:
I see two _combining_ causes.

First, `Driver=` tells me that `usbhid` has probably been unloaded from this device, possibly by another software that uses libusb (or hidapi-libusb) and doesn't reattach the kernel driver at the end.  Check the troubleshooting entry for how to [manually rebind `usbhid`](https://github.com/jonasmalacofilho/liquidctl#device-not-listed-linux).

Second, you have the last stable version of liquidctl, which doesn't yet include the support for the H100i PRO XT.  Take a look at the readme and try to install the latest code from the main branch.

douginoz, 2020-07-23:
I'm trying to manually add the <bus>-<port> but I get 'No such device'.  What's the value I need to use?  I've tried:
echo '01-4:1.0' | sudo tee /sys/bus/usb/drivers/usbhid/bind
echo '01-04:1.0' | sudo tee /sys/bus/usb/drivers/usbhid/bind
echo '1-4:1.0' | sudo tee /sys/bus/usb/drivers/usbhid/bind

I did some research and a partial extract of $ tree's output is:
 -- usb
----- 1-6.3 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.3
----- 1-6.4 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.4
-- usbhid
----- 1-6.3:1.0 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.3/1-6.3:1.0

So it seems logical to me that I need to add the Corsair (Bus 01.Port 6.Port 4) as
'1-6.4:1.0' or perhaps 
'1-4.6:1.0'
but again, nothing works:

```
$ echo '1-4.6:1.0' | sudo tee /sys/bus/usb/drivers/usbhid/bind
1-4.6:1.0
tee: /sys/bus/usb/drivers/usbhid/bind: No such device
# echo -n "1-6.4:1.0" | sudo tee /sys/bus/usb/drivers/usbhid/bind
1-6.4:1.0tee: /sys/bus/usb/drivers/usbhid/bind: No such device

Thinking that this error might be because of my shell or the syntax, I also tried:
# sudo sh -c 'echo -n "1-6.4:1.0" > /sys/bus/usb/drivers/usbhid/bind'
sh: 1: echo: echo: I/O error

```
I've been assuming that it's 01-04 because it's the one without any value for 'Driver=":

/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 6: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
        |__ Port 4: Dev 6, If 0, Class=Vendor Specific Class, Driver=, 12M
            ID 1b1c:0c15 Corsair 
```

douginoz, 2020-07-23:
Here's the folder:
```
ll /sys/bus/usb/drivers/usbhid/
total 0
drwxr-xr-x 2 root root    0 Jul 23 11:25 ./
drwxr-xr-x 7 root root    0 Jul 23 11:25 ../
lrwxrwxrwx 1 root root    0 Jul 23 14:54 1-12:1.0 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-12/1-12:1.0/
lrwxrwxrwx 1 root root    0 Jul 23 14:54 1-12:1.1 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-12/1-12:1.1/
lrwxrwxrwx 1 root root    0 Jul 23 14:54 1-14:1.0 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/
lrwxrwxrwx 1 root root    0 Jul 23 14:54 1-6.3:1.0 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.3/1-6.3:1.0/
lrwxrwxrwx 1 root root    0 Jul 23 14:54 1-8:1.0 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/
lrwxrwxrwx 1 root root    0 Jul 23 14:54 1-8:1.1 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/
lrwxrwxrwx 1 root root    0 Jul 23 14:54 1-8:1.2 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/
--w------- 1 root root 4096 Jul 23 15:28 bind
lrwxrwxrwx 1 root root    0 Jul 23 14:54 module -> ../../../../module/usbhid/
-rw-r--r-- 1 root root 4096 Jul 23 14:54 new_id
-rw-r--r-- 1 root root 4096 Jul 23 14:54 remove_id
--w------- 1 root root 4096 Jul 23 11:25 uevent
--w------- 1 root root 4096 Jul 23 14:54 unbind
```

jonasmalacofilho, 2020-07-23:
I would expected it be:

```
$ echo '1-4:1.0' | sudo tee /sys/bus/usb/drivers/usbhid/bind
```

The device would only appear in `/sys/bus/usb/drivers/usbhid/` after it was bound that that driver, before that it would have only been listed in `/sys/bus/usb/devices/`.

---

That said, I just noticed the `Class=Vendor Specific Class` part.

Your device is neither a "H100i PRO XT", as you believe, nor a "H100i Platinum", as its USB descriptor states.

It's a H100i PRO (non XT), made for Corsair by a different company (Asetek instead of CoolIT), requiring a different driver and, more importantly, not yet supported by liquidctl.

You can follow the progress in #137, and maybe even help out in porting the driver from OCL.

P.S. Tthis is probably the fourth time that this PRO vs. PRO XT vs. Platinum bits me...

closed 2020-07-23 by jonasmalacofilho

douginoz, 2020-07-23:
In the words of Cartman, "goddammit"!  Sorry about that.  I don't know where I got the "XT" from.

jonasmalacofilho, 2020-07-23:
Don't worry about, I should have noticed it too... and Corsair could be a little less chaotic in its naming.