liquidctl/liquidctl #157: Fails to run on Linux Debian

jabra98, 2020-07-24:
Hi, I had to reinstall the project and it fails to run now (both the PyPI version and the repo). Here is the error message when trying to run, seems to be something with cpython:
```
Traceback (most recent call last):
  File "/usr/local/bin/liquidctl", line 5, in <module>
    from liquidctl.cli import main
  File "/usr/local/lib/python3.8/dist-packages/liquidctl/cli.py", line 75, in <module>
    from liquidctl.driver import *
  File "/usr/local/lib/python3.8/dist-packages/liquidctl/driver/__init__.py", line 21, in <module>
    import liquidctl.driver.asetek
  File "/usr/local/lib/python3.8/dist-packages/liquidctl/driver/asetek.py", line 34, in <module>
    from liquidctl.driver.usb import UsbDriver
  File "/usr/local/lib/python3.8/dist-packages/liquidctl/driver/usb.py", line 63, in <module>
    import hidraw as hid
ImportError: /usr/lib/python3/dist-packages/hidraw.cpython-38-x86_64-linux-gnu.so: undefined symbol: hid_get_input_report
```
The listed dependencies are installed of course.

jonasmalacofilho, 2020-07-24:
This is caused by trezor/cython-hidapi#90: `0.9.0.post3` references an unreleased API from `libhidapi`.

I see that Debian (unstable) has updated `python3-hid` to this problematic version; you should open a bug report on Debian, they are going to need to patch it. : \

---

On to how to deal with this here.  I have refrained from pinning the cython-hidapi version because that would cause compatibility issues on distros that have already updated to `0.9.0.post3`.

You can sidestep the problem by using a Python virtual environment and pre-installing `0.9.0.post2` from PyPI within it, or by downgrading `python3-hid` (assuming you're using it instead of the corresponding `hidapi` PyPI package) globally.

jonasmalacofilho, 2020-07-24:
I just updated the previous comment with clearer instructions for the workarounds.

---

When opening the issue you can adjust and reuse the following minimal examples:

```
$ python3 -c "import hidraw"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: /usr/lib/python3/dist-packages/hidraw.cpython-38-x86_64-linux-gnu.so: undefined symbol: hid_get_input_report
```

```
$ python3 -c "import hid"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: /usr/lib/python3/dist-packages/hid.cpython-38-x86_64-linux-gnu.so: undefined symbol: hid_get_input_report
```

Additionally, trying to run the [upstream supplied test file](https://github.com/trezor/cython-hidapi/blob/master/tests.py) should result in a similar error.

jabra98, 2020-07-24:
Alright, I submitted a bug report to the Debian package maintainers. Will use the workaround in the meantime.

jonasmalacofilho, 2020-07-24:
Thanks, @jabra98!

---

For others arriving here...

Debian bug: [#966195: python3-hid: references an unreleased API from libhidapi, resulting in undefined symbols](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=966195)

jabra98, 2020-07-24:
The fix was just pushed to the Debian repositories. I think this can be closed now.

jonasmalacofilho, 2020-07-24:
It's still propagating through all archs, but `amd64` is already live.

I don't expect any problems at all, but please let me know if you're able to install the new package and successfully use liquidctl (with it).

jabra98, 2020-07-24:
Yeah I installed it already, everything works now.

jonasmalacofilho, 2020-07-24:
Thanks again!

closed 2020-07-24 by jonasmalacofilho

jonasmalacofilho, 2020-07-24:
@jabra98, any chance I can get you to adopt #62 ("Package for Debian/Ubuntu")?

It would already help if you could file the Request For Package on Debian.  But it would be especially awesome if you could take over the actual packaging (in which case it would be more appropriate to file an Intent to Package).

jabra98, 2020-07-24:
I have never done any packaging for Debian before. Have no clue how it works :/

jonasmalacofilho, 2020-07-26:
It really doesn't seem to be the simplest thing: each step is not very complicated, but there are several of them.

But, thanks any way!