liquidctl/liquidctl #158: Support for HydroX series

izolyda, 2020-07-26:
Can you please add Linux support for HydroX series (XD5 pump, XC9 CPU block, XG7 GPU block)? 
Please disregard if not applicable.

Thank you.

jonasmalacofilho, 2020-07-26:
I'm taking a look at Corsair's website.  It seems that those require a Commander PRO or Lighting Node PRO, is that right?

jonasmalacofilho, 2020-07-27:
I've been able to (reasonably) confirm that a Commander/Lighting Node PRO/Core is necessary, at least in the Corsair ecosystem.  You can follow the progress on the development of those drivers in #154 and #155.

By the way, you can also (probably) control them with an Arduino or Raspberry PI (by making either a RGB controller of your own).

closed 2020-07-27 by jonasmalacofilho