liquidctl/liquidctl #159: NZXT x53 OSError on ubuntu 20.04

OkuChou, 2020-07-26:
I'm trying liquidctl to control my newly bought x53. But when i  implement command "python3 -m liquidctl.cli status --debug" got followe error message.
[DEBUG] __main__: running liquidctl v1.3.3
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RGBFusion2Driver, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 5702 Controller (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 056e:1050
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:406a
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:0411
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:5411
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0026
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] __main__: device: NZXT Kraken X (X53, X63 or X73) (experimental)
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/xxxx/Download/liquidctl/liquidctl/cli.py", line 327, in <module>
    main()
  File "/home/xxxx/Download/liquidctl/liquidctl/cli.py", line 297, in main
    dev.connect(**opts)
  File "/home/xxxx/Download/liquidctl/liquidctl/driver/usb.py", line 115, in connect
    self.device.open()
  File "/home/xxxx/Download/liquidctl/liquidctl/driver/usb.py", line 363, in open
    self.hiddev.open_path(self.hidinfo['path'])
  File "hidraw.pyx", line 72, in hidraw.device.open_path
OSError: open failed

I've tried command "sudo apt install python3 libusb-1.0-0 python3-pkg-resources python3-docopt python3-usb python3-hid", but it can't help.
My motherboard is gigabyt z490 vision -d.
Can anyone help me. Thanks.

OkuChou, 2020-07-26:
python3 -m liquidctl.cli initialize all
OSError: open failed

jonasmalacofilho, 2020-07-26:
This is probably a permission error.  Take a look at [Access denied or open failed (Linux)](https://github.com/jonasmalacofilho/liquidctl#access-denied-or-open-failed-linux) in the troubleshooting section of the readme.

OkuChou, 2020-07-26:
@jonasmalacofilho Problem solved. Yes, you are right. Thank you so much!

closed 2020-07-26 by jonasmalacofilho