liquidctl/liquidctl #160: Incorrect temperature reading from a Kraken X3 cooler

jonasmalacofilho, 2020-07-26:
_Originally posted by @msigut in https://github.com/jonasmalacofilho/liquidctl/issues/79#issuecomment-663978449_

Hi,

please look at this message: **280.5 °C** . I hope that temperature is incorrect ... 
Latest [HEAD] commit and also some from history, always the same ...

Please, **help me with this** ... thank you.

```
NZXT Kraken X (X53, X63 or X73) (experimental)
├── Liquid temperature    280.5  °C
├── Pump speed             2739  rpm
└── Pump duty               100  %
```

debug info:
```
sudo liquidctl status --debug
[DEBUG] liquidctl.cli: running liquidctl v1.4.0a0 (0d2cdad0f223)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, KrakenX3Driver, KrakenZ3Driver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1781:0e39
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0ffc:0026
[DEBUG] liquidctl.driver.usb: probing drivers for device 258f:0b04
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2006
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 043e:9a11
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:0411
[DEBUG] liquidctl.driver.usb: probing drivers for device 043e:9a10
[DEBUG] liquidctl.driver.usb: probing drivers for device 1781:0e39
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:5411
[DEBUG] liquidctl.cli: device: NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.kraken_gen4: received 75 02 20 00 36 00 0b 51 53 58 34 35 33 20 01 ff ff cc 0a 64 64 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
NZXT Kraken X (X53, X63 or X73) (experimental)
├── Liquid temperature    280.5  °C
├── Pump speed             2764  rpm
└── Pump duty               100  %

[DEBUG] liquidctl.cli: device: NZXT Smart Device V2
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 02 2f 00 39 00 07 51 56 56 39 30 30 20 03 00 02 02 02 00 00 00 00 00 d4 02 11 02 c7 02 00 00 00 00 00 00 00 00 00 00 32 32 32 00 00 00 00 00 32 32 32 00 00 00 00 00 30 00 00 00 00 00 00 00
NZXT Smart Device V2
├── Fan 1 duty      50  %
├── Fan 1 speed    724  rpm
├── Fan 2 duty      50  %
├── Fan 2 speed    529  rpm
├── Fan 3 duty      50  %
├── Fan 3 speed    711  rpm
└── Noise level     48  dB
```

jonasmalacofilho, 2020-07-26:
@msigut,

I moved this into a separate issue.  But it's a bug alright.

Can you tell me a few things?

 - did you `initialize` the Kraken before executing `status`?
 - what firmware version it is currently running (it's reported by `initialize`)?
 - did you observe this issue before?  did you ever get sensible values for the liquid temperature?
 - what's your ambient temperature?
 - based on your experience (possibly on Windows/with CAM), what would be the expected temperature value in this scenario (it's obviously not 280.5°C, but it helps if I know what I'm looking for)?

msigut, 2020-07-26:
OK, what I did:

- get latest HEAD of Master (da3fc627618e8b0b69d5dbe3200c2c9ab079987f)
- install by: `sudo python3 setup.py install`
- initialize Kraken by: `liquidctl --match kraken initialize --debug` (detail list bellow)
- regarding question: initialize before status: yes
- regarding question: firmware version is: `1.8.0`
- regarding question: observe issue before: I was using some older commit that works, truly I can't remember point in time ...
- regarding question: ambient temperature is like 25-27 'C
- regarding question: CAM experience: sorry I can't run Windows with CAM at that machine

```
liquidctl --match kraken initialize --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (da3fc627618e)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RGBFusion2Driver, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1781:0e39
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2006
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2006
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 043e:9a10
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0ffc:0026
[DEBUG] liquidctl.driver.usb: probing drivers for device 1781:0e39
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 043e:9a11
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:0411
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:5411
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 258f:0b04
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.cli: device: NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x10 with 63 bytes: 01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x20 with 63 bytes: 03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x70 with 63 bytes: 02:01:b8:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x70 with 63 bytes: 01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 11:01:20:00:36:00:0b:51:53:58:34:35:33:20:07:20:02:01:08:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 71:01:20:00:36:00:0b:51:53:58:34:35:33:20:01:00:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 75:02:20:00:36:00:0b:51:53:58:34:35:33:20:01:ff:ff:e6:0a:64:64:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 21:03:20:00:36:00:0b:51:53:58:34:35:33:20:03:00:00:00:00:00:00:10:00:00:00:00:00:11:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
NZXT Kraken X (X53, X63 or X73) (experimental)
├── Firmware version       1.8.0
[... cut ...]
[DEBUG] liquidctl.driver.usb: read 64 bytes: 67:02:2f:00:39:00:07:51:56:56:39:30:30:20:03:00:02:02:02:00:00:00:00:00:d5:02:17:02:d0:02:00:00:00:00:00:00:00:00:00:00:32:32:32:00:00:00:00:00:32:32:32:00:00:00:00:00:30:00:00:00:00:00:00:00
NZXT Smart Device V2
├── Fan 1 duty      50  %
├── Fan 1 speed    725  rpm
├── Fan 2 duty      50  %
├── Fan 2 speed    535  rpm
├── Fan 3 duty      50  %
├── Fan 3 speed    720  rpm
└── Noise level     48  dB

[DEBUG] liquidctl.cli: device: Gigabyte RGB Fusion 2.0 5702 Controller (experimental)
[INFO] liquidctl.driver.rgb_fusion2: Status reports not available from Gigabyte RGB Fusion 2.0 5702 Controller (experimental)
```

jonasmalacofilho, 2020-07-26:
@pelifix,

Sorry to pull you in here, but in the past you reported another issue with the temperature value and ran a few comparisons against HWiNFO.

Could you do that again with the code currently in the main branch (commit da3fc627618e), and compare not only against HWiNFO but also against CAM?

As usual, please include the liquidctl output with `--debug`.

msigut, 2020-07-26:
After firmware update to `2.1.0` (by CAM, virtual Windows installation) it's OK: `Liquid temperature    33.1  °C`
```
NZXT Kraken X (X53, X63 or X73) (experimental)
├── Firmware version       2.1.0
├── Pump Logo LEDs      detected
└── Pump Ring LEDs      detected
```

```
sudo liquidctl status --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (da3fc627618e)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RGBFusion2Driver, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 05ac:1392
[DEBUG] liquidctl.driver.usb: probing drivers for device 1781:0e39
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 5702 Controller (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2006
[DEBUG] liquidctl.driver.usb: probing drivers for device 05ac:1392
[DEBUG] liquidctl.driver.usb: probing drivers for device 043e:9a10
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 1781:0e39
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 043e:9a11
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:0411
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:5411
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 258f:0b04
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:5702
[DEBUG] liquidctl.cli: device: NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: read 64 bytes: 75:02:20:00:36:00:0b:51:53:58:34:35:33:20:01:21:01:a8:06:35:35:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
NZXT Kraken X (X53, X63 or X73) (experimental)
├── Liquid temperature    33.1  °C
├── Pump speed            1704  rpm
└── Pump duty               53  %

[DEBUG] liquidctl.cli: device: NZXT Smart Device V2
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: read 64 bytes: 67:02:2f:00:39:00:07:51:56:56:39:30:30:20:03:00:02:02:02:00:00:00:00:00:e6:02:1d:02:cb:02:00:00:00:00:00:00:00:00:00:00:32:32:32:00:00:00:00:00:32:32:32:00:00:00:00:00:30:00:00:00:00:00:00:00
NZXT Smart Device V2
├── Fan 1 duty      50  %
├── Fan 1 speed    742  rpm
├── Fan 2 duty      50  %
├── Fan 2 speed    541  rpm
├── Fan 3 duty      50  %
├── Fan 3 speed    715  rpm
└── Noise level     48  dB
```

jonasmalacofilho, 2020-07-26:
I'm glad to hear.  Just to double check, do liquidctl and CAM agree on the specific temperature value?

msigut, 2020-07-27:
Yes , it's correct.
Something change in code that some-time before was firmware 1.8.0 correct and now you needs 2.1.0 ?

jonasmalacofilho, 2020-07-27:
No, there were no code changes on our side, it seems that your device was simply behaving badly.

I also noticed that it was running the pump at full speed, which suggests the controller itself was also "seeing" the same 280.5 °C value.

closed 2020-07-27 by jonasmalacofilho