liquidctl/liquidctl #162: H115i PRO XT doesn't connect on Linux platforms

hosaka-corp, 2020-07-28:
On my Arch Linux box, the introduction of 4028c26e seems to prevent me from using `initialize`.

```
[DEBUG] __main__: device: Corsair H115i PRO XT (experimental)
self.address: "/dev/hidraw1"
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/meta/src/liquidctl/liquidctl/cli.py", line 327, in <module>
    main()
  File "/home/meta/src/liquidctl/liquidctl/cli.py", line 297, in main
    dev.connect(**opts)
  File "/home/meta/src/liquidctl/liquidctl/driver/hydro_platinum.py", line 158, in connect
    loc = 'loc' + '_'.join((num.decode() for num in re.findall(b'\\d+', self.address)))
  File "/usr/lib/python3.8/re.py", line 241, in findall
    return _compile(pattern, flags).findall(string)
TypeError: cannot use a bytes pattern on a string-like object
```

jonasmalacofilho, 2020-07-28:
Thanks for reporting this.  Can you confirm that it's fixed in the latest HEAD (d8f74b01d4f7)?

hosaka-corp, 2020-07-28:
No problem! Yep, no issue on d8f74b0 :ok_hand:

closed 2020-07-28 by jonasmalacofilho