liquidctl/liquidctl #163: X52 Fan Speed Control

zwu513, 2020-08-01:
In 1.4.0, fan speed control on the X52 does not work. Fans are running at 100% even when it says that my fan curve is a valid command. See the attached image. 

[image]

jonasmalacofilho, 2020-08-01:
Hi,

I have a X62 with the same firmware version, but cannot reproduce this incorrect behavior.

<details>

```
$ liquidctl list
Device ID 0: NZXT Smart Device (V1)
Device ID 1: NZXT Kraken X (X42, X52, X62 or X72)

$ liquidctl -m kraken initialize

$ liquidctl -m kraken set pump speed 100

$ liquidctl -m kraken set fan speed 30 35 35 45 40 70 45 90 50 100 --verbose
INFO: setting fan PWM duty to 35% for liquid temperature >= 20°C
INFO: setting fan PWM duty to 35% for liquid temperature >= 21°C
INFO: setting fan PWM duty to 35% for liquid temperature >= 22°C
[...]

$ liquidctl -m kraken set sync color off

$ liquidctl -m kraken status
NZXT Kraken X (X42, X52, X62 or X72)
├── Liquid temperature     27.9  °C
├── Fan speed               477  rpm
├── Pump speed             2814  rpm
└── Firmware version      6.0.2  
```
</details>

Can you clarify whether this only started to happen after you updated to 1.4.0?

Des the following produce the expected results?

```
$ liquidctl -m kraken set pump speed 50
$ liquidctl -m kraken set pump speed 75
$ liquidctl -m kraken set fan speed 50
$ liquidctl -m kraken set fan speed 75
```

And does reverting to 1.3.3 fix the problem?

Can you also provide me with the complete output of the following command?

```
$ liquidctl -m kraken set fan speed 30 35 35 45 40 70 45 90 50 100 --debug
```

And, finally, have there been any other software or hardware changes that could also explain this behavior?  On the software side, another program could be controlling the Kraken dynamically and overruling the settings uploaded by liquidctl.  On the hardware side, you should try swapping the fan that's connected to the first position in the chain: if there's a problem on that connection, or with that specific fan, that fan will run at maximum speed and that's the only value actually read by the Kraken.

zwu513, 2020-08-01:
This only started after 1.4.0. Setting pump speed works. Setting fan speed does not. 1.3.3 does not fix this issue. Pump speed works but not fan control on 1.3.3. Might be a hardware issue on my end.

Raw output from CMD. 

> C:\Users\USER>liquidctl -m kraken set fan speed 30 35 35 45 40 70 45 90 50 100 --debug
> [DEBUG] __main__: running liquidctl v1.4.0 (AppVeyor; liquidctl-1.4.0-bin-windows-x86_64)
> [DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
> [DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b55
> [DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
> [DEBUG] liquidctl.driver.usb: probing drivers for device 258a:0036
> [DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
> [DEBUG] liquidctl.driver.usb: probing drivers for device 258a:0036
> [DEBUG] liquidctl.driver.usb: probing drivers for device 258a:0036
> [DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
> [DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
> [DEBUG] liquidctl.driver.usb: probing drivers for device 258a:0036
> [DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b55
> [DEBUG] liquidctl.driver.usb: probing drivers for device 258a:0036
> [DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b55
> [DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b55
> [DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b55
> [DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b55
> [DEBUG] liquidctl.driver.usb: probing drivers for device 258a:0036
> [DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
> [DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1b55
> [DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0610
> [DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
> [DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
> [DEBUG] liquidctl.driver.usb: probing drivers for device 258a:0036
> [DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
> [DEBUG] liquidctl.driver.usb: probing drivers for device 0781:5597
> [DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0616
> [DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
> [DEBUG] liquidctl.driver.usb: probing drivers for device 10de:1ad6
> [DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
> [DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
> [DEBUG] __main__: device: NZXT Kraken X (X42, X52, X62 or X72)
> [DEBUG] liquidctl.driver.usb: read 64 bytes: 04:2a:01:08:7f:0a:d1:00:00:00:a8:06:00:00:02:1e:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [INFO] liquidctl.driver.kraken2: setting fan PWM duty to 35% for liquid temperature >= 20°C
> [DEBUG] liquidctl.driver.usb: writting report 0x02 with 64 bytes: 4d:80:14:23:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [INFO] liquidctl.driver.kraken2: setting fan PWM duty to 35% for liquid temperature >= 21°C
> [DEBUG] liquidctl.driver.usb: writting report 0x02 with 64 bytes: 4d:81:15:23:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [INFO] liquidctl.driver.kraken2: setting fan PWM duty to 35% for liquid temperature >= 22°C
> [DEBUG] liquidctl.driver.usb: writting report 0x02 with 64 bytes: 4d:82:16:23:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [INFO] liquidctl.driver.kraken2: setting fan PWM duty to 35% for liquid temperature >= 23°C
> [DEBUG] liquidctl.driver.usb: writting report 0x02 with 64 bytes: 4d:83:17:23:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [INFO] liquidctl.driver.kraken2: setting fan PWM duty to 35% for liquid temperature >= 24°C
> [DEBUG] liquidctl.driver.usb: writting report 0x02 with 64 bytes: 4d:84:18:23:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [INFO] liquidctl.driver.kraken2: setting fan PWM duty to 35% for liquid temperature >= 25°C
> [DEBUG] liquidctl.driver.usb: writting report 0x02 with 64 bytes: 4d:85:19:23:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [INFO] liquidctl.driver.kraken2: setting fan PWM duty to 35% for liquid temperature >= 26°C
> [DEBUG] liquidctl.driver.usb: writting report 0x02 with 64 bytes: 4d:86:1a:23:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[... cut ...]
> [DEBUG] liquidctl.driver.usb: writting report 0x02 with 64 bytes: 4d:a2:3a:64:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> [INFO] liquidctl.driver.kraken2: setting fan PWM duty to 100% for liquid temperature >= 60°C
> [DEBUG] liquidctl.driver.usb: writting report 0x02 with 64 bytes: 4d:a3:3c:64:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
> 
> C:\Users\USER>

jonasmalacofilho, 2020-08-01:
Can you run the same command again, but with [liquidctl-fcb18d1fad46-bin-windows-x86_64.zip](https://ci.appveyor.com/api/buildjobs/w13msedqq8peb7r6/artifacts/liquidctl-fcb18d1fad46-bin-windows-x86_64.zip)?  I enabled additional checks which may help identify some possible problems.

zwu513, 2020-08-01:
It seems to be fixed after I plugged all the headers in. Liquidctl is now displaying the correct fan speed. The issue might just be that my X52 fan connector is getting lose.

closed 2020-08-01 by zwu513