liquidctl/liquidctl #164: NZXT Kraken X61 set pump odd behavior

MrEricM, 2020-08-02:
I have an NZXT X61 and an X53 on the same system. The X53 seems to work fine, but overall seems to be more capable/updated. I'm getting strange results with the X61 when it comes to setting the pump speed. It just doesn't seem like the value is setting, and the resulting pump RPM just doesn't seem to make sense. (Sometimes it's higher, lower, or is sometimes different for the same value.) The fan control for the X61 works as expected though.

Here is how I'm initializing everything at system boot:
```
liquidctl.exe --match x53 initialize
liquidctl --match x53 set pump speed 35 20  50 80  55 100
liquidctl.exe --legacy-690lc --match 690LC initialize
liquidctl.exe --legacy-690lc --match 690LC set logo color fixed ffffff
liquidctl.exe --legacy-690lc --match 690LC set pump speed 20
liquidctl.exe --legacy-690lc --match 690LC set fan speed 50
```

However, when I set different pump speeds, it doesn't really seem to take effect, or at least not what I expect.

For example:
liquidctl.exe --legacy-690lc --match 690LC set pump speed 20

Then: liquidctl.exe --legacy-690lc --match 690LC status
```
Asetek 690LC (assuming NZXT Kraken X) (experimental)
├── Liquid temperature       41.5  °C
├── Fan speed                 780  rpm
├── Pump speed               2280  rpm
└── Firmware version      0.7.0.0
```
20% Duty = 2280 rpm

Now: liquidctl.exe --legacy-690lc --match 690LC set pump speed 50

Status returns:
```
Asetek 690LC (assuming NZXT Kraken X) (experimental)
├── Liquid temperature       41.3  °C
├── Fan speed                 780  rpm
├── Pump speed               2400  rpm
└── Firmware version      0.7.0.0
```
50% duty = 2400 rpm

Now: liquidctl.exe --legacy-690lc --match 690LC set pump speed 100

Status:
```
Asetek 690LC (assuming NZXT Kraken X) (experimental)
├── Liquid temperature       41.3  °C
├── Fan speed                 780  rpm
├── Pump speed               1920  rpm
└── Firmware version      0.7.0.0
```
100% duty = 1920 rpm??

Here are some debug outputs:
liquidctl.exe --legacy-690lc --match 690LC set pump speed 20 --debug

```
[DEBUG] __main__: running liquidctl v1.4.0 (AppVeyor; liquidctl-1.4.0-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 06a3:5021
[DEBUG] liquidctl.driver.usb: probing drivers for device 06a3:5081
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1297
[DEBUG] liquidctl.driver.usb: probing drivers for device 0fd9:006c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1290
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[... 249 lines cut ...]
[DEBUG] liquidctl.driver.usb: selected interface: 0
[DEBUG] liquidctl.driver.asetek: set clear to send = True
[DEBUG] liquidctl.driver.usb: sending control transfer with (64, 2, 2), {}
[DEBUG] liquidctl.keyval: data in C:\Users\Eric\AppData\Local\Temp\liquidctl\vid2433_pidb200\bususb2_port11_2\legacy
[DEBUG] liquidctl.driver.asetek: begin transaction
[DEBUG] liquidctl.driver.usb: explicitly claim interface
[DEBUG] liquidctl.driver.usb: sending control transfer with (64, 2, 1), {}
[DEBUG] liquidctl.keyval: loaded pump_duty=100 (from C:\Users\Eric\AppData\Local\Temp\liquidctl\vid2433_pidb200\bususb2_port11_2\legacy\pump_duty)
[INFO] liquidctl.driver.asetek: setting pump duty to 100%
[DEBUG] liquidctl.driver.usb: writting 2 bytes: 13:64
[DEBUG] liquidctl.keyval: loaded fan_duty=50 (from C:\Users\Eric\AppData\Local\Temp\liquidctl\vid2433_pidb200\bususb2_port11_2\legacy\fan_duty)
[INFO] liquidctl.driver.asetek: setting fan duty to 50%
[DEBUG] liquidctl.driver.usb: writting 2 bytes: 12:32
[DEBUG] liquidctl.driver.usb: read 32 bytes: 03:0c:fe:eb:12:5f:7c:f7:07:80:27:12:ff:5c:0a:18:e7:18:fe:eb:20:dd:00:00:07:00:00:d3:47:80:67:11
[DEBUG] liquidctl.driver.usb: ensure interface is released
Asetek 690LC (assuming NZXT Kraken X) (experimental)
├── Liquid temperature       40.0  °C
├── Fan speed                 780  rpm
├── Pump speed               1920  rpm
└── Firmware version      0.7.0.0

[DEBUG] liquidctl.driver.usb: ensure interface is released
```

Just let me know if there are any other outputs that would be helpful, or anything I can try!

jonasmalacofilho, 2020-08-03:
Hi @MrEricM,

This is indeed weird behavior...

Unfortunately, I have no idea what's causing it.  As far as I can tell, all messages sent are exactly those (theoretically) known to work.
_(Although I admit that I haven't confirmed that for myself, since I only have access to a more modern Asetek 690LC)._

If it isn't too much trouble, please test with liquidctl 1.3.3, just so we can rule out bugs being introduced in the last (very minor) changes to the driver.

Assuming it's still reproduceable with 1.3.3, then I'll ask you to capture some USB traffic between CAM and the device while several fan/pump speed combinations are set.  Use Wireshark for that (I can give you more pointers if you need), and send me the capture files.  In all likelihood it should be an easy fix.

---

P.S. This device requires a custom driver that makes it unnaccessible to CAM, but please let me know if you have any other programs that can write or read to this device (or if nowadays CAM can do that too, even with the custom driver).  One of the quirks of these older Krakens is that reading requires writing some settings first, so even a innocent looking monitoring tool will end up controlling the device.

MrEricM, 2020-08-03:
It does unfortunately seem to do the same thing for me in 1.3.3 as well. The lighting control and fan control work, but pump control is a little off.

If you just let me know what you need from Wireshark or can point to a guide, I should be able to get those captures.

Regarding the driver portion, CAM does still see it but asks to re-install the driver before reading or controlling it. Oddly enough, HWInfo64 reads the device with the native NZXT driver: https://i.imgur.com/A3KezfW.png 

I did make sure to close it though (Although, it no longer reads it with the WinUSB driver, so probably not needed).

jonasmalacofilho, 2020-08-04:
Ok, let's take a closer look then.

There are a few guides online on how to capture USB traffic, but I'm not sure how up-to-date they are.  Plus, it's probably wise to have our own documentation on the subject, tailored to our specific use cases...

I just added a quick guide that covers how to capture USB traffic on a Windows host, using Wireshark and USBPcap: [docs/developer/capturing-usb-traffic.md](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/developer/capturing-usb-traffic.md).

Please let me know of any problems/difficulties you experience, so I can make the necessary improvements.

jonasmalacofilho, 2020-08-04:
Oh, the USB vendor and product IDs for your Kraken X61 are: `0x2433` and `0xb200`.

MrEricM, 2020-08-04:
Thanks for that, it looks easy enough.

So - I don't know exactly at what point this happened, but I re-installed CAM and the drivers but it looks like CAM doesn't see the ability to control the pump RPM at all. I can change profiles, but they don't affect the pump RPM, I can't set custom values on it either. Fan controller works fine, as it does with liquidctl. 

I'm pretty sure the WinUSB driver is completely removed, it's now seen as an Asetek VI USB Device in device manager. 

I'm going to guess that something is mucked up and maybe the pump is operating in some kind of fail-safe mode. That was probably the problem the whole time, and I don't really know how long that would have been happening, I'd guess its probably been two months since I've messed with the CAM profiles. 

Let's just close it at that, if I get a replacement or find anything else out I'll leave a comment.

closed 2020-08-04 by MrEricM

jonasmalacofilho, 2020-08-04:
Ah, ok, thanks for letting me know.