liquidctl/liquidctl #165: liquidctl v1.4.0 Error

jt7005, 2020-08-04:
I am trying to use version 1.4.0, but I keep getting this error: no devices matches available drivers and selection criteria
Everything worked fine in 1.33, but using the same command on 1.4.0  I get the error. I have gone back to 1.3.3

**liquidctl v1.4.0 (AppVeyor; liquidctl-1.4.0-bin-windows-x86_64)**

liquidctl initialize
Error: no devices matches available drivers and selection criteria

liquidctl list --verbose
Gives me a blank line

liquidctl status
Error: no devices matches available drivers and selection criteria

However.....
liquidctl.exe --version
**liquidctl v1.3.3 (AppVeyor; liquidctl-1.3.3-bin-windows-x86_64)**

liquidctl.exe status
NZXT Kraken X (X42, X52, X62 or X72)
├── Liquid temperature     26.9  °C
├── Fan speed                 0  rpm
├── Pump speed             2128  rpm
└── Firmware version      6.0.2

NZXT Smart Device V2 (experimental)
└── Noise level    50  dB

initialize all
NZXT Smart Device V2 (experimental)
├── Firmware version                      1.5.0
├── LED 1 accessory 1    HUE 2 LED Strip 300 mm
└── LED 1 accessory 2    HUE 2 LED Strip 300 mm

I am running windows 10

jonasmalacofilho, 2020-08-04:
Have you replaced the standard drivers using Zadig (or manually)?  If that's the case, restore the original Microsoft Generic HID driver; liquidctl 1.4.0 is a bit more strict about this than 1.3.3.

jt7005, 2020-08-04:
Yes that was it, Thank you very much for all your hard work, love this program.
jt

closed 2020-08-04 by jt7005