liquidctl/liquidctl #166: NZXT E850 status now throws error with 1.4.0

mindleak, 2020-08-04:
The previous release read all the stats just fine, but now when I run the status command, I get this:

PS E:\liquidctl> .\liquidctl status
NZXT Kraken X (X42, X52, X62 or X72)
├── Liquid temperature     30.4  °C
├── Fan speed              1100  rpm
├── Pump speed             2703  rpm
└── Firmware version      6.0.2

ERROR: Unexpected error with NZXT E850 (experimental)
Traceback (most recent call last):
AssertionError: invalid response (attempts=3)

And here is the debug output:

[DEBUG] __main__: device: NZXT E850 (experimental)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0xad with 63 bytes: 00:03:01:60:fc:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: aa:04:02:08:d2:45:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0xad with 63 bytes: 00:03:01:60:fc:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: aa:04:02:08:d2:45:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0xad with 63 bytes: 00:03:01:60:fc:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: aa:04:02:08:d2:45:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[ERROR] __main__: Unexpected error with NZXT E850 (experimental)
Traceback (most recent call last):
  File "liquidctl\cli.py", line 302, in main
  File "liquidctl\driver\nzxt_epsu.py", line 59, in get_status
  File "liquidctl\driver\nzxt_epsu.py", line 145, in _get_fw_versions
  File "liquidctl\driver\nzxt_epsu.py", line 113, in _exec_read
AssertionError: invalid response (attempts=3)

I know it can't currently control fan profiles or anything on the PSU, only read status, but wanted to put this out there so you knew.

Thanks!

jonasmalacofilho, 2020-08-04:
Thanks.  Can you show me the comparable debug output with 1.3.3?

mindleak, 2020-08-04:
Sure, here it is with 1.3.3:

[DEBUG] liquidctl.driver.usb: ensure interface is released
[DEBUG] __main__: device: NZXT E850 (experimental)
[DEBUG] liquidctl.driver.seasonic: write ad 00 03 01 60 fc (and 58 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 03 12 41 ef 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 03 01 60 8e (and 58 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 03 39 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 03 01 60 90 (and 58 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 03 00 00 ef 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 03 04 60 06 02 00 20 (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 03 01 19 8f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 00 8b (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 12 06 d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 00 8c (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 98 c9 ae 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 00 96 (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 6b e2 bb 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 03 04 60 06 02 01 20 (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 03 01 19 ed 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 01 8b (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 14 06 87 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 01 8c (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 4b c9 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 01 96 (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 ed db f5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 03 04 60 06 02 02 20 (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 03 01 19 4b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 02 8b (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 18 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 02 8c (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 85 c8 45 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 02 96 (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 30 d3 e5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 03 04 60 06 02 03 20 (and 55 padding bytes)
[DEBUG] liquidctl.driver.seasonic: received aa 03 01 19 29 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.seasonic: write ad 00 04 04 60 06 02 03 8b (and 55 padding bytes)
[... cut ...]
├── +12V #2 output power           31.41  W
├── +12V #3 output voltage         12.19  V
├── +12V #3 output current          1.04  A
├── +12V #3 output power           12.75  W
├── +5V output voltage              4.94  V
├── +5V output current              5.00  A
├── +5V output power               24.69  W
├── +3.3V output voltage            3.27  V
├── +3.3V output current            2.20  A
└── +3.3V output power              7.14  W

[DEBUG] liquidctl.driver.usb: ensure interface is released

jonasmalacofilho, 2020-08-04:
There are some things I can double check, but it's noteworthy that the PSU appears to have responded differently to the same commands.

Can you please also show me the complete output of

```
@with liquidctl 1.4.0
liquidctl --match e850 status --debug
```

with liquidctl 1.4.0.  I'm now specifically interested in the device probing preamble.

---

Another thing that may help me is to see the output from

```
@with liquidctl 1.3.3
liquidctl --match e850 status --debug --hid hid
```

which forces liquidctl 1.3.3 to use the same HID backend that's used in 1.4.0.

mindleak, 2020-08-05:
Sure, here is the output from 1.4.0:

.\liquidctl --match e850 status --debug
[DEBUG] __main__: running liquidctl v1.4.0 (AppVeyor; liquidctl-1.4.0-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0f08
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 9512:9512
[DEBUG] liquidctl.driver.usb: probing drivers for device 9512:9512
[DEBUG] liquidctl.driver.usb: probing drivers for device 9512:9512
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0f08
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 9512:9512
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c231
[DEBUG] liquidctl.driver.usb: probing drivers for device 9512:9512
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0f08
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0f08
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:0a5b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0f08
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 9512:9512
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0f08
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 045e:02ff
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:0a5b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0f08
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c232
[DEBUG] liquidctl.driver.usb: probing drivers for device 051d:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c0b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:0a5b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 7793:2500
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT E850 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:0a5b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0610
[DEBUG] liquidctl.driver.usb: probing drivers for device 051d:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0620
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:0a5b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c0b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:43d5
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 7793:2500
[DEBUG] liquidctl.driver.usb: probing drivers for device 045e:02d1
[... 83 lines cut ...]
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xa5b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0xc0b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1022:0x149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1022:0x43d5
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1532:0xc00
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x7793:0x2500
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x45e:0x2d1
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1532:0xf08
[... cut ...]
[DEBUG] liquidctl.driver.seasonic: received aa 04 02 92 cb 69 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ERROR] __main__: Unexpected error with NZXT E850 (experimental)
Traceback (most recent call last):
  File "liquidctl\cli.py", line 309, in main
  File "liquidctl\driver\seasonic.py", line 78, in get_status
  File "liquidctl\driver\seasonic.py", line 156, in _get_fw_versions
  File "liquidctl\driver\seasonic.py", line 124, in _exec_read
AssertionError: invalid response (attempts=3)

jonasmalacofilho, 2020-08-05:
Can you give [liquidctl-3d2c61e1eb35-bin-windows-x86_64.zip](https://ci.appveyor.com/api/buildjobs/7f8gcuwgaflsr9u0/artifacts/liquidctl-3d2c61e1eb35-bin-windows-x86_64.zip) a try and see if it helps?

mindleak, 2020-08-06:
That one works!

Here's the debug just in case you need it:

.\liquidctl --match e850 status --debug
[DEBUG] __main__: running liquidctl v1.4.0 (AppVeyor; liquidctl-3d2c61e1eb35-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0f08
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 9512:9512
[DEBUG] liquidctl.driver.usb: probing drivers for device 9512:9512
[DEBUG] liquidctl.driver.usb: probing drivers for device 9512:9512
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0f08
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 9512:9512
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c231
[DEBUG] liquidctl.driver.usb: probing drivers for device 9512:9512
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0f08
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0f08
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:0a5b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0f08
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 9512:9512
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0f08
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 045e:02ff
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:0a5b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0f08
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c232
[DEBUG] liquidctl.driver.usb: probing drivers for device 051d:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c0b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:0a5b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 7793:2500
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT E850 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:0a5b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0610
[DEBUG] liquidctl.driver.usb: probing drivers for device 051d:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0620
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0046
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0203
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:0a5b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c0b
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:149c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 1022:43d5
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0c00
[... 47 lines cut ...]
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: ad:00:04:04:60:06:02:04:8b:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: aa:04:02:a3:01:3b:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: ad:00:04:04:60:06:02:04:8c:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[... cut ...]
├── +12V motherboard/PCI-e output voltage         12.14  V
├── +12V motherboard/PCI-e output current          0.92  A
├── +12V motherboard/PCI-e output power           11.19  W
├── +5V combined output voltage                    4.95  V
├── +5V combined output current                    5.02  A
├── +5V combined output power                     24.78  W
├── +3.3V combined output voltage                  3.27  V
├── +3.3V combined output current                  2.55  A
└── +3.3V combined output power                    8.36  W

jonasmalacofilho, 2020-08-06:
Happy to hear!  And the rest of debug output also looks good to me.

Thanks!

---

Status: fix available in the main branch; will be included in liquidctl 1.4.1 (in a few days).

closed 2020-08-06 by jonasmalacofilho