liquidctl/liquidctl #167: Aorus Liquid Cooler support?

ArshZilpe, 2020-08-04:
Hey! I saw that liquidctl supports the NZXT Z63 and Z73 coolers, which is great.

I have the Aorus Liquid Cooler, which is essentially the exact same hardware, screen, and pump design since it's made by Asetek too. They have the exact same screens. I would love to see support for it, since it has awful software compared to NZXT's CAM.

Hopefully it's achievable. I am more than happy to do any testing if required.

Thanks!

jonasmalacofilho, 2020-08-04:
That's interesting, I didn't know they were so similar.  Though just to adjust expectations, they could still use very different firmware and protocols, as is the case with the 5th-gen. Asetek coolers.

But let's take a look.  Can you follow [Capturing USB traffic on a native Windows host](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/developer/capturing-usb-traffic.md#capturing-usb-traffic-on-a-native-windows-host) and share the PCAP capture files with me?

jonasmalacofilho, 2020-08-10:
Let me know if you need more help with the traffic captures.

ArshZilpe, 2020-08-13:
Hello. Sorry for the late reply. I have been trying to follow the 'Capturing USB traffic on a native Windows host' guide you linked, but I'm stuck. You say 'For this example, assume the target device has vendor and product IDs 0x1e71 and 0x170e, respectively.'

So for the Aorus Liquid Cooler, how am I meant to know which traffic is coming from it? It's not going to have the same IDs' right?

Thank you!

jonasmalacofilho, 2020-08-13:
Hi @ArshZilpe,

Right!  You can find right IDs in Device Manager (device -> ... -> Hardware IDs).

Additionally, feel free to send me the unfiltered captures, I may be able to spot the device through other means.

jonasmalacofilho, 2020-08-28:
Hi @ArshZilpe, any news on your end?

jonasmalacofilho, 2020-11-03:
Thanks to a kind soul on Reddit:

```
    Bus 001 Device 003: ID 1044:7a46 Chu Yuen Enterprise Co., Ltd USB-MASS STORAGE
    Couldn't open device, some information will be missing
    Device Descriptor:
      bLength                18
      bDescriptorType         1
      bcdUSB               1.10
      bDeviceClass            0 
      bDeviceSubClass         0 
      bDeviceProtocol         0 
      bMaxPacketSize0        64
      idVendor           0x1044 Chu Yuen Enterprise Co., Ltd
      idProduct          0x7a46 
      bcdDevice            1.00
      iManufacturer           1 
      iProduct                2 
      iSerial                 3 
      bNumConfigurations      1
      Configuration Descriptor:
        bLength                 9
        bDescriptorType         2
        wTotalLength       0x0040
        bNumInterfaces          2
        bConfigurationValue     1
        iConfiguration          0 
        bmAttributes         0x80
          (Bus Powered)
        MaxPower              100mA
        Interface Descriptor:
          bLength                 9
          bDescriptorType         4
          bInterfaceNumber        0
          bAlternateSetting       0
          bNumEndpoints           2
          bInterfaceClass         8 Mass Storage
          bInterfaceSubClass      6 SCSI
          bInterfaceProtocol     80 Bulk-Only
          iInterface              0 
          Endpoint Descriptor:
            bLength                 7
            bDescriptorType         5
            bEndpointAddress     0x81  EP 1 IN
            bmAttributes            2
              Transfer Type            Bulk
              Synch Type               None
              Usage Type               Data
            wMaxPacketSize     0x0040  1x 64 bytes
            bInterval               0
          Endpoint Descriptor:
            bLength                 7
            bDescriptorType         5
            bEndpointAddress     0x02  EP 2 OUT
            bmAttributes            2
              Transfer Type            Bulk
              Synch Type               None
              Usage Type               Data
            wMaxPacketSize     0x0040  1x 64 bytes
            bInterval               0
        Interface Descriptor:
          bLength                 9
          bDescriptorType         4
          bInterfaceNumber        1
          bAlternateSetting       0
          bNumEndpoints           2
          bInterfaceClass         3 Human Interface Device
          bInterfaceSubClass      0 
          bInterfaceProtocol      0 
          iInterface              0 
            HID Device Descriptor:
              bLength                 9
              bDescriptorType        33
              bcdHID               1.10
              bCountryCode           33 US
              bNumDescriptors         1
              bDescriptorType        34 Report
              wDescriptorLength      47
             Report Descriptors: 
               ** UNAVAILABLE **
          Endpoint Descriptor:
            bLength                 7
            bDescriptorType         5
            bEndpointAddress     0x83  EP 3 IN
            bmAttributes            3
              Transfer Type            Interrupt
              Synch Type               None
              Usage Type               Data
            wMaxPacketSize     0x0040  1x 64 bytes
            bInterval               1
          Endpoint Descriptor:
            bLength                 7
            bDescriptorType         5
            bEndpointAddress     0x04  EP 4 OUT
            bmAttributes            3
              Transfer Type            Interrupt
              Synch Type               None
              Usage Type               Data
            wMaxPacketSize     0x0040  1x 64 bytes
            bInterval               1
```

---

It's unlikely we'll find any similarities between this cooler and the NZXT Z63/Z73.

jonasmalacofilho, 2020-11-03:
More info:

> When changing the image or the orientation of the screen, the windows tool seems to recompile and upload the whole firmware again.

ArshZilpe, 2020-11-15:
Aw that sucks. I apologise for not responding again (I hadn't checked GitHub in a while). Thank you for all the efforts anyway!

I did notice one thing while looking around myself. In the Program Files for Aorus Engine (used to control the AIO), there is a 'FlashAIOImage.exe', 'FlashAIOImage.pdb', and 'FlashAIOImage.vhost.exe'.

Then there is also a folder named 'images' which has three .bin files named 0, 90 and 180. There are also two other folders with .recourse.dll files in them.

I have uploaded the folder is that is any help to you. @jonasmalacofilho 

[AIO.zip](https://github.com/jonasmalacofilho/liquidctl/files/5543312/AIO.zip)

ArshZilpe, 2020-11-23:
@jonasmalacofilho

jonasmalacofilho, 2020-11-23:
No news on my side, I haven't been able to decode those `.bin` files yet.

They may be the key to understanding the traffic captures.  By the way, protocol looks nothing like is used used by the Kraken Z coolers.

ArshZilpe, 2020-11-23:
I see. That's a shame. Thank you anyway! I'll keep digging around.

alkanna, 2021-03-24:
Hey, I started digging around and found some patterns already :
It seems that on my 360 version of the cooler (3 fans) there are always 5 data packets being sent when changing the configuration of the lightning (not this is only RGB control testing for now, I haven't dug into temperature reporting yet)

Let's take these 4 packets :
```
B0 02 FF 21 00 FF 21 00
B1 02 FF 21 00 FF 21 00
B2 02 FF 21 00 FF 21 00
B3 02 FF 21 00 FF 21 00
```

This happens on saving some RGB config for the AIO. So far I can only guess that the B0/B1/B2/B3 are the three fans and something else (haven't found that out yet), the next byte (here 02) is the current RGB mode. This AIO supports Static, pulse, flash, double flash etc... (11 total including off), and it seems to start at 01 with 01 being Static, 02 is Pulse, and so on.

The next 6 bytes are RGB, for some reason sent two times for each fan (FF 21 00 being 255/33/0 in RGB)

There is also another packet send right before these 4 which contains some more settings, let's take for example this one :

```
C9 02 64 05 00 00 00 00
```

First byte - No idea yet
Second byte (02) - Speed of the effect, ranging from 00 to 05 I think
For the rest I got no clue yet, but I haven't played a lot with the rest of the settings.

At first I wanted to make a golang daemon to control this from OSX, but it seems wiser to add suppot for it in liquidctl so I hope someone else will help 😄

jonasmalacofilho, 2021-03-24:
Very nice! Can you attach a traffic capture?

---

To get started writing a liquidctl driver, I suggest you begin with basic temperature/fan speed monitoring, then go into fan speed control, and let fan LED control [and the screen] for last.

[Porting drivers from OpenCorsairLink](https://github.com/liquidctl/liquidctl/blob/master/docs/developer/porting-drivers-from-opencorsairlink.md), despite the name and obvious focus on a specific scenario, should help get give you an overview of how liquidctl is structured.

And you can also ask me questions on our [developer's Discord server](https://discord.gg/CFwbjqPx).

alkanna, 2021-03-25:
@jonasmalacofilho The discord invite is expired is seems :)

jonasmalacofilho, 2021-03-25:
@alkanna updated.

ArshZilpe, 2021-03-25:
@jonasmalacofilho Hey! I have attached a traffic capture (I'm pretty sure). I'm still learning how to do this haha.

One thing I noticed was that every time the temperature changed on the display, there is a chunk of `SET_REPORT` Request and Response packets. I'm not 100% sure if they're related though.

[AIOcapture.zip](https://github.com/liquidctl/liquidctl/files/6208489/AIOcapture.zip)

jonasmalacofilho, 2021-05-10:
Not that I'm aware.  But I'm available to help anyone working on adding support for this device.

fearless-spider, 2021-11-29:
> Not that I'm aware. But I'm available to help anyone working on adding support for this device.

Hello, I have this device too, version Waterforce X 360. Im debuging it. It works as HID Interrupt Transfer. How can I send data to this device?

Thanks

jonasmalacofilho, 2021-11-29:
@fearless-spider ,

For HIDs, we use [hidapi](https://github.com/libusb/hidapi/) through [cython-hidapi](https://github.com/trezor/cython-hidapi).  You can also experiment with [hidpytoy](https://github.com/todbot/hidpytoy) (I haven't used it myself _yet)._

One thing though: according to its USB [descriptors](https://github.com/liquidctl/liquidctl/issues/167#issuecomment-721178804), that device has two interfaces, and you'll probably need to use both to control all features.  For the non-HID interface you can use [libusb](https://github.com/libusb/libusb) (we use it through [PyUSB](https://github.com/pyusb/pyusb)).

That said, it's possible that the HID interface is enough for status monitoring and fan/pump speed control, maybe even LED control.

fearless-spider, 2021-11-29:
Thank You @jonasmalacofilho I will play with it

thankjura, 2021-12-16:
> Thank You @jonasmalacofilho I will play with it

Did anything work out?

fearless-spider, 2021-12-16:
> @fearless-spider
> 
> > I will play with it
> 
> Did anything work out?

I had no time. A lot of work at the end of year.

jonasmalacofilho, 2022-07-03:
Triage: is anyone interested in working on this?

Wapitiii, 2022-09-16:
> Triage: is anyone interested in working on this?

I would be interested.

jonasmalacofilho, 2022-09-21:
@Wapitiii,

Great. I think you could start by taking a look at the previous comments and at our [developers docs](https://github.com/liquidctl/liquidctl/tree/main/docs#documentation-for-developers).  And I'm around if you have questions or need help.

flisboac, 2022-10-01:
If I may, I also want to express some support for Gigabyte AIOs in general. I have an Aorus Waterforce 240, and it would be awesome to be able to control fan and pump speed from liquidctl on Linux. Right now, the device isn't even recognized by liquidctl, but I can see it:

```sh
$ lsusb | grep -i gigabyte
Bus 003 Device 004: ID 1044:7a51 Chu Yuen Enterprise Co., Ltd GIGABYTE CPU Cooler

$ lsusb -s 003:004 -vvv
Bus 003 Device 004: ID 1044:7a51 Chu Yuen Enterprise Co., Ltd GIGABYTE CPU Cooler
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1044 Chu Yuen Enterprise Co., Ltd
  idProduct          0x7a51 
  bcdDevice            1.00
  iManufacturer           1 ROCCAT
  iProduct                2 GIGABYTE CPU Cooler
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0062
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      24
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      34
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Warning: Descriptor too short
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         2
          bDescriptorType        34 Report
          wDescriptorLength      47
          bDescriptorType         0 (null)
          wDescriptorLength       0
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

flisboac, 2022-10-02:
If it may help, I made some USB captures for the Aorus Waterforce 240, following the guidelines on the documentation. All of them were made with Wireshark, on Windows, using USBPcap. On some of the captures, I manipulated Aorus Engine, as noted in the capture files' names.

https://drive.google.com/drive/folders/1kp7IBvVG80fekMy7w9_8m9bmitH86COp?usp=sharing

In the folder, there's also a hardware report I generated with hwinfo64.

namidairo, 2022-11-28:
Looked at a few captures of my Waterforce X 360. (1044/7A4D)

The traffic looks quite a bit different from the other captures here, but I can at least spot the status messages on endpoint 0x81.

Most of the messages in the hid data (there's many with 0 length) appear to go something along the lines of:

byte magic (Always 0x99)
byte command/response
and then the command/response data, padded out to 0x100 or 0x1800.

For the status response (0xDA on the first device), it looks like a bunch of null separated values for the fan speed, pump speed, an unknown 4 byte, and the liquid temperature.

I believe most of the settings write end up on the second endpoint, as I can observe fan curve edits etc in various messages. (Along with constant updates like CPU name, usage % etc.)
[Waterforce X 360 Capture.zip](https://github.com/liquidctl/liquidctl/files/10100728/Waterforce.X.360.Capture.zip)

namidairo, 2022-12-06:
I'm not too sure how one would implement this well for the Waterforce X series, as the fan/pump curves use a service on the host to keep it updated with the cpu temp value (via the 0xE0 command type), and the static duty set command appears to be a no-op for me. (???)

I could just set a flat fan curve as a jank workaround, but then you still have the issue of the screen elements that update based on the 0xE0 command. Or perhaps extend yoda out.

Side note: I also briefly managed to put mine in a weird state where it was returning garbage rpms and corrupted the fan curve. Odd, considering I hadn't even attempted to set it yet.

amazonparrot, 2022-12-12:
I just bought the waterforce x 240 and found that it's not supported in linux, thx to this post and especially @namidairo decoding, I can now control the fan speed by feeding in my cpu temperature with a python, using pyusb. Let me know if anyone is interested in the code. It does not integrate with liquidctl though.

fearless-spider, 2022-12-12:
Hi @amazonparrot , interesting, can you share your python code?  I have wateerforce x360 .

amazonparrot, 2022-12-13:
@fearless-spider, code as follow
you might need to install psutil and pyusb

import usb.core
import usb.util
import psutil
import time
def get_cpu_temp():
    t = psutil.sensors_temperatures()
    for x in ['coretemp']:
        if x in t:
            return int(t[x][0].current)
    print("Warning: Unable to get CPU temperature!")
    return 0 
def makecpuupdatestr():
  #change to your number of core 
  ncore=24
  #change to your number of threads
  nthread=32
  #change to your cpu Ghz
  ghz=5.5
  
  cputemp="{:02x}".format(get_cpu_temp())
  ncorestr="{:02x}".format(ncore)
  nthreadstr="{:02x}".format(nthread)
  ghz1="{:02x}".format(int(ghz))
  ghz2="{:02x}".format(int( (ghz-int(ghz))*10) )
  print(ghz1)
  print(ghz2)
  data="99e000"+cputemp+"18"+ghz1+ghz2+ncorestr+nthreadstr
  print(data)
  data+="0"*(6144*2-len(data))
  barray=bytes.fromhex(data)
#  print(len(barray))
  return barray
def readhexdump(filename):
  f=open(filename,"r")
  data=f.read()
  barray=bytes.fromhex(data)
  return barray
# find our device
dev = usb.core.find(idVendor=0x1044, idProduct=0x7a4d)

# was it found?
if dev is None:
    raise ValueError('Device not found')
bdata=makecpuupdatestr()
endpoint = dev[0][(0,0)][0]
endpoint2 = dev[0][(0,0)][1]
interface=0
if dev.is_kernel_driver_active(interface) is True:
  # tell the kernel to detach
  dev.detach_kernel_driver(interface)
  usb.util.claim_interface(dev, interface)
while(True):
 try:
     bdata=makecpuupdatestr()
     dev.write(endpoint2,bdata)
 except usb.core.USBError as e:
        print(e)
 time.sleep(2)

amazonparrot, 2022-12-13:
@fearless-spider code as attached, you would need to install pyusb and psutil, help it would work on your waterforce x 360 also 
[testpyusb.py.txt](https://github.com/liquidctl/liquidctl/files/10213586/testpyusb.py.txt)

fearless-spider, 2022-12-13:
Thank You @amazonparrot :)

amazonparrot, 2022-12-13:
I am not sure what's the status of others development, so took some time to found the following, hope it might help, will write some python code based on the following for some poc, later.

for setting the fan and pump mode
99(magic) e5(set fan pump op) 01(01=fan,02=pump) 06(mode)
mode code as follow:
07 zero
00 balance
05 performance
06 quiet
04 max
02 default
01 custom

below not tested yet:
for setting the custom curve for fan, it seems to be the following:
99 e6 01 01 00(degree) 03 ef(speed 1007) ...... (seems up to 4 (degree,speed) tuples)

for setting the custom curve for pump, it might be the following:
99 e6 04 02 00(degree) 03 ef(speed 1007) ...... (seems up to 4 (degree,speed) tuples)

for reading cooler stat
99 da ..... 
return 
99 da 62 03(fan speed) 00 f5 04(pump speed) 00 [20 25 1f a1](unknown) 00 1a(liquid degree)

99 d9 01 .... (read custom fan curve)
99 d9 02 .... (read custom pump curve)

namidairo, 2022-12-13:
> for reading cooler stat
99 da .....
return
99 da 62 03(fan speed) 00 f5 04(pump speed) 00 20 25 1f a1 00 1a(liquid degree)

Bytes [8] and [9] appear to be the fan and pump duty percentages, respectively. Thus, you are running the fan and pump at 32/37% in that example.

Additionally, if you're going to start setting pump curve, you should be checking the model + firmware and clamping the rpm where necessary? Gigabyte seems to set the lower boundary at 750rpm, and the upper at 2800 or 3200 depending on firmware version. (There are also some model checks involved, but they are against what looks like OEM versions of the cooler with different usb PID.)

There's also a command to get the model variant from the device.

Eg. https://github.com/namidairo/liquidctl/commit/a56db61350d01db8c3a008bb8816d870f6f2350d

```
Init
Gigabyte AORUS WATERFORCE X (240, 280, 360)
├── Firmware version                F1.6
└── Model variant       WATERFORCE X 360

Status
Gigabyte AORUS WATERFORCE X (240, 280, 360)
├── Liquid temperature      26.0  °C
├── Fan speed               1324  rpm
├── Pump speed              1841  rpm
├── Fan duty                  50  %
├── Pump duty                 57  %
├── Fan mode              custom
└── Pump duty             custom
```

As I mentioned before, most of the communication for this is now handled in managed code in Gigabyte Control Centre, which makes it easier to inspect. Take a look at `C:\Program Files\GIGABYTE\Control Center\Lib\GBT_Cooler\GcLib.dll`, `GcLib.PanelApi.Aio2Panel`. Basically everything minus the fan rgb is handled through here. Hence the appearance of a static speed command that does... nothing. Note that the get/set colour functions are in reference to the panel accent colours, not the fans.

joaortizro, 2023-04-24:
any updates? . I have the same cooling system , if there's anything I can do willing to help, I'm seriously thinking to buy a noctua  since the noise on Linux is not desirable when working  I guess it starts at default operation

fearless-spider, 2023-04-24:
Hi @joaortizro 

Here is my copy of the python script, it works on my computer, I tested it on Fedora and Arch linux. To run it after boot, you have to create systemd task and install pyusb, psutils for root.

https://github.com/fearless-spider/python_playground/blob/master/testpyusb.py

joaortizro, 2023-04-24:
@fearless-spider ty will take a look

amazonparrot, 2023-04-25:
I am using this https://github.com/amazonparrot/waterforce_x240_poc/blob/main/testpyusb.py, but I think @namidairo has make some code for liquidctl too, but I haven't got the time to test it yet. @fearless-spider I have added in code to control the fan color also so that we would know it's working, feel free to play around with it. My code also made use of some other information from @namidairo and other folks in this post, I hope someone could put this to the kernel so that we could use it in proxmox too.....

namidairo, 2023-04-25:
Suggest you take a look at https://github.com/liquidctl/liquidtux instead, as those are actually going upstream. (And thus got the old LKML treatment in terms of code review)

amazonparrot, 2023-04-25:
> Suggest you take a look at https://github.com/liquidctl/liquidtux instead, as those are actually going upstream. (And thus got the old LKML treatment in terms of code review)
Thx @namidairo, will have a look.

aleksamagicka, 2023-04-25:
Yes, kraken drivers in liquidtux are certainly up to date on the hwmon part.

For the Waterforce driver, your previous message inspired me. [Here it is](https://github.com/aleksamagicka/waterforce-hwmon/), compile tested only as I don't have the devices myself. Offsets taken from @namidairo's code - for now I've lumped them all into RPM readings, but duty readings should become more specialized when this is tested to be working.

To test, just run `make dev`. Proper README coming soon. Appreciate testing and comments, hopefully I didn't screw anything up!

aleksamagicka, 2023-04-25:
A basic readme is up. Can you please check that `cat /sys/class/hwmon/*/name | grep waterforce` lists it? It should at least be there.

amazonparrot, 2023-04-27:
> A basic readme is up. Can you please check that `cat /sys/class/hwmon/*/name | grep waterforce` lists it? It should at least be there.
thx a lot for your help.
my bad, I didn't reattach the device to kernel after running my python code, after I reattached, I can see the Waterforce showing in the sensors output but the reading are zero, last I tested. I suspect that we might need to use the usb_interrupt_msg to send command to the waterforce. based on your code I was testing with usb_bulk_msg to send the command, but it didnt work, I will try usb_interrupt_msg later.

amazonparrot, 2023-04-28:
> Been looking some more at your python script. `0x99` looks like the HID report ID?
> 
> Edit: from the descriptors shared above, `0x02` is the OUT of the non HID endpoint, which is what the python code uses. But, unless I'm missing something, the liquidctl code uses the HID endpoint, which suggests that they might be interchangeable, at least for the sensor stuff.
> 
> `usb_bulk_msg()` and `usb_interrupt_msg()` are the same - see [Elixir](https://elixir.bootlin.com/linux/v6.3/source/drivers/usb/core/message.c#L327).

I just managed to get some output, based on your code, the u16 speed_input[2]; should be u16 speed_input[4];
waterforce_raw_event should read the data variable instead of the buffer.
temperature sensor offset I need to double check, but the reading is in Celsius we would need to convert back to F?
my previous attempt to use it as usb device seems too complicated as it crash with the usbhid driver I think.
Glad to have your help, great work from you considering you don't have the device. The rest of the goal would be to get the temperature of the CPU and feed it to the Waterforce. 

Could some other folks test this also? coz I just have the Waterforce x240 model.

amazonparrot, 2023-04-28:
output from sensors now:
waterforce-hid-3-6
Adapter: HID adapter
Fan speed:    1848 RPM
Pump speed:   2430 RPM
Fan duty:       64 RPM <-- might need fixing the unit for this
Pump duty:      65 RPM <-- might need fixing the unit for this
Coolant temp:  +26.3°C

aleksamagicka, 2023-04-28:
Terribly sorry for the oversights (am juggling this and graduation). Will fix immediately.

Temps in hwmon are in Celsius. Userspace can convert to F. As for pump and fan duty, I'll rework them to be exposed as PWM values (0-255) - for now, this was just a test to fetch them correctly.

aleksamagicka, 2023-04-28:
Some questions regarding the device:

- How important are the CPU cores and frequency [here](https://github.com/amazonparrot/waterforce_x240_poc/blob/main/testpyusb.py#L110)?
- Setting points for [a custom curve](https://github.com/amazonparrot/waterforce_x240_poc/blob/main/testpyusb.py#L19). Do we know how many points that curve has? Are the degrees relative to something (a temp of 2C seems weird for a loop)? What is the speed referring to (RPM)? It looks like its big endian?

The fan and pump modes could be implemented as pwm_enable entries, so I'm trying to learn more about the custom mode. If implementing the custom mode is feasible, then additional handling for fancontrol will most likely be needed.

amazonparrot, 2023-04-29:
> Some questions regarding the device:
>
> * How important are the CPU cores and frequency [here](https://github.com/amazonparrot/waterforce_x240_poc/blob/main/testpyusb.py#L110)?
> * Setting points for [a [...]
The CPU temperature is for the fan to look up the corresponding fan speed it should use. The display of the waterforce will also show the CPU temperature and the frequency. 

for setting the custom curve for fan, it seems to be the following:
99 e6 01 01 00(degree celsius) 03 ef(speed 1007) ...... (seems up to 4 (degree,speed) tuples)

for setting the custom curve for pump, it might be the following:
99 e6 04 02 00(degree celsius) 03 ef(speed 1007) ...... (seems up to 4 (degree,speed) tuples)

The speed is in RPM.
The curve is a lookup table for controlling the fan and pump speed given the CPU temperature.

amazonparrot, 2023-04-29:
https://github.com/amazonparrot/waterforce_x240_poc/blob/main/waterforce.c 
I am currently using this code, but I have yet to find out how to trigger a event to do the update.

joaortizro, 2023-04-29:
Btw i have the x360 , how i can help ?

aleksamagicka, 2023-04-29:
@joaortizro, you can help test. Clone [this repo](https://github.com/aleksamagicka/waterforce-hwmon/), run `make dev` in the terminal and look up `sensors`. The device parameters should show up under `waterforce`. Please report if anything looks off, or doesn't show up at all. Thanks!

@amazonparrot, from the driver we can only expose writing some temp to the device. The temp value is provided by userspace, we can't look it up from the driver. As far as the driver is concerned, it may not even be running on x86 machines. Same goes for RGB stuff - that's suited for liquidctl which is specifically geared towards that, as the kernel does not (yet) concern itself with that.

As for the custom curve, looks like that needs additional exploration, so I think it would be good to implement all the other fixed modes for now. Custom curve support can always come later, both to the driver and liquidctl at some point.

joaortizro, 2023-04-29:
@aleksamagicka Ok, testing now

amazonparrot, 2023-04-29:
> @joaortizro, you can help test. Clone [this repo](https://github.com/aleksamagicka/waterforce-hwmon/), run `make dev` in the terminal and look up `sensors`. The device parameters should show up under `waterforce`. Please report if anything looks off, or doesn't show up at all. Thanks!
> 
> @amazonparrot, from the driver we can only expose writing some temp to the device. The temp value is provided by userspace, we can't look it up from the driver. As far as the driver is concerned, it may not even be running on x86 machines. Same goes for RGB stuff - that's suited for liquidctl which is specifically geared towards that, as the kernel does not (yet) concern itself with that. That's why it enables hidraw in the probe method.
> 
> As for the custom curve, looks like that needs additional exploration, so I think it would be good to implement all the other fixed modes for now. Custom curve support can always come later, both to the driver and liquidctl at some point.

agree that we shouldn't get the CPU temp and change RGB color in the module, I am just doing it for testing and also for my use.
somehow the latest verison of your code will lock up or crash the kernel, I have fixed that in my version. changes are as follow:
change the PWM code and the MAX_PACKET_LENGTH to 6144. @joaortizro please do see if @aleksamagicka would work on your side, as somehow I need to modify something before it worked.
also the TEMP offset is changed.

joaortizro, 2023-04-29:
```
waterforce-hwmon on  master via C v11.3.0-gcc 
❯ make dev
make clean
make[1]: Entering directory '/home/jonas/waterforce-hwmon'
make -C /lib/modules/5.19.0-40-generic/build M=$PWD clean
make[2]: Entering directory '/usr/src/linux-headers-5.19.0-40-generic'
make[2]: Leaving directory '/usr/src/linux-headers-5.19.0-40-generic'
make[1]: Leaving directory '/home/jonas/waterforce-hwmon'
make
make[1]: Entering directory '/home/jonas/waterforce-hwmon'
make -C /lib/modules/5.19.0-40-generic/build M=$PWD modules
make[2]: Entering directory '/usr/src/linux-headers-5.19.0-40-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
  CC [M]  /home/jonas/waterforce-hwmon/waterforce.o
  MODPOST /home/jonas/waterforce-hwmon/Module.symvers
  CC [M]  /home/jonas/waterforce-hwmon/waterforce.mod.o
  LD [M]  /home/jonas/waterforce-hwmon/waterforce.ko
  BTF [M] /home/jonas/waterforce-hwmon/waterforce.ko
Skipping BTF generation for /home/jonas/waterforce-hwmon/waterforce.ko due to unavailability of vmlinux
make[2]: Leaving directory '/usr/src/linux-headers-5.19.0-40-generic'
make[1]: Leaving directory '/home/jonas/waterforce-hwmon'
sudo rmmod waterforce || true
rmmod: ERROR: Module waterforce is not currently loaded
sudo insmod waterforce.ko

```

amazonparrot, 2023-04-29:
> ```
> waterforce-hwmon on  master via C v11.3.0-gcc
> ❯ make dev
> make clean
> make[1]: Entering directory '/home/jonas/waterforce-hwmon'
> make -C /lib/modules/5.19.0-40-generic/build M=$PWD clean [...]

this is normal output from compiling, I got the same output as well.

aleksamagicka, 2023-04-29:
@joaortizro, the driver is now inserted, look up `sensors`.

@amazonparrot, OK, so the report size is 6144, will update. It was 64 in liquidctl code, which is where I got it from. What is changed regarding PWM?

joaortizro, 2023-04-29:
i guess thats a sudo apt-get sensors? sorry if this is obvious for you guys

aleksamagicka, 2023-04-29:
I believe it's `sudo apt install lm-sensors`.

[71 more replies]

closed 2025-11-15 by jonasmalacofilho