liquidctl/liquidctl #168: NZXT RGB & Fan Controller can not be detected under hackintosh

zk2004mb, 2020-08-05:
Hi! Thank you so much for creating such a great app. I'm running an AMD based hackintosh and my NZXT RGB & Fan Controller can not be found by liquidctl(1.4.0). I can use the device normally in Windows 10 by CAM. Do you have any suggestions?

jonasmalacofilho, 2020-08-05:
Can you show me the output of `liquidctl list --debug` on macOS?

zk2004mb, 2020-08-05:
[DEBUG] liquidctl.cli: running liquidctl v1.4.0 (homebrew; liquidctl 1.4.0)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device 046a:00b4
[DEBUG] liquidctl.driver.usb: probing drivers for device 046a:00b4
[DEBUG] liquidctl.driver.usb: probing drivers for device 046a:00b4
[DEBUG] liquidctl.driver.usb: probing drivers for device 046a:00b4
[DEBUG] liquidctl.driver.usb: probing drivers for device 046a:00b4
[DEBUG] liquidctl.driver.usb: probing drivers for device 046a:00b4
[DEBUG] liquidctl.driver.usb: probing drivers for device 046a:00b4
[DEBUG] liquidctl.driver.usb: probing drivers for device 043e:9a39
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0765:5020
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0265
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0265
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0265
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0265
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0265
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0265
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0265
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0265
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0265
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:ace1
[DEBUG] liquidctl.driver.usb: probing drivers for device 046a:00b4
[DEBUG] liquidctl.driver.usb: probing drivers for device 043e:9a39
[DEBUG] liquidctl.driver.usb: probing drivers for device 0765:5020
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c531
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0743
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bc2:ab28
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8140

jonasmalacofilho, 2020-08-05:
Thanks!

It seems that your mac OS isn't seeing the device.  I know tuning in a working USB configuration is a big part of building a Hackintosh, so that's probably where the problem lies.

zk2004mb, 2020-08-05:
I actually finished the USB mapping. But sometimes the internal usb port can be tricky. I will try to remap internal ports and see if there is any difference. Thank you! Will report back!

jonasmalacofilho, 2020-08-10:
Did you manage to sort it?

zk2004mb, 2020-08-17:
Sorry for the late response. The problem was on the usb port mapping. I remap the internal ports and the controller now can be detected.

jonasmalacofilho, 2020-08-17:
Glad to hear that, thanks for letting me know!