liquidctl/liquidctl #170: systemctl service failing on boot

brandon-feder, 2020-08-08:
Hello. I created a systemctl service as follows:
```
[Unit]
Description=AIO startup service

[Service]
Type=oneshot
ExecStart=/home/brandonf/Documents/scripts/init_liquidctl

[Install]
WantedBy=default.target
```
in `/etc/systemd/system/liquidcfg.service`. (Note that its `systemd` not `system.d` at least on my Manjaro system.) However, after reloading the ademon, enabling, then starting the service, during boot I get the following errors:

```
Aug 08 09:19:27 BrandonsWorkstation systemd[1]: Starting AIO startup service...
Aug 08 09:19:27 BrandonsWorkstation init_liquidctl[621]: Error: no devices matches available drivers and selection criteria
Aug 08 09:19:27 BrandonsWorkstation init_liquidctl[731]: Error: no devices matches available drivers and selection criteria
Aug 08 09:19:27 BrandonsWorkstation init_liquidctl[734]: Error: no devices matches available drivers and selection criteria
Aug 08 09:19:27 BrandonsWorkstation systemd[1]: liquidcfg.service: Main process exited, code=exited, status=1/FAILURE
Aug 08 09:19:27 BrandonsWorkstation systemd[1]: liquidcfg.service: Failed with result 'exit-code'.
Aug 08 09:19:27 BrandonsWorkstation systemd[1]: Failed to start AIO startup service.
```

I am not sure whether I am doing something wrong or there is a bug, or there is supposed to be a `system.d` folder but there is not, etc. Please help. If it matter, I am using liquidctl with a Corsair H!00i Platinum SE which I know is not yet official supported. However, using the unofficial repository, I was able to get it and it works great so I am relatively confident it is not an error with liquidctl itself.

jonasmalacofilho, 2020-08-08:
Let me take a look at this `init_liquidctl` script.

Also, what's the output of `liquidctl list -g`?

brandon-feder, 2020-08-08:
Here is `init_liquidctl`:
```
#!/bin/bash
liquidctl initialize
liquidctl set led color fixed ffffff
liquidctl set fan speed 20 20 30 25 40 30 50 40 60 50 70 90 75 100
```
If I run it normally, everything works as expected so I am sure its not a problem with the script on its own.

Here is the output of `liquidctl list -g`:
```
[DEBUG] liquidctl.cli: running liquidctl v1.4.0 (2b59f9574ff0)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c19
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H100i Platinum SE (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:005c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:005c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:005c
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0105
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0105
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0105
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 24f0:0105
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:005c
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:082d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bda:5401
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18f3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0930:6545
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c19
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
Device ID 0: Corsair H100i Platinum SE (experimental)
├── Vendor ID: 0x1b1c
├── Product ID: 0x0c19
├── Release number: 0x0100
├── Bus: hid
├── Address: /dev/hidraw0
└── Driver: HydroPlatinum using module hid
[DEBUG] liquidctl.cli: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice
```

jonasmalacofilho, 2020-08-08:
It's possible that the service is trying to start before the kernel the device is enumerated.

You should be able to confirm if this is case by taking a look at the rest of logs of the same boot, looking for a block of kernel messages that starts with a line that begins with:

```
New USB device found, idVendor=1b1c
```

brandon-feder, 2020-08-08:
Yep. I see it. Here is the output of `journalctl | grep "New USB device found, idVendor=1b1c"`:
```
Aug 06 16:44:52 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
Aug 06 16:51:28 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
Aug 06 17:09:13 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
Aug 06 18:55:04 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
Aug 07 14:05:47 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
Aug 07 14:06:26 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
Aug 07 14:36:08 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
Aug 08 06:51:55 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
Aug 08 07:09:52 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
Aug 08 09:06:38 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
Aug 08 09:08:25 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
Aug 08 09:14:22 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
Aug 08 09:16:16 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
Aug 08 09:19:26 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
```
What should I do about it? I am fairly new to linux so please give me detailed instructions.

jonasmalacofilho, 2020-08-08:
Actually, 

```
Aug 08 09:19:26 BrandonsWorkstation kernel: usb 1-5: New USB device found, idVendor=1b1c, idProduct=0c19, bcdDevice= 1.00
       ^^^^^^^^
```

would suggest that the device could have been enumerated before the service tried to start.

Of course things are happening in parallel, and it takes a while from when the device is first enumerated on the USB to when it's available on the HID bus.

After a while does `systemctl start liquidcfg` work?

brandon-feder, 2020-08-08:
No. On startup I get the an error that says something like `AIO startup service failed to start` in red text. (Don't remember exactly) after which my PCs rainbow lights come on.

jonasmalacofilho, 2020-08-09:
Hm, in that case, add:

```
which liquidctl
liquidctl list --debug
```

at the begenning of the `init_liquidctl` script, and try to start the service again.

```
systemctl start liquidcfg
journalctl -u liquidcfg.service --since '5 minutes ago'
```

jonasmalacofilho, 2020-08-10:
I'm not sure if you have resolved the issue by now, but just so you know where I was getting to: since you reported that starting the service always fails, even after the system has been on for a while (and manually entering `liquidctl list` on the command line does report your device), a possible explanation would be that the service is using another liquidctl binary.

This can happen because the service runs in a different environment than you as a user.  The PATH variable will usually be different, so it's somewhat easy to get to a state where, for example, you have two installations, one global at `/usr/bin` (that the service sees) and one local at `/usr/local/bin` (which is only in the PATH for your user).  The issue can also affect Python environments if you're activating them in your bashrc or something like it.

Anyway, by adding those two lines at the beginning of the script you will be able to (by comparing the output from starting the service vs. manually/directly running the script): (a) confirm that this is the case and (b) see which binary is being used in each case.

P.S. Small quirk: `liquidctl` isn't actually a binary, but a tiny script that starts liquidctl from the Python package.  So if both service and user can use same ~~binary~~ executable but still have an issue with which Python package installation is used, since that can also depend on the environment.  But this is probably less common...

brandon-feder, 2020-08-10:
Okay. I did the fix you proposed and... It Worked! Thanks so much. I appreciate it.

jonasmalacofilho, 2020-08-10:
Hm, that wasn't a fix per say, just adding more information to the output of the service to help debug the problem.

Or did you figure out and already fixed the underlying issue?

closed 2020-08-14 by jonasmalacofilho

trentech, 2021-11-14:
Hate to bring up and old closed issue, but I'm having this issue and it doesn't seem like this was ever really resolved. Everything works fine when executing manually but when I attempt to create a startup service I can run it manually, but when actually running at startup it fails. Below is my service file (added the extra debug commands for more information) and the errors thrown upon startup. Any help would be appreciated. 

```
[Unit]
Description=Corsiar startup service

[Service]
Type=oneshot
ExecStart=which liquidctl
ExecStart=liquidctl list --debug
ExecStart=liquidctl initialize all

[Install]
WantedBy=default.target
```t

```
Nov 14 12:52:44 terrenced02 systemd[1]: Starting Corsiar startup service...
Nov 14 12:52:44 terrenced02 which[684]: /usr/bin/liquidctl
Nov 14 12:52:44 terrenced02 liquidctl[686]: [DEBUG] cli (main): running liquidctl v1.7.1 (Arch Linux; 1.7.1-1)
Nov 14 12:52:44 terrenced02 liquidctl[686]: [DEBUG] smbus (find_devices): searching LinuxI2c (Ddr4Temperature, VengeanceRgb, EvgaPascal, RogTuring)
Nov 14 12:52:44 terrenced02 liquidctl[686]: [DEBUG] smbus (find_devices): found I²C bus i2c-0
Nov 14 12:52:44 terrenced02 liquidctl[686]: [DEBUG] smbus (find_devices): ignoring 0-0008, cannot infer bus number
Nov 14 12:52:44 terrenced02 liquidctl[686]: [DEBUG] usb (find_devices): searching HidapiBus (CommanderCore, CommanderPro, CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver)
Nov 14 12:52:44 terrenced02 liquidctl[686]: [DEBUG] usb (find_devices): searching PyUsbBus (Hydro690Lc, Legacy690Lc, Modern690Lc, _Base690Lc, _ModernBase690Lc, CorsairAsetekProDriver)
Nov 14 12:52:45 terrenced02 liquidctl[686]: [DEBUG] usb (find_devices): found USB device 1d6b:0003
Nov 14 12:52:45 terrenced02 liquidctl[686]: [DEBUG] usb (find_devices): found USB device 1d6b:0002
Nov 14 12:52:45 terrenced02 liquidctl[686]: [DEBUG] usb (find_devices): found USB device 0bda:0411
Nov 14 12:52:45 terrenced02 liquidctl[686]: [DEBUG] usb (find_devices): found USB device 0bda:8153
Nov 14 12:52:45 terrenced02 liquidctl[686]: [DEBUG] usb (find_devices): found USB device 0424:5537
Nov 14 12:52:45 terrenced02 liquidctl[686]: [DEBUG] usb (find_devices): found USB device 1d6b:0003
Nov 14 12:52:45 terrenced02 liquidctl[686]: [DEBUG] usb (find_devices): found USB device 046d:c52b
Nov 14 12:52:45 terrenced02 liquidctl[686]: [DEBUG] usb (find_devices): found USB device 0424:2137
Nov 14 12:52:45 terrenced02 liquidctl[686]: [DEBUG] usb (find_devices): found USB device 1d6b:0002
Nov 14 12:52:45 terrenced02 liquidctl[686]: [DEBUG] usb (find_devices): found USB device 1d6b:0003
Nov 14 12:52:45 terrenced02 liquidctl[686]: [DEBUG] usb (find_devices): found USB device 8087:0aa7
Nov 14 12:52:45 terrenced02 liquidctl[686]: [DEBUG] usb (find_devices): found USB device 1d6b:0002
Nov 14 12:52:45 terrenced02 liquidctl[1288]: Error: no devices matches available drivers and selection criteria
Nov 14 12:52:45 terrenced02 systemd[1]: fan.service: Main process exited, code=exited, status=1/FAILURE
Nov 14 12:52:45 terrenced02 systemd[1]: fan.service: Failed with result 'exit-code'.
Nov 14 12:52:45 terrenced02 systemd[1]: Failed to start Corsiar startup service.

```

jonasmalacofilho, 2021-11-14:
@trentech,

Being able to start the service manually, after the system has been on for a while, suggests that the issue is simply that, during startup, the service is trying to run before the device has been found by the kernel.

The debug output you posted seems to corroborate that, but you haven't mentioned which device you have so I don't know exactly what I'm looking for.

You can find some information on how to solve this in [`docs/linux/making-systemd-units-wait-for-devices.md`](https://github.com/liquidctl/liquidctl/blob/main/docs/linux/making-systemd-units-wait-for-devices.md).

And for a quick ~~fix~~ test, you could try adding a small sleep.

P.S. The original issue was different, since the service could not be started manually either.

trentech, 2021-11-14:
This is exactly what I needed. Adding the device rules resolved my issue. Thanks for the speedy response.