liquidctl/liquidctl #172: Wrong temperature readout X63

monitorherocg, 2020-08-12:
Hello guys,

I am referring to this issue here since I get the wrong temperature readout
https://github.com/jonasmalacofilho/liquidctl/issues/160
I wanted to avoid installing NZXT CAM alltogether so I straight went to liquidctl. I am not too familiar with command line tools but I hope you can help me. I am mostly trying to avoid the NXZT auto firmware update. The pump is used as GPU cooler for a 1080ti.
Under load I usually don't go above 60°C on the GPU die under heavy load but I am not able to readout the water temperature with HWINFO64.
Is there a way to fix this issue without installing CAM?
Thank you.

jonasmalacofilho, 2020-08-12:
I'm not sure.

As far as we were able to gather in that issue the problem seems to be related to the firmware (possibly a bug in how the analog thermocouple is "read" in some scenarios).  But it could also be that we're missing some rare and obscure packet that CAM can occasionally send to clear faults like these.

First, what firmware version do you have?  This is returned by `initialize`.

Second, try turning off the system completely (including the PSU), and keeping it that way for a few minutes.  This will clear the RAM in cooler's embedded controller and could clear a "soft" fault (one that depends on some packet we're not yet familiar with).

jonasmalacofilho, 2020-08-12:
See if this is also happening with your unit (observation from the other issue):

> I also noticed that it was running the pump at full speed, which suggests the controller itself was also "seeing" the same 280.5 °C value.

Try to set some simple yet sensible profile, like `set pump speed 20 50 50 100`, and post the output of `status`.

monitorherocg, 2020-08-12:
Hey thanks for the quick answer. Firmware is 1.8.0. The pump speed that HWINFO64 provides is between 1800 and 2800rpm. I will do a shutdown and power disconnect and see where that gets me.

monitorherocg, 2020-08-12:
So the reset worked. I now see 38.4°C temperature at 35% speed with 1255 rpm. Its pretty hot here with 30°C ambient. Not sure if that is a bad liquid temperature. I upped the pumpspeed to 100%. Let's see if it helps.

jonasmalacofilho, 2020-08-12:
Ok, thanks for letting me know.  And those results seem compatible with the high ambient temperature (it's pretty hot here too).

---

I did a search on r/NZXT and this really appears to be a firmware problem: [here's](https://www.reddit.com/r/NZXT/comments/hsmpcu/cam_not_displaying_liquid_temperature_pump_stuck/) one user reporting the same problem within CAM (and there a couple of cases that also appear to be related).

_(The difference between our 280.5°C and CAM's 0°C results is almost certainly only cosmetic and due to the different equations we use to convert the raw bytes; which in this case are `0xffff`)._

If the issue comes back you can also try updating to the new 2.1.0 firmware.  But you'll need CAM to do that.

monitorherocg, 2020-08-12:
Ok maybe I will update it at some point. Under load my water temps currently rise up to 46°C.

jonasmalacofilho, 2020-08-12:
> Is there a way to link the GPU temperature to the fans of the AIO?

There is, but not using the cooler as the controller, since (as far as I've been told and have been able to confirm) in these X3 models the fans can't even be connected to the pump block.

Depending on the motherboard/GPU combo you have you may be able to use do this is BIOS, for example with ASUS GPUs on ASUS motherboards.  Otherwise you'll need some software solution that runs constantly, supports both your GPU and your motherboard (or other fan controller) and periodically updates the fan speed.

monitorherocg, 2020-08-12:
Ok then I would need a special software for that I guess. Thank you for the help.

closed 2020-08-12 by jonasmalacofilho

basmith7, 2021-01-02:
I had the same problem and turn the computer off for a few minutes fixed it. 

This was the only resource I could find on the problem, so I wanted to post for posterity.

GitPullNow, 2022-08-03:
Just wanted to confirm that I to fixed it by turning the PC off and waiting for the little light on the NZXT Smart Device to turn off. It can be you don't use the Smart Device but it was a good indicator to see that power was really gone.

Confirmation that it had reset is also shown that the pump circle (i have it set to static white) now was turning circles on boot.

commit a48ab956 (2020-08-12): kraken3: warn if the temperature reading is faulty

Kraken X3 coolers, at least those with firmware 1.8.0, occasionally
report faulty temperature readings.  This has been reported with both
liquidctl and CAM.[1]

Warn if this happens and suggest a few immediate actions.  Additionally,
point the user to the (latest) issue where this is discussed.

Closes: #172 ("Wrong temperature readout X63")
Related: #160 ("Incorrect temperature reading from a Kraken X3 cooler")

[1] https://www.reddit.com/r/NZXT/comments/hsmpcu/cam_not_displaying_liquid_temperature_pump_stuck/