liquidctl/liquidctl #173: no response from liquidctl

markusmoormann, 2020-08-14:
I installed liquidctl today and at first had issues with OSError but could fix it with the help of [issue153](https://github.com/jonasmalacofilho/liquidctl/issues/153)

`#SUBSYSTEM=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c21", MODE="0666"
#SUBSYSTEM=="hidraw", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1b1c", ATTRS{idProduct}=="0c21", MODE="0666"
`
after a reboot everything worked fine. After some reboots I can't get any response from liquidctl. list returns the correct device. But status somehow seems to be stuck

```
% liquidctl list -g
[DEBUG] liquidctl.cli: running liquidctl v1.4.1
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 2ccf:0880
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:1939
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c21
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H115i PRO XT (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4041
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c318
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c318
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8140
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8140
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c318
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:3410
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c21
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0610
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:1939
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 2ccf:0880
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
Device ID 0: Corsair H115i PRO XT (experimental)
├── Vendor ID: 0x1b1c
├── Product ID: 0x0c21
├── Release number: 0x0100
├── Bus: hid
├── Address: /dev/hidraw5
└── Driver: HydroPlatinum using module hidraw
[DEBUG] liquidctl.cli: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice
```

```
% liquidctl status -g
[DEBUG] liquidctl.cli: running liquidctl v1.4.1
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 2ccf:0880
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:1939
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c21
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H115i PRO XT (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4041
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c318
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c318
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8140
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8140
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c318
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:3410
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c21
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0610
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:1939
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 2ccf:0880
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair H115i PRO XT (experimental)
[DEBUG] liquidctl.keyval: data in /run/user/1000/liquidctl/vid1b1c_pid0c21/loc5 (within XDG_RUNTIME_DIR)
[DEBUG] liquidctl.keyval: loaded sequence=4 (from /run/user/1000/liquidctl/vid1b1c_pid0c21/loc5/sequence)
[DEBUG] liquidctl.keyval: stored sequence=5 (in /run/user/1000/liquidctl/vid1b1c_pid0c21/loc5/sequence)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 3f:28:ff:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:d8
```

```
% lsusb -d 1b1c:0c21 -v

Bus 001 Device 006: ID 1b1c:0c21 Corsair 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c21 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
``` 

Any advices?

jonasmalacofilho, 2020-08-14:
Hi @markusmoormann,

Hm, I don't really know what's going on.

First, can you clarify whether you're running `initialize` after every reboot?

markusmoormann, 2020-08-14:
hi @jonasmalacofilho 

yes. But even that fails currently:
```
% liquidctl initialize all -g
[DEBUG] liquidctl.cli: running liquidctl v1.4.1
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 2ccf:0880
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:1939
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c21
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair H115i PRO XT (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4041
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c318
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c318
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8140
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8140
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c318
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:3410
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c21
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0610
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:1939
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 2ccf:0880
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: Corsair H115i PRO XT (experimental)
[DEBUG] liquidctl.keyval: data in /run/user/1000/liquidctl/vid1b1c_pid0c21/loc5 (within XDG_RUNTIME_DIR)
[DEBUG] liquidctl.keyval: stored pump_mode=1 (in /run/user/1000/liquidctl/vid1b1c_pid0c21/loc5/pump_mode)
[DEBUG] liquidctl.keyval: no data (file) found for fan1_mode
[DEBUG] liquidctl.driver.hydro_platinum: falling back to FIXED_DUTY for _FanMode(None)
[DEBUG] liquidctl.keyval: no data (file) found for fan1_duty
[INFO] liquidctl.driver.hydro_platinum: setting fan1 to 100% duty cycle
[DEBUG] liquidctl.keyval: no data (file) found for fan2_mode
[DEBUG] liquidctl.driver.hydro_platinum: falling back to FIXED_DUTY for _FanMode(None)
[DEBUG] liquidctl.keyval: no data (file) found for fan2_duty
[INFO] liquidctl.driver.hydro_platinum: setting fan2 to 100% duty cycle
[INFO] liquidctl.driver.hydro_platinum: setting pump mode to balanced
[DEBUG] liquidctl.keyval: no data (file) found for sequence
[DEBUG] liquidctl.keyval: stored sequence=1 (in /run/user/1000/liquidctl/vid1b1c_pid0c21/loc5/sequence)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x00 with 64 bytes: 3f:08:14:00:ff:05:ff:ff:ff:ff:ff:02:00:00:00:00:ff:02:00:00:00:00:ff:01:00:00:00:00:00:07:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:7f
```

markusmoormann, 2020-08-14:
I don't know why. But shutting the computer down and disconnecting power for 10 seconds solved the issue. Restarting didn't solve the issue. Maybe the microcontroller in the pump hung up, due to a wrong command?

jonasmalacofilho, 2020-08-14:
> I don't know why. But shutting the computer down and disconnecting power for 10 seconds solved the issue. Restarting didn't solve the issue.

I was just going to suggest this; completely powering down the device (which in some cases may even require turning off the PSU for a few minutes) does tend to fix problems like these (but not their root causes).

> Maybe the microcontroller in the pump hung up, due to a wrong command?

That's probably what happened, unfortunately.  Did you experiment with lighting commands before the cooler became unresponsive?

Could I also bother your for some captures of USB traffic between the device and iCue, while you cycle through a few cooling and lighting options?

> BTW: Thanks for implementing this piece of software. I really really like it

Thanks! : )

markusmoormann, 2020-08-14:
Yes, I was using some led commands which worked flawlessly apart from the notice that the commands would only work with the extra parameter. I can send you some USB captures next week.

jonasmalacofilho, 2020-08-14:
> which worked flawlessly

Glad to hear!

But it's possible that some sent (or missing) packet for LED control is what caused the cooler to hang.  Please let me know if this happens again, and if you spot any obvious relationship with previously executed actions.

> I can send you some USB captures next week.

That's fine, thanks!

jonasmalacofilho, 2020-08-14:
Note to myself: this could be related to the probably unnecessary¹ `_CMD_SET_LIGHTING2` packet clashing with some other protocol feature on PRO XT coolers.

https://github.com/jonasmalacofilho/liquidctl/blob/0ccb9d9887df7597087dd60114a8d4c9894cd133/liquidctl/driver/hydro_platinum.py#L263-L266

¹ Probably unnecessary because these coolers only have 16 LEDs, which fit in a single packet.

markusmoormann, 2020-08-17:
I couldn't reproduce the isssue. I remember that I first hat the fan commands in my .zshrc.local and I had two tabs of my terminal open. Maybe a parallel send command to pump got it stuck?
```
liquidctl initialize
liquidctl set fan speed  20 0  25 10  30 50  40 90  50 100
liquidctl set led color off
liquidctl set led color fixed ff8000
liquidctl set led color off         
liquidctl set fan speed  100                              
liquidctl status
liquidctl status
liquidctl status
liquidctl status
liquidctl set fan speed  20 0  25 10  30 50  40 90  50 100
liquidctl status
liquidctl set led color fixed "hsv(90,85,70)"
liquidctl set led color fixed "rgb(255, 127, 63)"
liquidctl set led color fading "hsv(0,80,100)" "hsv(180,80,100)" 
liquidctl set led color off                                     
```

[wireshark.tar.gz](https://github.com/jonasmalacofilho/liquidctl/files/5082331/wireshark.tar.gz)

jonasmalacofilho, 2020-08-17:
Thanks, but something went wrong and the capture file is entirely made out of nulls:

```
$ hexdump wireshark.pcapng 
0000000 0000 0000 0000 0000 0000 0000 0000 0000
*
09e9c50 0000 0000                              
09e9c54
```

---

> Maybe a parallel send command to pump got it stuck?

It's possible.  Perhaps a different command between `_CMD_SET_LIGHTING1` and `_CMD_SET_LIGHTING2`, or perhaps simply because of the packet sequence order getting messed up.

It would be interesting to know if it's possible to reproduce the problem with multiple simultaneous commands of all types, and then if it's still possible to reproduce it with all _but_ lighting commands.

markusmoormann, 2020-08-18:
I'll give it a try and try to reproduce it. Can't promise any eta.

jonasmalacofilho, 2020-08-18:
That's fine.  But please don't forgot to check/try to fix the all-nulls capture, as that's likely to reveal important information.

jonasmalacofilho, 2020-08-28:
Hi @markusmoormann, any news on your end?

markusmoormann, 2020-08-28:
Hi @jonasmalacofilho,
thanks for the reminder ;-)
I tried to export a new dump, but couldn't really export the needed part. I think, this now contains everything.
Unfortunately I could not reproduce the issue yet

[wireshark.zip](https://github.com/jonasmalacofilho/liquidctl/files/5140195/wireshark.zip)

closed 2020-12-10 by jonasmalacofilho