liquidctl/liquidctl #174: H80i V2 Fan Speed shows only above ~1600

andrefchaves, 2020-08-17:
My fans speed are most of the time 0rpm but it shows when reaches >1600
I'm able to change the lighting but fans and pump speeds don't seem to be applied to the AIO. I'm not getting any errors or anything so I don't know where to start looking. I've installed the WinUSB driver through windows and even there I'm getting these same results.

this is my status --debug result:
```
liquidctl -v status -g
[DEBUG] liquidctl.cli: running liquidctl v1.4.1 (homebrew; liquidctl 1.4.1)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c077
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c077
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:026c
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c08
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair Hydro H80i v2
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c077
[DEBUG] liquidctl.driver.usb: probing drivers for device 0bc2:231a
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.cli: device: Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[INFO] liquidctl.driver.rgb_fusion2: Status reports not available from Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.cli: device: Corsair Hydro H80i v2
[DEBUG] liquidctl.driver.usb: selected interface: 0
[DEBUG] liquidctl.driver.asetek: set clear to send = True
[DEBUG] liquidctl.driver.usb: sending control transfer with (64, 2, 2), {}
[DEBUG] liquidctl.driver.asetek: begin transaction
[DEBUG] liquidctl.driver.usb: explicitly claim interface
[DEBUG] liquidctl.driver.usb: sending control transfer with (64, 2, 1), {}
[DEBUG] liquidctl.driver.usb: writting 4 bytes: 14:00:00:00
[DEBUG] liquidctl.driver.usb: read 32 bytes: 00:00:01:32:00:06:00:9f:04:92:1d:14:00:00:05:f9:00:22:10:05:00:ff:09:02:08:00:00:00:01:8f:11:43
[DEBUG] liquidctl.driver.usb: ensure interface is released
Corsair Hydro H80i v2
├── Liquid temperature       29.5  °C
├── Fan speed                   0  rpm
├── Pump speed               1170  rpm
└── Firmware version      2.8.0.0 
```

jonasmalacofilho, 2020-08-18:
Opa, tudo bem @andrefchaves ?

The fan and pump don't respond to configuration changes, but the fan still ramps up sometimes, reaching 1600+ rpm?  Is this due to a (perhaps default) fan/temperature profile?

Another thing, are you able to run iCue (or CorsairLink) on Windows and see if both (fan speed and fan/pump control) issues still reproduce when using it?

andrefchaves, 2020-08-18:
Oi Jonas, tudo ótimo e muito obrigado pelo seu tempo desenvolvendo essas ferramentas!

Yes, the fans and the pump ramp up and go full speed during full load. They're definitely working and cooling properly. It could be a default fan/temp profile but I really don't know where it could be coming from... I've already set the cpu_fan in bios to full speed and the fans settings in iStats are set to System Controlled.

The pump in iCue was running full speed and the fans had an Extreme curve profile before I installed the WinUSB driver but since then I've lost control of the AIO through iCue/Link. It's not finding any devices.

The pump speed and liquid temperature seem to be displaying correctly as they are fluctuating through time but the fan speed is at 0 rpm. As a reference, through another monitoring app the fans are at 664 rpm

andrefchaves, 2020-08-18:
One thing that I've noticed is that apparently the set fan/pump speed is working but with offset values maybe? or it is controlling the speed on top of a prior profile, is that possible?
If I set the pump speed to 100%, it stays around 1350 in idle but if I set it to 50% it goes down to 730. When in a full load, the pump speed ramps up to 2880. I believe that with the fan speed is occurring something similar but haven't been able to find applied values that I could use as a comparison. I think if I change the fan speeds during a full load I might be able to see some results

andrefchaves, 2020-08-18:
I'm so sorry, my bios got misconfigured during troubleshooting. Everything is working as expected now :)

jonasmalacofilho, 2020-08-18:
Oh, right, varying voltage would cause some very weird and fun issues : )

closed 2020-08-18 by jonasmalacofilho