liquidctl/liquidctl #175: Corsair H115i PRO XT is not detected on Linux

lbrande, 2020-08-19:
I also have a Corsair RM850i which is detected correctly. I'm running Ubuntu Linux 20.04 and installed liquidctl from PyPi. To make anything work at all I installed the suggested udev rules file.

jonasmalacofilho, 2020-08-19:
Hi @lbrande ,

Can you show me the output of `lsusb -tv` and `liquidctl list -g`?

lbrande, 2020-08-19:
Output of `lsusb -tv`:
```bash
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 6: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0610 Genesys Logic, Inc. 4-port hub
        |__ Port 4: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 0b05:18a3 ASUSTek Computer, Inc. 
        |__ Port 2: Dev 4, If 0, Class=Vendor Specific Class, Driver=, 12M
            ID 1b1c:0c13 Corsair 
    |__ Port 7: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        ID 04d9:4545 Holtek Semiconductor, Inc. Keyboard [Diatec Majestouch 2 Tenkeyless]
    |__ Port 7: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        ID 04d9:4545 Holtek Semiconductor, Inc. Keyboard [Diatec Majestouch 2 Tenkeyless]
    |__ Port 8: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1532:0062 Razer USA, Ltd 
    |__ Port 8: Dev 5, If 2, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1532:0062 Razer USA, Ltd 
    |__ Port 8: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1532:0062 Razer USA, Ltd 
    |__ Port 11: Dev 7, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1b1c:1c0c Corsair RM850i Power Supply
    |__ Port 13: Dev 8, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0606 Genesys Logic, Inc. USB 2.0 Hub / D-Link DUB-H4 USB 2.0 Hub
        |__ Port 1: Dev 10, If 2, Class=Audio, Driver=snd-usb-audio, 480M
            ID 046d:085c Logitech, Inc. C922 Pro Stream Webcam
        |__ Port 1: Dev 10, If 0, Class=Video, Driver=uvcvideo, 480M
            ID 046d:085c Logitech, Inc. C922 Pro Stream Webcam
        |__ Port 1: Dev 10, If 3, Class=Audio, Driver=snd-usb-audio, 480M
            ID 046d:085c Logitech, Inc. C922 Pro Stream Webcam
        |__ Port 1: Dev 10, If 1, Class=Video, Driver=uvcvideo, 480M
            ID 046d:085c Logitech, Inc. C922 Pro Stream Webcam
    |__ Port 14: Dev 9, If 0, Class=Audio, Driver=snd-usb-audio, 12M
        ID 054c:09cc Sony Corp. DualShock 4 [CUH-ZCT2x]
    |__ Port 14: Dev 9, If 1, Class=Audio, Driver=snd-usb-audio, 12M
        ID 054c:09cc Sony Corp. DualShock 4 [CUH-ZCT2x]
    |__ Port 14: Dev 9, If 2, Class=Audio, Driver=snd-usb-audio, 12M
        ID 054c:09cc Sony Corp. DualShock 4 [CUH-ZCT2x]
    |__ Port 14: Dev 9, If 3, Class=Human Interface Device, Driver=usbhid, 12M
        ID 054c:09cc Sony Corp. DualShock 4 [CUH-ZCT2x]
```

Output of `liquidtcl list -g`:
```bash
[DEBUG] liquidctl.cli: running liquidctl v1.4.1
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1c0c
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair RM850i
[DEBUG] liquidctl.driver.usb: probing drivers for device 054c:09cc
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 04d9:4545
[DEBUG] liquidctl.driver.usb: probing drivers for device 04d9:4545
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0062
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0062
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0062
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1532:0062
[DEBUG] liquidctl.driver.usb: probing drivers for device 04d9:4545
[DEBUG] liquidctl.driver.usb: probing drivers for device 0b05:18a3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c13
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0610
[DEBUG] liquidctl.driver.usb: probing drivers for device 054c:09cc
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:085c
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0606
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1c0c
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
Device ID 0: Corsair RM850i
├── Vendor ID: 0x1b1c
├── Product ID: 0x1c0c
├── Release number: 0x0002
├── Bus: hid
├── Address: /dev/hidraw6
└── Driver: CorsairHidPsu using module hidraw
[DEBUG] liquidctl.cli: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice
```

jonasmalacofilho, 2020-08-20:
The kernel driver appears to have been detached from the cooler.

```
        |__ Port 2: Dev 4, If 0, Class=Vendor Specific Class, Driver=, 12M
            ID 1b1c:0c13 Corsair 
```

This is [usually](https://github.com/jonasmalacofilho/liquidctl#device-not-listed-linux) caused by programs that use libusb or hidapi-libusb.

It's also possible that the kernel was unable to bind a driver to the device due to some other problem; you can check this with `dmesg`.

lbrande, 2020-08-20:
`dmesg` shows the following lines for my H115i PRO:
```bash
[    1.482207] usb 1-6.2: Product: H115i Platinum
[    1.482208] usb 1-6.2: Manufacturer: Corsair
[    1.482208] usb 1-6.2: SerialNumber: 7289_2.0
```

I'm a bit unsure on what numbers from `lsusb -tv` I should substitute \<bus\> and \<port\> for in the guide for **Device not listed (Linux)**.

lbrande, 2020-08-20:
When running `echo -n "1-6.2:1.0" > sudo /sys/bus/usb/drivers/usbhid/bind` nothing seems to happen. I also tried first running `echo -n "1-6.2:1.0" > sudo /sys/bus/usb/drivers/usb/unbind` but that didn't work either.

jonasmalacofilho, 2020-08-20:
Your command would write to a file called `sudo`, which isn't what you want.  But `1-6.2` should be the correct bus/port spec, so the following should work:

```
echo '1-6.2:1.0' | sudo tee /sys/bus/usb/drivers/usbhid/bind
```

This is just a temporary fix though, and you probably want to try to figure out what program is detaching the kernel driver.

As for the output of dmesg, this shows that there's no kernel-side problem.  But the kernel logs don't show (at least in non-debug mode) when the kernel driver is detached _unless_ that was done by the kernel itself, for example because there were connection issues.

jonasmalacofilho, 2020-08-20:
Can you also show me the output of `lsusb -vd 1b1c:0c13`?
_(this could show if the device is stuck is firmware update mode, which would be another reason for a detached `usbhid`)_

lbrande, 2020-08-20:
Executing `echo '1-6.2:1.0' | sudo tee /sys/bus/usb/drivers/usbhid/bind` outputs:
`tee: /sys/bus/usb/drivers/usb/unbind: No such device`

Output from `lsusb -vd 1b1c:0c13`:
```bash
Bus 001 Device 004: ID 1b1c:0c13 Corsair 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c13 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower               64mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
```

jonasmalacofilho, 2020-08-20:
This is wrong:

```
      bInterfaceClass       255 Vendor Specific Class
```

Either your device is messed up... or it isn't a H115i PRO **XT**.

Yeah, it's the latter, please see #137.

closed 2020-08-20 by jonasmalacofilho

lbrande, 2020-08-20:
Thanks for the thorough investigation.