liquidctl/liquidctl #177: Corsiar H100i Platinum not being detected

GhostCode64, 2020-08-25:
liquidctl init output:
```
➜  ~ sudo liquidctl initialize       
Error: no devices matches available drivers and selection criteria
```

lsusb:
```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 046d:c336 Logitech, Inc. Gaming Keyboard G213
Bus 001 Device 004: ID 046d:c084 Logitech, Inc. G203 Gaming Mouse
Bus 001 Device 003: ID 1b1c:0c15 Corsair H100i Platinum
Bus 001 Device 002: ID 1b1c:0a56 Corsair Corsair VOID ELITE USB Gaming Headset
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

lsusb -vt:
```
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 4: Dev 2, If 3, Class=Human Interface Device, Driver=usbhid, 12M
        ID 1b1c:0a56 Corsair 
    |__ Port 4: Dev 2, If 1, Class=Audio, Driver=snd-usb-audio, 12M
        ID 1b1c:0a56 Corsair 
    |__ Port 4: Dev 2, If 2, Class=Audio, Driver=snd-usb-audio, 12M
        ID 1b1c:0a56 Corsair 
    |__ Port 4: Dev 2, If 0, Class=Audio, Driver=snd-usb-audio, 12M
        ID 1b1c:0a56 Corsair 
    |__ Port 5: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 12M
        ID 1b1c:0c15 Corsair 
    |__ Port 8: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        ID 046d:c084 Logitech, Inc. G203 Gaming Mouse
    |__ Port 8: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 046d:c084 Logitech, Inc. G203 Gaming Mouse
    |__ Port 9: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 046d:c336 Logitech, Inc. 
    |__ Port 9: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        ID 046d:c336 Logitech, Inc. 
```

liquidctl init -g:
```
[DEBUG] liquidctl.cli: running liquidctl v1.4.1 (Arch User Repository; liquidctl 1.4.1-1)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c084
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c084
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c336
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c336
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c336
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c084
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c15
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0a56
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
Error: no devices matches available drivers and selection criteria
```

All liquidctl commands ran as sudo
Not sure if this is related to #136 , I thought I'd make a ticket to be sure. if it is, no harm no foul just wanted to make sure I'm not going crazy

jonasmalacofilho, 2020-08-25:
We use the names that are on the box, not the USB product descriptor.  The produce descriptor can be, to put it mildly, very inconsistent.

So, (to us) that's a H100i PRO.  Take a look at #137, @jlask is working on a driver for it.

closed 2020-08-25 by jonasmalacofilho

GhostCode64, 2020-08-25:
Thank you, will follow the progress. For now I'll look into setting my fan speed in windows if they get reset (which happens sometimes).

Just a curious note, OpenCorsairLink used to be able to detect my AIO, but not anymore. Irrelevant anyway I suppose, that software is discontinued. Thanks again and keep up the good work!