liquidctl/liquidctl #178: Asus Aora Support

gusdrawn, 2020-08-28:
Hi,

You have great project, the only device to complete all support for my devices is Asus Aura, I see to support you need capture when use the device, but I dont have idea. I believe when  I install Wireshark could check to capture USB devices, change color to my asus graphics card y so on.

If yes, tell me, I could be try that.

jonasmalacofilho, 2020-08-28:
Hi Gustavo,

I'm very curious about how Aura works on compatible graphics cards.  Do you have any obvious (Asus or ITE) USB device listed on your OS for the card?

That said, a lot of Aura stuff uses SMBus, not USB, and there's a big chance Asus graphics cards do this too.

---

P.S. You should also check out [OpenRGB](https://gitlab.com/CalcProgrammer1/OpenRGB).

jonasmalacofilho, 2020-09-14:
Did you find anything out?

By the way, you could just try to capture the traffic.  For that [Capturing USB traffic](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/developer/capturing-usb-traffic.md) may help.

gusdrawn, 2020-09-14:
Yes, You are right, I mistake because I believe you need to capture someone traffic to support AURA but the OpenRGB its OK and work for me, but its another app to controller only VGA, the principal difference your app and OpenRGB for me, is liquidctl control the Gigabyte and NZXT AIO and OpenRGB is only to AORA Graphics.

Another Idea is liquidctl check the color from Logitech Hub and copy&paster color to all devices from liquidctl.

jonasmalacofilho, 2020-09-15:
I just checked the source code of OpenRGB and it seems that Aura GPUs are controlled through SMBus, not USB, and I'm not currently looking into supporting non-USB devices.

> Another Idea is liquidctl check the color from Logitech Hub and copy&paster color to all devices from liquidctl.

This would be fairly specific to Windows + Logitech Hub, and probably even some specific devices (to read the Logitech settings from).

So I don't think it really fits into liquidctl, but (the liquidctl part) would be be fairly simple for a user to implement as a Python script that calls the liquidctl APIs.  Of course, someone interested in this would need to figure out how to extract the settings from Logitech Hub...

---

Closing this issue because the specific Aura device it's about (a GPU) uses SMBus, and I'm not looking into this type of device at the moment.  Still, please feel free to ask more questions or continue the discussion.

closed 2020-09-15 by jonasmalacofilho