liquidctl/liquidctl #182: OSError: open failed on NZXT Smart Device (V1) on my hackintosh

kingo132, 2020-09-08:
My OS: macOS Catalina Version 10.15.6
My PC case: NZXT h200i

There is a Smart Device used to control the fan and led in h200i case, it's called: NZXT Smart Device (V1).
The liquidctl command failed to read NZXT Smart Device (V1) and shows: OSError: open failed
However, if I switch to Windows 10 and open CAM one time and then reboot to mac os. Then everything works, liquidctl can read and control this Smart Device without any problem.
However, if I power off and then power on the computer. Then liquidctl fails to read again. Every time I boot my computer, I have to boot to Windows 10 first to open CAM and then reboot to mac os to use liquidctl. Otherwise, liquidctl will fail.

Here is the output of liquidctl when everything works:
NZXT Kraken X (X42, X52, X62 or X72)
├── Liquid temperature     36.7  °C
├── Fan speed               516  rpm
├── Pump speed             2148  rpm
└── Firmware version      6.0.2  

NZXT E850 (experimental)
├── Temperature                                    46.0  °C
├── Fan speed                                         0  rpm
├── Firmware version                         A017/40983  
├── +12V peripherals output voltage               11.91  V
├── +12V peripherals output current                3.55  A
├── +12V peripherals output power                 42.25  W
├── +12V EPS/ATX12V output voltage                11.98  V
├── +12V EPS/ATX12V output current                 2.28  A
├── +12V EPS/ATX12V output power                  27.31  W
├── +12V motherboard/PCI-e output voltage         11.93  V
├── +12V motherboard/PCI-e output current          1.01  A
├── +12V motherboard/PCI-e output power           12.02  W
├── +5V combined output voltage                    4.88  V
├── +5V combined output current                    3.45  A
├── +5V combined output power                     16.81  W
├── +3.3V combined output voltage                  3.11  V
├── +3.3V combined output current                  1.38  A
└── +3.3V combined output power                    4.27  W

NZXT Smart Device (V1)
├── Fan 1                          —  
├── Fan 2                        PWM  
├── Fan 2 current               0.01  A
├── Fan 2 speed                 1054  rpm
├── Fan 2 voltage              12.04  V
├── Fan 3                        PWM  
├── Fan 3 current               0.03  A
├── Fan 3 speed                 1657  rpm
├── Fan 3 voltage              12.04  V
├── Firmware version           1.0.7  
├── LED accessories                1  
├── LED accessory type    HUE+ Strip  
├── LED count (total)             10  
└── Noise level                   57  dB

And here is the output when failed to read NZXT Smart Device  (V1)
NZXT Kraken X (X42, X52, X62 or X72)
├── Liquid temperature     36.7  °C
├── Fan speed               516  rpm
├── Pump speed             2148  rpm
└── Firmware version      6.0.2  

OSError: open failed

jonasmalacofilho, 2020-09-08:
Can you check whether `liquidctl initialize -m smart --debug` also fails on Windows (before opening CAM)?  Can you also provide the `--debug` output for the same command on mac OS (you only showed it for `status`, not `initialize`)?

---

I have this device, with the same firmware version as yours, and cannot reproduce the problem on both Windows and Linux.  So it's either a mac-only problem, or a problem with your device.

In the former case it could be that it and mac OS aren't playing nice during the initialization of the USB or HID kernel device, which could be affected by your specific USB setup (I know this is one of the tricky parts of setting up a hackintosh).  Maybe you could also try other USB headers or ports.

robbins, 2020-09-08:
On Linux I had to run the initialization command as root, otherwise I would get the same OSError.

kingo132, 2020-09-09:
> liquidctl initialize -m smart --debug

Here is my test result
### 1. Runing "liquidctl initialize -m smart --debug" on mac when everything works fine
jason@myhm ~ % liquidctl initialize -m smart --debu
[DEBUG] liquidctl.cli: running liquidctl v1.4.1 (homebrew; liquidctl 1.4.1)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 7793:2500
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:1714
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device (V1)
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 04b4:120d
[DEBUG] liquidctl.driver.usb: probing drivers for device 04b4:120d
[DEBUG] liquidctl.driver.usb: probing drivers for device 04b4:120d
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:1714
[DEBUG] liquidctl.driver.usb: probing drivers for device 7793:2500
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 04b4:120d
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.cli: device: NZXT Smart Device (V1)
[DEBUG] liquidctl.driver.usb: writting report 0x01 with 64 bytes: 5c:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x01 with 64 bytes: 5d:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00

### 2. Runing "liquidctl initialize -m smart --debug" before CAM start in Windows 10 when everything works fine on mac
PS C:\Users\wangg> E:\winsoft\liquidctl-1.4.1-bin-windows-x86_64\liquidctl-1.4.1-bin-windows-x86_64\liquidctl.exe initialize -m smart --debug
[DEBUG] __main__: running liquidctl v1.4.1 (AppVeyor; liquidctl-1.4.1-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 04b4:120d
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:1714
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device (V1)
[DEBUG] liquidctl.driver.usb: probing drivers for device 7793:2500
[DEBUG] liquidctl.driver.usb: probing drivers for device 04b4:120d
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 04b4:120d
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 04b4:120d
[DEBUG] liquidctl.driver.usb: probing drivers for device 7793:2500
[DEBUG] liquidctl.driver.usb: probing drivers for device 0a5c:4500
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:1714
[DEBUG] liquidctl.driver.usb: probing drivers for device 05ac:828d
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c53f
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 8086:a36d
[DEBUG] __main__: device: NZXT Smart Device (V1)
[DEBUG] liquidctl.driver.usb: writting report 0x01 with 64 bytes: 5c:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[... cut ...]
├── +5V combined output current                    2.11  A
├── +5V combined output power                     10.30  W
├── +3.3V combined output voltage                  3.16  V
├── +3.3V combined output current                  1.37  A
└── +3.3V combined output power                    4.32  W

### 4. Have not opened CAM, just reboot computer and log into mac. Everything works fine.
### 5. Turn off the computer, unplug power cable, wait NZXT Smart Device led indicator turned off, power on computer and directly log into mac. 
Failed with OSError again.

jonasmalacofilho, 2020-09-09:
Hi again @kingo132,

Can you try the following test?

- cold boot¹ into Windows
- _don't_ run `liquidctl` or CAM
- reboot to mac OS and see if `liquidctl initialize` works then
- _if it doesn't, try runing `liquidctl status` on Windows (which will hang, but that's fine; still _don't_ run `liquidctl initialize` [or CAM]) and rebooting into mac again_

This would tell me whether it is liquidctl or Windows that is making the device "acceptable" to mac OS.

You also mentioned there is only one USB 2.0 header on that board.  But can you try connecting the device to an external USB port (not from the NZXT Hub), or even another mac?

Finally, @CaseySJ, do you have any other tips on how to fix (or at least debug) a problem like this?

---

P.S. I'll check if there haven't been any mac-specific changes to Hidapi that could explain this.

¹ As you've been already doing, not only cold boot but also wait for the Smart Device LED indicator to turn off.

CaseySJ, 2020-09-09:
@kingo132 

Let's get some additional background information:

- We know you're running Catalina 10.15.6.
- Which motherboard are you using?
- Which CPU?
- Are you booting using OpenCore or Clover?
- Are you configuring your USB ports using "USBPorts.kext" or USBInjectAll (and SSDT-UIAC.aml)?
- Are you running any malware checkers, virus checkers, etc.?
- Do you have any experience using "WireShark" for macOS to capture USB traffic?
- Please download and run "IORegistryExplorer" and select "File --> Save As..." to save the IOReg dump. Then post that file. It will show us the USB configuration.
- What is your USB configuration in BIOS? Do you have XHC Handoff enabled? 

Not all of the above items are directly relevant to solving the problem, but they do provide a better context.

kingo132, 2020-09-09:
> * We know you're running Catalina 10.15.6.
> * Which motherboard are you using?
> * Which CPU?
> * Are you booting using OpenCore or Clover?
> * Are you configuring your USB ports using "USBPorts.kext" or [...]

1. Which motherboard are you using?
Asrock Z390 Phantom Gaming Itx
2. Which CPU?
i9-9900k
3. Are you booting using OpenCore or Clover?
Clover
4. Are you configuring your USB ports using "USBPorts.kext" or USBInjectAll (and SSDT-UIAC.aml)?
Yes, I configured the USB using hackintool and exported SSDT-EC-USBX.aml and SSDT-UIAC.aml, and I putted them in Clover/ACPI/patched
5. Are you running any malware checkers, virus checkers, etc.?
No
6. Do you have any experience using "WireShark" for macOS to capture USB traffic?
Yes, I have some knowledge. I can have a try.
7. Please download and run "IORegistryExplorer" and select "File --> Save As..." to save the IOReg dump. Then post that file. It will show us the USB configuration.
[myhm.ioreg.zip](https://github.com/jonasmalacofilho/liquidctl/files/5192486/myhm.ioreg.zip)
8. What is your USB configuration in BIOS? Do you have XHC Handoff enabled?
I only enabled XHC Handoff in BIOS, other USB configurations are all default.

kingo132, 2020-09-09:
> Hi again @kingo132,
> 
> Thanks!
> 
> Can you try the following additional test?
> 
> * cold boot¹ into Windows
> * _don't_ run `liquidctl` or CAM
> * reboot to mac OS and see if `liquidctl initialize` works then
> * _if it doesn't, try runing `liquidctl status` on Windows (which will hang, but that's fine; still _don't_ run `liquidctl initialize` [or CAM]) and rebooting into mac again_
> 
> This would tell me whether it is liquidctl or Windows that is making the device "acceptable" to mac OS.
> 
> You also mentioned there is only one USB 2.0 header on that board. But can you try connecting the device to an external USB port (not from the NZXT Hub), or even another mac?
> 
> Finally, @CaseySJ, do you have any other tips on how to fix (or at least debug) a problem like this?
> 
> P.S. I'll check if there haven't been any mac-specific changes to Hidapi that could explain this.
> 
> ¹ As you've been already doing, not only cold boot but also wait for the Smart Device LED indicator to turn off.

1. Cold boot to windows, do not run anything, reboot to mac
Still OSError
2. Cold boot to windows, just run liquidctl status, hang, close Powershell, reboot to mac
Works ok, no OSError
3. Connect NZXT Smart Device to an external USB
I don't have 9pin to USB-A connector, I'll try to make one and post the result later

kingo132, 2020-09-09:
> Hi again @kingo132,
>
> Thanks!
>
> Can you try the following additional test?
>
> * cold boot¹ into Windows
> * _don't_ run `liquidctl` or CAM
> * reboot to mac OS and see if `liquidctl initialize` works then
> * [...]

I made an ugly 9pin to USB-A cable. I tried several external USB ports, it still failed with OSError.

CaseySJ, 2020-09-09:
Two immediate issues/suggestions are:

1. Too many USB ports have been defined. There's a limit of 15 ports per USB controller, but you have 19.
2. Google Chrome is attached to nearly all of your USB ports. We encountered this issue before. It may be a good idea to fully delete Google Chrome, reboot, run IORegistryExplorer once again, and check if those entries in blue are gone.

[image]
[image]

kingo132, 2020-09-11:
> Two immediate issues/suggestions are:
>
> 1. Too many USB ports have been defined. There's a limit of 15 ports per USB controller, but you have 19.
> 2. Google Chrome is attached to nearly all of your [...]

I uninstalled Chrome using CleanMyMac. These entries are gone. However, liquidctl still got OSError.
Here are the screenshot and IORegistryExplorer dump.
[image]
[myhm.ioreg.zip](https://github.com/jonasmalacofilho/liquidctl/files/5205470/myhm.ioreg.zip)

CaseySJ, 2020-09-12:
@kingo132 

Your system still has 19 USB ports defined instead of 15. Please try temporarily disabling 4 USB ports such as SS01, SS02, SS03, and SS04.

I would like to rule this out as a possible source of error.

kingo132, 2020-09-13:
> @kingo132
> 
> Your system still has 19 USB ports defined instead of 15. Please try temporarily disabling 4 USB ports such as SS01, SS02, SS03, and SS04.
> 
> I would like to rule this out as a possible source of error.

I'll try to minimize USB port number under 15 to test if it works. 
However, I have applied the port number limit unlock patch using Clover Configurator. I don't have problems using these 19 USB ports with other APPs.

jonasmalacofilho, 2020-09-13:
I reviewed exactly what would be sent to the device in last test scenario (`liquidctl status` on Windows, uninitialized device), and I still can't find anything that would explain why mac OS can't normally open the device.[1]

Can you provide a Wireshark capture on mac OS of the traffic between an uninitialized (=from cold boot of the device) execution of `liquidctl status` and the device?

Another thing I noticed in your IORegistryExplorer screenshots (I don't have a tool to open the report itself on Linux) is that one NZXT device is attached to `IOHIDInterface` and  `IOHIDLibUserClient`, while the other only to the former.

The relevant screenshot doesn't show all devices, and I also don't know if the device paths haven't changed since the first post.  Can you show this information for all the NZXT devices (including the hub), as well as marking which one is which?

[1] Capture file added to liquidctl-device-data in jonasmalacofilho/liquidctl-device-data@24bf0f11bbab .

jonasmalacofilho, 2020-09-13:
The HID report descriptor looks ok as well:

```
$ usbhid-dump -a 1:4
001:004:000:DESCRIPTOR         1600002923.244703
 06 00 FF 09 01 A1 01 85 01 09 01 15 00 26 FF 00
 75 08 95 01 B1 82 85 01 09 01 91 82 85 02 09 02
 15 00 26 FF 00 75 08 95 40 B1 82 85 02 09 02 91
 82 85 03 09 03 15 00 26 FF 00 75 08 95 40 B1 82
 85 03 09 03 91 82 85 04 09 04 75 08 95 14 81 82
 85 05 09 05 75 08 95 01 81 82 C0
```

decoded:

```
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x09, 0x01,        //   Usage (0x01)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x01,        //   Report Count (1)
0xB1, 0x82,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x01,        //   Report ID (1)
0x09, 0x01,        //   Usage (0x01)
0x91, 0x82,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x02,        //   Report ID (2)
0x09, 0x02,        //   Usage (0x02)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x40,        //   Report Count (64)
0xB1, 0x82,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x02,        //   Report ID (2)
0x09, 0x02,        //   Usage (0x02)
0x91, 0x82,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x03,        //   Report ID (3)
0x09, 0x03,        //   Usage (0x03)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x75, 0x08,        //   Report Size (8)
0x95, 0x40,        //   Report Count (64)
0xB1, 0x82,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x03,        //   Report ID (3)
0x09, 0x03,        //   Usage (0x03)
0x91, 0x82,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Volatile)
0x85, 0x04,        //   Report ID (4)
0x09, 0x04,        //   Usage (0x04)
0x75, 0x08,        //   Report Size (8)
0x95, 0x14,        //   Report Count (20)
0x81, 0x82,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x85, 0x05,        //   Report ID (5)
0x09, 0x05,        //   Usage (0x05)
0x75, 0x08,        //   Report Size (8)
0x95, 0x01,        //   Report Count (1)
0x81, 0x82,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection

// 91 bytes
```

jonasmalacofilho, 2020-10-25:
@kingo132, can you provide any more feedback?

closed 2020-10-31 by jonasmalacofilho