liquidctl/liquidctl #184: Gigabyte FI27G-P with RGB backlighting.

sgtsixpack, 2020-09-13:
I wonder if CaseySJ would support this with his fork?  I can do some wireshark data collecting.

jonasmalacofilho, 2020-09-13:
Is there an USB device associated with it?

If so, can you list the vendor (VID) and product (PID) IDs?  On Windows you can find them in driver details -> hardware IDs.

sgtsixpack, 2020-09-13:
Looked at the device manager and found MMDEVAPI\AudioEndpoints (value) for Hardware ID of an AMD high definition audio device in brackets next to the name of the monitor.

Device SWD\MMDEVAPI\{0.0.0.00000000}.{8ead5960-7199-4eb2-b1f4-787a385cddb4} was configured.

Driver Name: audioendpoint.inf
Class Guid: {c166523c-fe0c-4a94-a586-f1a80cfbbf3e}
Driver Date: 12/06/2019
Driver Version: 10.0.19041.1
Driver Provider: Microsoft
Driver Section: NO_DRV
Driver Rank: 0xFF0000
Matching Device Id: MMDEVAPI\AudioEndpoints
Outranked Drivers: c_swdevice.inf:SWD\GenericRaw:00FF3001
Device Updated: false
Parent Device: HDAUDIO\FUNC_01&VEN_1002&DEV_AA01&SUBSYS_00AA0100&REV_1007\7&1c799c31&0&0001

Device SWD\MMDEVAPI\{0.0.0.00000000}.{8ead5960-7199-4eb2-b1f4-787a385cddb4} was started.

Driver Name: audioendpoint.inf
Class Guid: {c166523c-fe0c-4a94-a586-f1a80cfbbf3e}
Service: 
Lower Filters: 
Upper Filters: 

Also have a generic PnP monitor in device manager, I new had an drivers to change that.  Hardware IDs  monitor\GBT2707.

Device DISPLAY\GBT2707\7&25f76f65&2&UID264 was configured.

Driver Name: monitor.inf
Class Guid: {4d36e96e-e325-11ce-bfc1-08002be10318}
Driver Date: 06/21/2006
Driver Version: 10.0.19041.1
Driver Provider: Microsoft
Driver Section: PnPMonitor.Install
Driver Rank: 0xFF2000
Matching Device Id: *PNP09FF
Outranked Drivers: 
Device Updated: false
Parent Device: PCI\VEN_1002&DEV_687F&SUBSYS_6B761002&REV_C3\6&2d57b13c&0&00000019

Device DISPLAY\GBT2707\7&25f76f65&2&UID264 was started.

Driver Name: monitor.inf
Class Guid: {4d36e96e-e325-11ce-bfc1-08002be10318}
Service: monitor
Lower Filters: 
Upper Filters: 

I could not find an obvious source of control of rgb in device manager or devices and printers.

sgtsixpack, 2020-11-03:
Did I give the correct info?

I verified by pulling the cable out, this is the correctly identified device and I believe the corresponding USB hardware IDs:

https://1drv.ms/u/s!AhCKpFOJjf6jgjd9BU2Km2lt9npK?e=8YAM4p

jonasmalacofilho, 2020-11-03:
Sorry for the delay, I had planned to post more detailed instructions but never got around to booting Windows (to prepare them).

---

This device (`0bda:1100`) announces itself as a Realtek device, which is a bit surprising.  The thing is, disconnecting the cable to a modern monitor disconnects a whole bunch of things, and I would also be surprised if the only USB device embedded in this monitor is the LED controller.

At this point I think it would help to look at a complete tree of all the USB devices in the system, as well as all of their string descriptors.  But I have no idea how to get this on Windows... there's some tool in some Windows SDK that does this, I just can't remember the specifics.

Maybe someone else can help.  For reference, on Linux I would start with the outputs of:

```
# for the tree
lsusb -t

# for the tree plus driver bindings
lsusb -tv

# for all string descriptors
sudo lsusb
```

sgtsixpack, 2020-11-03:
https://1drv.ms/u/s!AhCKpFOJjf6jgj90tIshEmgAFUZD?e=MxZJOV

I used HWiNFO & USB device tree viewer.  The device has a generic unidentifiable name so I have included a few possibilities (not including the printer) in the screenshots.  If you need the extended info from USB device tree viewer I can copy and paste for that particular device.  I think waiting for your response is better than posting all of the details for each device at once.

jonasmalacofilho, 2020-11-03:
Thanks!

On the same hub you have:

- `0bda:1100` this vendor-defined HID from Realtek with `PID=F192` in the name
- `0d8c:01c2`: a C-Media audio device
- `048d:8295`: another vendor-defined HID from ITE

Given that the hub is probably in the monitor, that the `F192` string has been previously used in Realtek network controllers, and that Gigabyte already uses LED controllers from ITE (8297, 5702)... I'm thinking we're actually interested in this `048d:8295` device.

Can you do some initial traffic captures?

sgtsixpack, 2020-11-03:
I think I need a reboot and to acquire the device again (I guess same string) because I've been running into issues with hard drives being differently enumerated and also when I change configuration in HWiNFO it seems to never do any garbage control.  The author wanted me to reset user preferences but then what is the point in the user back preferences?  I went through and cleaned old entries.  I think I've only got 1 bug right now where one of the rows on my G15 will not show (possibly have a conflict).

What I'm saying is that I want to be sure to provide valid data.  I will get back to you once my 5hr unresumable download finishes. 👍

sgtsixpack, 2020-11-03:
I just downloaded Wireshark-win64-3.4.0.exe but I am unsure of the proceedure.  I was looking back in topics I've posted but can't find the one where I used it before.  I think there is an option that I need to check prior to installing wireshark (also need to install RGBfusion to get traffic to the monitor).  I sold my Corsair Vengence RGB pro but I have installed an m.2 cooler with RGB which is working correctly.  

I have restarted my computer (without touching HWiNFO) with theUSB leads back in.  This is the screenshot I took, showing the device in question found:
https://1drv.ms/u/s!AhCKpFOJjf6jgkAoi_oEX42e3XaT?e=E3dEcG

I think I will install wireshark and RGBfusion on a backup clone OS to keep my NVME boot drive copy clean of the dreaded manufacturer RGB software.

jonasmalacofilho, 2020-11-04:
Take a look at [Capturing USB traffic](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/developer/capturing-usb-traffic.md).

closed 2020-12-25 by jonasmalacofilho