liquidctl/liquidctl #185: Fedora 31 1.4.1 package

MarshallAsch, 2020-09-22:
The rpm package on https://src.fedoraproject.org/rpms/liquidctl seems to be out of date for fedora 31. 

Is there any plan to update the rpm?

Installing from source seems to run into a similar issue as #145, but reinstalling does not seem to solve the problem.

`sudo dnf install liquidctl`
- Liquidctl will run now with version `1.3.3` but cant find my Corsair H100i Platnum

At this point running `$ python -c "import usb; import hid; import hidraw"` fails with 
`ImportError: dynamic module does not define module export function (PyInit_hidraw)`

and if I were to remove the packaged version of liquidctl (keeping the dependancies) and try to run `pytest` from the git directory there are a ton of 
`ImportError: dynamic module does not define module export function (PyInit_hidraw)` errors.

Relevant versions:
```
Fedora: 31
Kernel: 5.8.10
Python: 3.7.9
pip: 19.1.1
libusb: 0.1.5 (from package manager)
setuptools: 41.6.5
hidapi: 0.9.0 (from system package manager)
pyusb: 1.1.0 (from pip)
```

jonasmalacofilho, 2020-09-22:
I'm not sure what is the policy on package updates on Fedora.  Perhaps they haven't packaged 1.4.1 due to a stability guarantee?  
_(1.4.1 would contain a major version number change, and some distros only backport patch number bumps)._

That said, the Fedora package is handled by their developers.  I can ping Arthur here, but it's probably better if you do that through the appropriate Fedora channel (?).

---

Hm, #145 is generally caused by broken installations of `python-hidapi` (`cython-hidapi`).

The issue is that if either your distro or you are using any non-default build options (such as `--with-system-hidapi`, which Fedora may use), you may end up in a scenario where your Python modules don't match your `.so`.  It's a mess...

The way to debug this is to manually inspect the `.so` files.  Something like this (from #145):

```
$ find `pip show hidapi | grep Location | awk '{print $2}'` -iname 'hid*.so' -exec ldd {} \;
```

MarshallAsch, 2020-09-22:
I'm not quite sure what I'm looking at with this output or what to do with this information, but these this is what that command gave me.

```
$ find `pip show hidapi | grep Location | awk '{print $2}'` -iname 'hid*.so' -exec ldd {} \; 
	linux-vdso.so.1 (0x00007ffc57283000)
	libhidapi-libusb.so.0 => /lib64/libhidapi-libusb.so.0 (0x00007efe416e0000)
	libpython3.7m.so.1.0 => /lib64/libpython3.7m.so.1.0 (0x00007efe41392000)
	libc.so.6 => /lib64/libc.so.6 (0x00007efe411c9000)
	librt.so.1 => /lib64/librt.so.1 (0x00007efe411be000)
	libusb-1.0.so.0 => /lib64/libusb-1.0.so.0 (0x00007efe411a2000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007efe41188000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007efe41164000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007efe4115d000)
	libutil.so.1 => /lib64/libutil.so.1 (0x00007efe41158000)
	libm.so.6 => /lib64/libm.so.6 (0x00007efe41012000)
	/lib64/ld-linux-x86-64.so.2 (0x00007efe4171f000)
	libudev.so.1 => /lib64/libudev.so.1 (0x00007efe40fe4000)
	linux-vdso.so.1 (0x00007ffe3ddf9000)
	libhidapi-hidraw.so.0 => /lib64/libhidapi-hidraw.so.0 (0x00007fb5e45c5000)
	libpython3.7m.so.1.0 => /lib64/libpython3.7m.so.1.0 (0x00007fb5e4277000)
	libc.so.6 => /lib64/libc.so.6 (0x00007fb5e40ae000)
	libudev.so.1 => /lib64/libudev.so.1 (0x00007fb5e4080000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007fb5e405e000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007fb5e4057000)
	libutil.so.1 => /lib64/libutil.so.1 (0x00007fb5e4050000)
	libm.so.6 => /lib64/libm.so.6 (0x00007fb5e3f0a000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fb5e4601000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007fb5e3ef0000)

```

suve, 2020-09-22:
The reason I decided not to update the `liquidctl` RPM for Fedora 31 and 32 is precisely because of the issues described in the OP - while *liquidctl* does not mention a minimum version of `python3-hidapi`, the test suite fails with v0.7.99.post20 (which is the version present in both F31 and F32). Here's a scratch build to demonstrate: https://koji.fedoraproject.org/koji/taskinfo?taskID=52049325

In F33, `python3-hidapi` has been updated to v0.9.0.post2 and the test suite passes a-ok. So this is either caused by the version of `python3-hidapi` being too old, or - as your adventures in #145 suggest - the `python3-hidapi` package is just broken in F32 and F31.

MarshallAsch, 2020-09-22:
Well thats frustrating. But makes sense I have seen several posts elsewhere about there being some issues with version 0.9.0 of the trezor/cython-hidapi program. Later tonight Ill give it a shot installing that from source and seeing if that changes anything. I don't think that will be too useful in general but it might be a workaround if it works.

jonasmalacofilho, 2020-09-22:
Thanks!

Digging a little deeper, the issue reproduces on python-hidapi 0.7.99.post20 with default build options and in a virtual environment:

```
$ python -m venv venv

$ venv/bin/pip install hidapi==0.7.99.post20
...
Successfully installed hidapi-0.7.99.post20

$ venv/bin/python -c "import hidraw"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: dynamic module does not define module export function (PyInit_hidraw)
```

I could be wrong, but this seems to have been solved in trezor/cython-hidapi@e32b7664fc9b63703578aa78ebd629c79cf35a2c, eventually landing in 0.7.99.post21.  Could the Fedora 31/32 python-hidapi packages be updated to that version?

jonasmalacofilho, 2020-09-22:
> I have seen several posts elsewhere about there being some issues with version 0.9.0 of the trezor/cython-hidapi program

One issue I'm aware of is that 0.9.0.post3 references an unreleased API from libusb/hidapi (trezor/cython-hidapi#90, trezor/cython-hidapi#92), and thus is a problem if `--with-system-libs` is used.

MarshallAsch, 2020-09-23:
I tried doing a manual install of trezor/cython-hidapi@2f769bee8a89331bbf5a0e40de38fb2dbb3a2baa following the instructions there to "use hidraw API instead of libusb", and that seemed to install `hidapi v0.9.0.post2` . But I still got the same error message. 

```
$ python -c "import hidraw"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: dynamic module does not define module export function (PyInit_hidraw)
```

I noticed that in version 1.3.3 the module `hidraw` was not required. Without going too far down this rabbit hole do you think it would be feasible if I were to make a modified version that did not use that dependancy and used one of the other hidapi implementations instead (Im not too familiar with how that library works)?

jonasmalacofilho, 2020-09-23:
The `hidraw` module is optional (now and back in 1.3.3): with cython-hidapi it may not exist or may simply be called `hid`, we never know.  The issue on 0.7.99post20 is that it exists, but is fundamentally broken.

That said, all later versions of cython-hidapi work (albeit with _other_ issues, like the `--with-system-libs` packaging problem with 0.9.0post3 mentioned before).

With `testall.sh`:

```sh
#/bin/bash

for version in 0.7.99post20 0.7.99post21 0.9.0 0.9.0post1 0.9.0post2 0.9.0post3
do
	python -m venv "$version"
	"$version/bin/pip" install "hidapi==$version" 2>/dev/null >"$version.log"
	if "$version/bin/python" -c "import hidraw" >>"$version.log"
	then
		"$version/bin/pip" install liquidctl 2>/dev/null >>"$version.log"
		if "$version/bin/liquidctl" list >>"$version.log"
		then
			echo cython-hidapi==$version: everything works
		else
			echo "cython-hidapi==$version: importing hidraw works but liquidctl FAILS"
		fi
	else
		echo cython-hidapi==$version: importing hidraw FAILS
	fi
	rm -rf "$version"
done
```

```
./testall.sh
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: dynamic module does not define module export function (PyInit_hidraw)
cython-hidapi==0.7.99post20: importing hidraw FAILS
cython-hidapi==0.7.99post21: everything works
cython-hidapi==0.9.0: everything works
cython-hidapi==0.9.0post1: everything works
cython-hidapi==0.9.0post2: everything works
cython-hidapi==0.9.0post3: everything works
```

You need to double check your installation since you're getting the 0.7.99post20 error even though you tried to install 0.9.0post2.

---

> Without going too far down this rabbit hole do you think it would be feasible if I were to make a modified version that did not use that dependancy and used one of the other hidapi implementations instead (Im not too familiar with how that library works)?

It is feasible, sure.  That said, I recommend that you stick to the underlying hidraw backend on Linux, even if using a different Python wrapper.  _(cython-hidapi wraps hidapi, which in turn is implemented [on Linux] both on top of libusb and directly on top of Linux hidraw)_

Let me know how this goes, and if the dependency you pick is just as widely available on Linux as cython-hidapi, and easily distributed and installed on both Windows and mac OS, I'll be happy to take a pull request.

closed 2020-09-30 by jonasmalacofilho

MarshallAsch, 2020-10-08:
I ran that script and it gave me the correct results. After I installed `pip install hidapi==0.9.0post3` installing liquidctl from source worked. :) yay