liquidctl/liquidctl #186: Product ID parsing glitch

andrejpodzimek, 2020-09-23:
I'm using version 1.4.1 [from AUR](https://aur.archlinux.org/packages/liquidctl/) and it doesn't like product IDs in the usual hexadecimal format (used by `lsusb`) when they start with `0`. This might have something to do with integer parsers that expect such values to be octal, so hexadecimal digits can be considered garbage or, in the worst case, misinterpreted.

The device:
```
Bus 003 Device 002: ID 1b1c:0c17 Corsair H115i Platinum
```

The command line:
```bash
liquidctl --vendor=1b1c --product=0c17 set fan2 speed 20 50 40 80 45 100
```

The error message:
```
ValueError: invalid literal for int() with base 0: '0c17'
```

A workaround is to use `0x` to let the parser know it's hexadecimal.
```bash
liquidctl --vendor=0x1b1c --product=0x0c17 set fan2 speed 20 50 40 80 45 100
```

jonasmalacofilho, 2020-09-23:
Agreed, that would be better.  Also more consistent, since colors don't even support the `0x` prefix.

Actually, both IDs and colors should _optionally_ accept the prefix (either way being parsed as in base 16).

---

This is a good first issue, please consider contributing the improvement.

_(The corresponding parsing functions need doctests showing that both cases [with and without the prefix] are handled appropriately)._

MarshallAsch, 2020-10-09:
Im going to take on this issue. and I wanted to clarify something. 

the values for the `product` and `colors` should accept hex values optionally with a `0x` prefix. This also applies to the `vendor` flag. Should the `vendor` and `product` be allowed to be given as decimal numbers or some other base number?

jonasmalacofilho, 2020-10-09:
Great, thanks!

I think we should disallow decimal values for the `--vendor` and `--product` IDs.  But there's not much we can do in the code, this is more of a documentation thing.

MarshallAsch, 2020-10-09:
Thats what I figured, especially since those values would be given elsewhere to the user as hex values, and what sane person would convert it to decimal to give it back to the program.

jonasmalacofilho, 2020-10-17:
Improved/fixed with #189.

closed 2020-10-17 by jonasmalacofilho