liquidctl/liquidctl #188: Deelopment of a GUI

Sporknife, 2020-10-09:
Yes, I know that there is an already existing issue but its old (2018). So here it goes.                                                                                 
I'm working on a GUI for liquidctl made in with Qt. Feel free to join me in this [this](https://discord.gg/xW5EYb) discord server.  
Currently its in early development and i am working on creating all widgets/elements in Qt Designer or in Python (whatever is better).

cluxter, 2020-12-04:
This sounds like a nice project. Do you have a Git repo somewhere to which we could contribute?

Sporknife, 2020-12-04:
@cluxter yea i do. I made it public some time ago again. Repo linked [here](https://github.com/Sporknife/Liquidctl-Qt).

closed 2021-02-27 by jonasmalacofilho