liquidctl/liquidctl #190: question: Corsair H100i Platinum `super-fixed` option

MarshallAsch, 2020-10-09:
When setting the RGB colours for the H100i Platinum cooler my understanding from the docs is that the command  `# liquidctl set led color super-fixed <up to 24 colors>` should set individual LED's to static colors and any unset values will be left off. 

When I tested it with `liquidctl set led color super-fixed FF0000 00FF00 0000FF` the LEDs change colour in a marquee like pattern, and it does not go around every single LED. It Simultaneously is set for the pump, fan1, and fan2. Im not sure if this is the desired behaviour or if it is a bug.

I have attached a video recording (in a zip folder) showing this behaviour. 

[IMG_0345.MOV.zip](https://github.com/jonasmalacofilho/liquidctl/files/5357108/IMG_0345.MOV.zip)

- Firmware version: 1.1.15
- Fedora31
- liquidctl 1.4.1

jonasmalacofilho, 2020-10-10:
That's certainly not the documented behavior.

And besides the possible/probable implementation error, it's also not how I understood the device/protocol to work.

(I was working with iCue and an emulated cooler – an Arduino with custom firmware – so I could only see what iCue showed me, not what a real device would do).

Hopefully you can help here too? : )

P.S. You may need to do some new traffic captures,<sup>[[1]](https://github.com/jonasmalacofilho/liquidctl/tree/master/docs/developer)</sup> or at least take a look at the ones we already have.<sup>[[2]](https://github.com/jonasmalacofilho/liquidctl-device-data)</sup>

MarshallAsch, 2020-10-10:
Yup I'd be happy to see if I can figure it out, (and also trying to get my hacktoberfest PRs in here).
 
I have never done any protocol reverse engineering before, so that might be a bit tough to figure out. Im trying to get a windows VM setup on this computer so I can put icue on that. (while Im figuring that out Ill try to fill out some information for [this](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/developer/capturing-usb-traffic.md#capturing-usb-traffic-on-linux) section of the documentation)

Do you have any tips on getting things setup to identify commands / test to see if sending a certain command actually does what I think it does?

jonasmalacofilho, 2020-10-10:
> (while Im figuring that out Ill try to fill out some information for this section of the documentation)

The gist is to enable the `usbmon` kernel driver for USB support in Wireshark, pass through the USB device to your Windows VM, and monitor the correct bus back in Wireshark on the Linux host.

> Do you have any tips on getting things setup to identify commands / test to see if sending a certain command actually does what I think it does?

For identifying things there are a few techniques documented in [techniques...md](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/developer/techniques-for-analyzing-usb-protocols.md); it also links to a script I used to test some ideas on the various captures for this family of devices.  Since you have access to one of these coolers you can also send packets directly from a modified driver, your own scripts, or even a Python REPL.

MarshallAsch, 2020-10-10:
I was able to confirm that the data payload that liquidctl is using is correct. the command that is sent for `super-fixed` does set static LEDs. however there is a different command that must be sent first to change it from cycle mode to fixed mode. (Im still trying to identify what it is. 

I have found [this](https://forum.corsair.com/v3/showthread.php?t=120092) that mentions some different LED modes for `static color`, `2 color cycle`, `4 color` and `tempature` mode. based on the behaviour that was being expressed before I believe that it was in a 4 color cycle mode. 

After changing the mode using icue in the VM to static color, when I send LED commands from liquidctl it behaves as expected.

commit 0daaa902 (2020-10-22): Merge #194: hydro_platinum: ensure super-fixed results in a fixed mode

Without some magic messages, the color may end up displaying a marquee-like pattern.

Closes #190 ("question: Corsair H100i Platinum `super-fixed` option").
closed 2020-10-22 by jonasmalacofilho