liquidctl/liquidctl #192: Unexpected error with Corsair HX1200i, 72 is not a valid OCPMode

urbinek, 2020-10-09:
Brand new HX1200i, runnng on Windows 10. After installation i had problem with usb backend so i've copied dll to system32 from https://netcologne.dl.sourceforge.net/project/libusb/libusb-1.0/libusb-1.0.21/libusb-1.0.21.7z 

When i try to initialize or display info about HX1200i i have errors
```PowerShell
> liquidctl initialize all
NZXT Kraken X (X53, X63 or X73) (experimental)
├── Firmware version       2.1.0
├── Pump Logo LEDs      detected
└── Pump Ring LEDs      detected

ERROR: Unexpected error with Corsair HX1200i
ValueError: 72 is not a valid OCPMode

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
ValueError: 72 is not a valid OCPMode
```
```PowerShell
> liquidctl status
NZXT Kraken X (X53, X63 or X73) (experimental)
├── Liquid temperature    35.0  °C
├── Pump speed            1463  rpm
└── Pump duty               42  %

ERROR: Unexpected error with Corsair HX1200i
Traceback (most recent call last):
OverflowError: Python int too large to convert to C int
```
```PowerShell
> liquidctl list -g
[DEBUG] liquidctl.cli: running liquidctl v1.4.1
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a35f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 054c:0cd3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c232
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c231
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1c08
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair HX1200i
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 17e9:6015
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 17a0:0001
[DEBUG] liquidctl.driver.usb: probing drivers for device 054c:0cd3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:1028
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a359
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8140
[DEBUG] liquidctl.driver.usb: probing drivers for device 10de:1ad8
[DEBUG] liquidctl.driver.usb: probing drivers for device 17e9:6015
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:1029
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b21:1242
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a357
[DEBUG] liquidctl.driver.usb: probing drivers for device 1a40:0101
[DEBUG] liquidctl.driver.usb: probing drivers for device 8086:a12f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a35f
[DEBUG] liquidctl.driver.usb: probing drivers for device 1a86:7523
[DEBUG] liquidctl.driver.usb: probing drivers for device 17a0:0001
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a35f
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a356
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1c08
Device ID 0: NZXT Kraken X (X53, X63 or X73) (experimental)
├── Vendor ID: 0x1e71
├── Product ID: 0x2007
├── Release number: 0x0201
├── Serial number: 00000000001A
├── Bus: hid
├── Address: \\?\hid#vid_1e71&pid_2007#6&1f09b849&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
└── Driver: KrakenX3 using module hid
[DEBUG] liquidctl.cli: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice

Device ID 1: Corsair HX1200i
├── Vendor ID: 0x1b1c
├── Product ID: 0x1c08
├── Release number: 0x0002
├── Bus: hid
├── Address: \\?\hid#vid_1b1c&pid_1c08#6&2e1af3c8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
└── Driver: CorsairHidPsu using module hid
[DEBUG] liquidctl.cli: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice
```

jonasmalacofilho, 2020-10-10:
Hi!

Since it's a brand new unit, it's possible that the protocol has changed, even if only a bit.  Can you also please show me the output of `liquidctl status --debug`?

There are also some things in the logs that hint that I might have broken something in last couple releases.  You could give liquidctl 1.3.3 a try.  If it's a bug on our side, it's very likely to have been caused by the transition on Windows from libusb to hidapi (for this particular device) in 1.4.0.

urbinek, 2020-10-10:
btw, iCUE is dissabled from autostart but i found 2 services 3 services running in background

- CorsairGamingAudioConfig

- CorsairLLAService

- CorsairService

If i disable them (wihout reboot) liquidctl is freezing on all commands related with psu 

```Powershell
> liquidctl status --debug
[DEBUG] liquidctl.cli: running liquidctl v1.4.1
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a35f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 054c:0cd3
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c232
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c231
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1c08
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair HX1200i
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1bf0
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 17e9:6015
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 17a0:0001
[DEBUG] liquidctl.driver.usb: probing drivers for device 054c:0cd3
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:1028
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a359
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8140
[DEBUG] liquidctl.driver.usb: probing drivers for device 10de:1ad8
[DEBUG] liquidctl.driver.usb: probing drivers for device 17e9:6015
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:1029
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b21:1242
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a357
[DEBUG] liquidctl.driver.usb: probing drivers for device 1a40:0101
[DEBUG] liquidctl.driver.usb: probing drivers for device 8086:a12f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a35f
[DEBUG] liquidctl.driver.usb: probing drivers for device 1a86:7523
[DEBUG] liquidctl.driver.usb: probing drivers for device 17a0:0001
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a35f
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a356
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1c08
[DEBUG] liquidctl.cli: device: NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: read 64 bytes: 75:02:30:00:2e:00:15:57:41:56:33:34:34:20:01:1e:02:6c:05:28:28:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
NZXT Kraken X (X53, X63 or X73) (experimental)
├── Liquid temperature    30.2  °C
├── Pump speed            1388  rpm
└── Pump duty               40  %

[DEBUG] liquidctl.cli: device: Corsair HX1200i
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x02 with 63 bytes: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: wrote 65 total bytes, expected 64
[DEBUG] liquidctl.driver.usb: read 64 bytes: 02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x03 with 63 bytes: d2:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00: [... cut ...]
    _print_dev_status(dev, dev.get_status(**opts))
  File "c:\users\miark\appdata\local\programs\python\python37\lib\site-packages\liquidctl\driver\corsair_hid_psu.py", line 113, in get_status
    ('Current uptime', self._get_timedelta(_CORSAIR_READ_UPTIME), ''),
  File "c:\users\miark\appdata\local\programs\python\python37\lib\site-packages\liquidctl\driver\corsair_hid_psu.py", line 179, in _get_timedelta
    return timedelta(seconds=secs)
OverflowError: Python int too large to convert to C int
```

urbinek, 2020-10-10:
It seems that on 1.3.3 works slightly better
```Powershell
PS C:\Users\miark\liquidctl-1.3.3-bin-windows-x86_64> .\liquidctl.exe list
Device ID 0: Corsair HX1200i (experimental)
```
```Powershell
PS C:\Users\miark\liquidctl-1.3.3-bin-windows-x86_64> .\liquidctl.exe status
WARNING: reading the +12V OCP mode is an experimental feature
Corsair HX1200i (experimental)
├── Current uptime             0:17:35
├── Total uptime              17:17:35
├── Temperature 1                 39.5  °C
├── Temperature 2                 30.5  °C
├── Fan control mode          Hardware
├── Fan speed                        0  rpm
├── Input voltage               230.00  V
├── Total power                  98.00  W
├── +12V OCP mode           Multi rail
├── +12V output voltage          12.16  V
├── +12V output current           7.00  A
├── +12V output power            82.00  W
├── +5V output voltage            5.00  V
├── +5V output current            2.00  A
├── +5V output power             10.00  W
├── +3.3V output voltage          3.30  V
├── +3.3V output current          1.88  A
└── +3.3V output power            6.00  W
```
```Powershell
PS C:\Users\miark\liquidctl-1.3.3-bin-windows-x86_64> .\liquidctl.exe initialize
```
```PowershellPS C:\Users\miark\liquidctl-1.3.3-bin-windows-x86_64> .\liquidctl.exe status
ERROR: Unexpected error with Corsair HX1200i (experimental)
Traceback (most recent call last):
OverflowError: Python int too large to convert to C int
```

jonasmalacofilho, 2020-10-10:
Can you try to disable all of these services then do a cold start (turn of the PSU long enough for everything to shutdown; some boards have LEDs to indicate this, if not just way ~5 minutes) and see if that helps both 1.3.3 and 1.4.1?

Having multiple tools accessing the same device, especially (but not only) if the aren't aware of each other, is likely to cause issues as some responses may reach the wrong applications.

P.S. 1.3.3 still had some issues, so I'm not sure if the successful `status` with that version wasn't just a bit of luck.

jonasmalacofilho, 2020-10-22:
@urbinek , did you managed to do the test mentioned in the previous comment?

urbinek, 2020-10-24:
@jonasmalacofilho, sorry I got distracted and dropped it for a while.

Regardless, I've uninstalled iCUE, switched PSU off for 10 min
```PowerShell
PS C:\Users\miark\liquidctl-1.3.3-bin-windows-x86_64> .\liquidctl.exe list --debug
[DEBUG] __main__: running liquidctl v1.3.3 (AppVeyor; liquidctl-1.3.3-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=usb, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17ef:0x1028
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17ef:0xa359
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x451:0x8140
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x10de:0x1ad8
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17e9:0x6015
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17ef:0x1029
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b21:0x1242
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17ef:0xa357
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1a40:0x101
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x8086:0xa12f
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc080
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x54c:0x5c4
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1a86:0x7523
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17a0:0x1
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17ef:0xa35f
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17ef:0xa356
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x451:0x8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x8087:0x29
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0x1c08
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair HX1200i (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17ef:0x1028
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17ef:0xa359
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x451:0x8140
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x10de:0x1ad8
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17e9:0x6015
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17ef:0x1029
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b21:0x1242
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17ef:0xa357
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1a40:0x101
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x8086:0xa12f
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0xc080
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x54c:0x5c4
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1a86:0x7523
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17a0:0x1
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17ef:0xa35f
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17ef:0xa356
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x451:0x8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x8087:0x29
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1b1c:0x1c08
Device ID 0: Corsair HX1200i (experimental)
├── Vendor ID: 0x1b1c
├── Product ID: 0x1c08
├── Release number: 0x0002
├── Bus: usb3
├── Address: 1
├── Port: 3
└── Driver: CorsairHidPsuDriver using module usb
[DEBUG] __main__: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver; PyUsbHid
```
```PowerShell
PS C:\Users\miark\liquidctl-1.3.3-bin-windows-x86_64> .\liquidctl.exe status --debug
[DEBUG] __main__: running liquidctl v1.3.3 (AppVeyor; liquidctl-1.3.3-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=usb, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17ef:0x1028
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x17ef:0xa359
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x451:0x8140
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x10de:0x1ad8
[... 396 lines cut ...]
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 054c:05c4
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a35f
[DEBUG] liquidctl.driver.usb: probing drivers for device 1a86:7523
[DEBUG] liquidctl.driver.usb: probing drivers for device 17a0:0001
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a35f
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a356
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[... cut ...] 0:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x03 with 63 bytes: d2:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: wrote 65 total bytes, expected 64
[DEBUG] liquidctl.driver.usb: wrote 65 total bytes, expected 64

```
And it just hangs up after this line, i had to kill liquidctl.exe

jonasmalacofilho, 2020-10-24:
Thanks!

On 1.3.3, aren't you `initialize`-ing the power supply before attempting to read from it?

Also, can you see if the latest code in the master branch helps?  It contains a small a fix for this driver, maybe it's more relevant than I initially realized.  The most recent build for Windows is: [liquidctl-c67da90d24b4-bin-windows-x86_64.zip](https://ci.appveyor.com/api/buildjobs/9eg9vckik3996cs7/artifacts/liquidctl-c67da90d24b4-bin-windows-x86_64.zip).

urbinek, 2020-10-25:
This version seems to work fine
```PowerShell
PS C:\Users\miark\liquidctl-c67da90d24b4-bin-windows-x86_64> .\liquidctl.exe initialize all --debug
[DEBUG] __main__: running liquidctl v1.4.1 (AppVeyor; liquidctl-c67da90d24b4-bin-windows-x86_64)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a35f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c232
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c231
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 054c:05c4
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1c08
[DEBUG] liquidctl.driver.usb: instanced driver for Corsair HX1200i
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:1028
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a359
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8140
[DEBUG] liquidctl.driver.usb: probing drivers for device 10de:1ad8
[DEBUG] liquidctl.driver.usb: probing drivers for device 17e9:6015
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:1029
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b21:1242
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a357
[DEBUG] liquidctl.driver.usb: probing drivers for device 1a40:0101
[DEBUG] liquidctl.driver.usb: probing drivers for device 8086:a12f
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c080
[DEBUG] liquidctl.driver.usb: probing drivers for device 054c:05c4
[DEBUG] liquidctl.driver.usb: probing drivers for device 1a86:7523
[DEBUG] liquidctl.driver.usb: probing drivers for device 17a0:0001
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a35f
[DEBUG] liquidctl.driver.usb: probing drivers for device 17ef:a356
[DEBUG] liquidctl.driver.usb: probing drivers for device 0451:8142
[DEBUG] liquidctl.driver.usb: probing drivers for device 1038:1607
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:1c08
[DEBUG] __main__: device: NZXT Kraken X (X53, X63 or X73) (experimental)
[DEBUG] liquidctl.driver.usb: discarded 0 previously enqueued reports
[DEBUG] liquidctl.driver.usb: writting report 0x10 with 63 bytes: 01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x20 with 63 bytes: 03:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x70 with 63 bytes: 02:01:b8:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: writting report 0x70 with 63 bytes: 01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 11:01:30:00:2e:00:15:57:41:56:33:34:34:20:07:20:02:02:01:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 71:01:30:00:2e:00:15:57:41:56:33:34:34:20:01:00:01:01:00:00:02:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: read 64 bytes: 75:02:30:00:2e:00:15:57:41:56:33:34:34:20:01:26:01:da:04:21:23:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[... cut ...]
├── Total power                       96.00  W
├── +12V OCP mode                Multi rail
├── +12V output voltage               12.19  V
├── +12V output current                7.00  A
├── +12V output power                 88.00  W
├── +5V output voltage                 4.98  V
├── +5V output current                 2.00  A
├── +5V output power                  10.00  W
├── +3.3V output voltage               3.30  V
├── +3.3V output current               1.88  A
└── +3.3V output power                 6.00  W
```

jonasmalacofilho, 2020-10-25:
Great! : )

Basically there was an off-by-one error, because of the change of what library is used and some quirkiness in the HID spec. Windows tried to fix the error for us, but that only made things worse.  But it's fixed now, and thanks for reporting/helping test this.

closed 2020-10-25 by jonasmalacofilho