liquidctl/liquidctl #193: Corsair H100i Platinum new hardware default mode.

MarshallAsch, 2020-10-10:
When testing things for #190 I noticed a new option in icue titled `Hardware lighting` which it describes as a setting to allow you to configure a single lighting effect for the device when icue is not running. This seems to be a method to override the built in rainbow cycling mode. 

The command for this seems to be a mix of `_CMD_GET_STATUS` value being set in byte[2] of the payload as well as the `_CMD_SET_LIGHTING[12]` value being `or`'d with the sequence number.  Im assuming that this would also require the led mode talked about in #190 to be changed for any effect that does not use the 4 color cycle mode.

(Im also going to work on an implementation for this too.)

MarshallAsch, 2020-10-12:
Never mind, after testing that mode in icue it does not actually do what I thought it did, it might actually add that ability properly in a future firmware update tho.

jonasmalacofilho, 2020-10-14:
So this isn't related to what iCue does when set to static color?

> After changing the mode using icue in the VM to static color, when I send LED commands from liquidctl it behaves as expected.<sup>[[1]](https://github.com/jonasmalacofilho/liquidctl/issues/190#issuecomment-706606523)

MarshallAsch, 2020-10-14:
That may have been related, I initially believed that the option in icue was to set one setting that would be stored on the cooler to set the colour that way before the program actually starts working. but that is not the case. 

This can probably be closed tho.

jonasmalacofilho, 2020-10-14:
I'm still interested in learning how this "Hardware lighting" works.  The liquidctl CLI can't (by itself) support modes that _aren't_ running on the cooler's hardware, and it's also more efficient to have the animation run on the cooler, instead of hammering the USB bus.

_(IIRC this family of devices is one that particularly massacres the USB bus with interrupt requests [which have high priority] when the software-backed animation modes in iCue are used)._

MarshallAsch, 2020-10-14:
> IIRC this family of devices is one that particularly massacres the USB bus

Yup that is correct. All of the lighting effects are controlled entirely by the icue software. The option may have been added in preparation for a future firmware update.

> I'm still interested in learning how this "Hardware lighting" works.

Me too. Using hardware to control the lighting would be far better than having to have a service figure it out, Ill check out the forms in the morning to see if Corsair talks about that feature at all.

jonasmalacofilho, 2020-10-14:
> check out the forms

Forms or forums?  I hopeful that it's the first and that it somehow means you got access to some official documentation, which would great! : )

MarshallAsch, 2020-10-14:
> somehow means you got access to some official documentation,

LMAO I wish, its forums, I just can't spell.

MarshallAsch, 2020-10-31:
After capturing some more data dumps of setting the hardware mode I have been able to identify part of the packet payload. 

I have found that there is a separate command to set the mode and to set the colors.
This command format seems to be the same as the "magic packets" that I noticed in #194.

I will keep documenting what I find elsewhere to add all at once later, but this update is here if anyone else is interested. 

the command format seems to be. something like this example:
`3fc10001ffffffffffffffffffffffffff7f7f7f7f7f09ba00000300ffffffff00ffffffff00ffffffff00ffffffff00ffffffffffffffffffffffffffffff84`

where the important stuff seems to be this chunk `09 ba 00 00 03 00`. the `09` seems to be the mode. `ba` is the speed of the animation, and the `03` seems to be the number of colors to use in the sequence -1. 

I have sort of identified the following commands

| Code | command |
| ------- | -------- |
| `00` | solid single color |
| `01` | Jump |
| `09` | smooth fade |
| `0b` | some sort of different fade |

The colors can be set using a normal set color message. However when the mode and color is sent by only sending those 3 messages it does not seem to be persistent across device reboots.

(you can reboot just the cooler by sending the reboot command to the device. 

To capture this data you can use icue to set a hardware mode, then shutdown the computer. (or just icue but Im not sure how to stop it) and the messages will be the last handful that are sent to the device

jonasmalacofilho, 2020-11-09:
Can you work on a pull request that exposes these modes through the `set_color` API and the `set <channel> color` CLI command?

It would also be nice to have documented what exactly persists across a `REBOOT` command, across a normal system shutdown and across a mechanical system shutdown (=PSU turned off for some time).

MarshallAsch, 2020-11-09:
Yup Im currently working on it. Im still trying to decode what the payload actually is. (it isn't what I though the first time, or the second time, or the third time, but I think the 4th seems to make sense). 

I have also more or less figured out what those 'magic' packets were that we were both not quite comfortable including in #194.

MarshallAsch, 2020-11-09:
Im also quite enjoying working on this project a ton of learning is going into this.

jonasmalacofilho, 2020-11-09:
Ok, thanks!

---

> Im also quite enjoying working on this project a ton of learning is going into this.

I'm glad to hear!  By the way, would you be interested in becoming a co-maintainer of the project?

MarshallAsch, 2020-11-09:
> By the way, would you be interested in becoming a co-maintainer of the project?

Sure! That sounds cool, I haven't been a maintainer on a project before (aside from my own)

jonasmalacofilho, 2020-11-11:
> Sure! That sounds cool, I haven't been a maintainer on a project before (aside from my own)

Great!  I'll set it up...

Do you have an email address, or maybe Discord, where I can reach you?

jonasmalacofilho, 2021-01-25:
Can we leave this for 1.6.0?  Do we want to do that?

MarshallAsch, 2021-01-26:
> Can we leave this for 1.6.0? Do we want to do that?

Yup that would make sense, I may not have a complete set of hardware modes. 
I did not work on this issue as much as I had intended too.

jonasmalacofilho, 2022-07-03:
For reference, RogueZeta @ OpenRGB has learned a few things about this:

https://gitlab.com/TheRogueZeta/OpenRGB/-/commits/h100i-aio

There's also some stuff in an (appropriately named) archived thread on liquidctl Discord.