liquidctl/liquidctl #195: Feature: initialize command gets some values to a config file

MarshallAsch, 2020-10-13:
I have been enjoying using this program, and I know that I can add a system service or set a script to run on boot, but what are your thoughts and createing a config file `.config/liquidctl`, .liquidctl`, etc. that will keep track of my preferred default LED settings, fan speeds, and pump speeds  and will set those when running `liquidctl initialize`?

If you think that this might be a useful addition I can look into adding this.

jonasmalacofilho, 2020-10-14:
That's a great idea, but let's do a quick design phase before committing to an implementation.

There's the obvious the question of which format to use.  On that, I've been doing a lot of Rust lately, so [TOML](https://github.com/toml-lang/toml) files are something I would suggest.  Even though I acknowledge that they have some issues, they have a spec of some sort and fairly simple for humans and (it seems) programs.  But we should consider other alternatives.

Besides the serialization format, we need to figure out what schema to use.  With TOML I would probably separate between identifying a device (with vendor/product IDs and/or bus name and addresses, and/or serial numbers) and specifying its initialization options (i.e. options to `initialize` itself) and which other actions to perform (like setting LED settings or fan speeds) by default.

Finally, there should be command line arguments and/or environment variables to disable the use of the config file, or to use a different one than that in the default location.

P.S. A general/global section could also solve #67 by allowing the user to set the legacy mode.

MarshallAsch, 2020-10-14:
I personally really like the [.ini](https://en.wikipedia.org/wiki/INI_file) like configuration files, [TOML](https://en.wikipedia.org/wiki/TOML), seems to be a formally structured variant of that. I generally don't think that json, or yaml make good configuration files they seem to verbose. 

I didn't think about specify the vendor/product IDs and be able to set `initialize` options for those, but that seems like a good idea since CPU coolers isn't the only type of device that this supports (If it was I would say I don't think that makes much sense because what normal person is going to have more than one AIO in their system). 

Ill try to think of some sort of potential format over the next couple days. 

Would adding this in cause any issues with public API changes within the application, I believe that was mentioned somewhere in a different issue.

jonasmalacofilho, 2020-10-14:
> I generally don't think that json, or yaml make good configuration files they seem to verbose.

Agreed.  I would argue JSON is not really human writable, and I can never get YAML configs right the first time, since things can work one way but not the other, and it's never that well documented (besides the most trivial cases).

> (If it was I would say I don't think that makes much sense because what normal person is going to have more than one AIO in their system).

Actually there is a fairly reasonable and not _that_ uncommon use case: GPU cooling with things like the Kraken G12.  Another use case is dual-socket systems, although that's fairly rare (I only remembered about it because I saw a picture last night of home dual-socket build server cooled by a couple of Corsair H80i somethings).

> Ill try to think of some sort of potential format over the next couple days.

Thanks, I'll try to think about it a bit too.

> Would adding this in cause any issues with public API changes within the application, I believe that was mentioned somewhere in a different issue.

That's a concern and backwards compatibility is a hard requirement.  But the explicit and user-controlled configuration file mitigates most potential problems I can think of.

jonasmalacofilho, 2020-10-14:
Sorry, forgot to add: my problem with INI files is that (AFAIK) there is no implementation-agnostic specification.  And implementations vary in subtle ways...

> This module provides the `ConfigParser` class which implements a basic configuration language which provides a structure similar to what’s found in Microsoft Windows INI files.<sup>[[1]](https://docs.python.org/3/library/configparser.html)</sup>

MarshallAsch, 2020-10-14:
Attempt 1 at a config file format:

using TOML

```TOML
[device_name]
vendor="hex str"
product="hex str"
serial="hex str"
bus="int"
address="int"
legacy=true         #bool (default false)

# for setting the LED settings
[color]
    [color.<channel>]
    mode="str for the color mode"
    colors="hex string of color values"
    speed="str or int depending on which the device supports"
    time_on="int"
    time_off="int"
    alert_color="hex str"
    alert_threshhold="value"

    #... can have multuple subsections here as long as the channel names are unique

# for setting the fan and pump speeds
[speed]
    [speed.<channel>]
    curve="temp duty pairs"
    const_rpm="rpm value for constant speed if device supports it"
    const_duty="duty value for constant percent"
    mode="for pump speed if it is set using a mode keyword"
```

In order for the devices to be matched properly vendor, product, bus, and address  values must be specified if there is more than one device. 

cli flags that would be added:
- `--file, -f` to override the default configuration file path
- `--test` to test that all of the options specified in the config file are correct without actually changing any settings of the devices
- `--ignore-config` optional command to ignore the configuration file if it does exist and to use the default values instead

### Default configuration file paths. 

Will search for a config file in `$XDG_CONFIG_HOME/.config/liquidctl/config`,  `~/.config/liquidctl/config`, then `~/.liquidctl` respectively according to the [XDG Base Directory Specification](https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html)

olielvewen, 2020-10-19:
Interesting structure. Good work. 

And if we have already a config file personalized ? can we create another one ?

Another idea : choose between config files i.e default file, first personalized file, second personalized file ?

MarshallAsch, 2020-10-19:
> And if we have already a config file personalized ? can we create another one ?

@olielvewen thats what the `-f` option would be for, to allow the user to specify a different config file.

The config file location according to the XDG specification allows you to have multiple different potential locations for the file and it will check in different spots for the file and just use the first one it finds be default.

MarshallAsch, 2020-10-19:
@jonasmalacofilho are you okay with this format if I were to start working on implementing that?

jonasmalacofilho, 2020-10-19:
I haven't really looked into this yet.  So far I kind of like it, even there are some issues when you consider multiple devices (probably just need `[<device name>.color.<channel>]` or something like that).

I would also like a bit more time to think about how these settings will map to the liquidctl APIs in a generic manner.  I see the pattern key -> argument for the API calls, and that should work well, but I'm not sure whether the speed settings should match the CLI or the APIs (the latter are split between `fixed` and `profile`).

MarshallAsch, 2020-10-20:
> multiple devices

your right. all of those segments should have been subsections under devices. 

> api calls vs cli

I haven't looked all that closely at the api calls so far. Ill take a look at that more when I get a chance

MarshallAsch, 2021-04-04:
@jonasmalacofilho I know you have assigned this to yourself but I started an implementation for this, (currently just adding the basics). 

After thinking about the file format a bit more, I no longer think that the file format I proposed makes sense. 

```toml
config_version="1"

[global]
setting="value"
#...

[dev1]
vendor=".."
product=""
type="aio,ram,gpu,fan,psu"

    [dev1.pump]
        mode=""
        duty=""
        rpm=""

    [dev1.leds]
         [dev1.leds.channel1]
             color=["ffffff", "00ff00"]
             mode=""

    [dev1.fans]
        [dev1.fans.fan1]
            rpm=""
            duty=""
            profile=[ {temp: "", duty: "", rpm: ""},  {temp: "", duty: "", rpm: ""} ]
          #  ....
```

This updated format will allow multiple devices to have different light channels and fan channels that are configured independently. Some of the values that would conflict such as `fan rpm` and `fan duty` can be set to be mutually exclusive in the validator. 

The `global` section is for application wide configuration values (that either currently exist or might be added in the future.
The `config_version` is a string that is the version number for the configuration file. This will allow the config file format to have breaking changes without requiring  liquidctl getting a major version update.  

This file format will make some fields mandatory at each level while also allowing any extra values to be accepted for use as device or driver specific configuration options.

tbaumann, 2021-04-08:
I'm glad I found this thread before I spent too much time adjusting the yoda script to my needs.

Is it planned to integrate dynamic fan control? I guess it makes sense to start with device initialization.

There is no method to reliably address specific devices. This whole match syntax seems utterly inadequate to be used in a config-file. Predictable and stable device identifiers would be nice.
Not sure how though. The current names collide easily if you have multiple of the same device.
Any ideas?

Btw, this is roughly how my conffile looks now. But it's bascially unimplemented
```yaml
profile:
  silent:
    - (40,0)
    - (60,20
    - (79,50)
    - (80, 100)

devices:
  kraken:
    control:
      - fan:
          sensor: "k10temp.tdie"
          curve: "silent"
  smart:
    control:
      - fan1:
          sensor: "nvme.composite"
          curve: "silent"
      - fan1:
          sensor: "bar"
          curve: "silent"
  controller:
    control:
      - sync:
          sensor: "kraken.liquid"
          curve: "silent"
```
In the future I would have added `set:` entries per device to set at boot.
Plus some kind of equivalent to fancurves for LED to visually react to temps.

tbaumann, 2021-04-08:
My 02 Eurocent. This makes absolutely zero sense as a configfile in the home directory. It should be system-wide and therefore in `/etc/`

jonasmalacofilho, 2021-04-08:
@tbaumann,

> Is it planned to integrate dynamic fan control? I guess it makes sense to start with device initialization.

That's essentially orthogonal to this issue (although it would presumably benefit from it), but, no, we are not currently looking into integrating dynamic (e.g. depending on CPU temperature) fan or lighting control into the main `liquidctl` executable.

This has been considered in the past (#260), but the main technical reasons for _not_ doing it are: the difficulty to come up with a good interface (in the `liquidctl` executable it would still be expected to work _without_ a config file); and the absence of reliable cross-platform support (essentially we can only really support this on Linux, thanks to hwmon; maybe on Windows, with some limitations, but someone would need to contribute the annoying parts).

> There is no method to reliably address specific devices. This whole match syntax seems utterly inadequate to be used in a config-file. Predictable and stable device identifiers would be nice.
> Not sure how though. The current names collide easily if you have multiple of the same device.
> Any ideas?

By definition, yes, model names collide if you have multiples of the same device.  That's where `--pick` (depends on OS enumeration order) and other filters come in.

`--address` (usually, but not necessarily always combined with `--bus`) is also a good way to filter devices, although it still depends on the OS enumeration order, which may or may not be predictable, depending on the OS, bus and device.

`--usb-port` and `--serial` are more stable, but require, respectively, a non-HID USB device and a device with a proper serial number (that is, one that doesn't abuse the serial number descriptor for something else, like versioning).

On Linux it may also be possible to use an udev to use other information (like the USB port for a USB HID device) to pin the device to some particular name, which a script could then use to figure out the correct `--address` at runtime.

> This makes absolutely zero sense as a configfile in the home directory.

Hardly.

> It should be system-wide and therefore in `/etc/`

But I suppose you're right that we should _also_ search for it in `$XDG_CONFIG_DIRS` (default: `/etc`).

jonasmalacofilho, 2021-04-08:
@MarshallAsch,

Sorry for the delay...

> I know you have assigned this to yourself but I started an implementation for this, (currently just adding the basics).

That's absolutely fine, thanks.

> This updated format will allow multiple devices to have different light channels and fan channels that are configured independently. 

Yes, that's something that was missing before.

Small note: "pump" is not special in the drivers, it's just another common cooling channel name.  And perhaps the "cooling" and "lighting" (more readable), or "speed" and "color" (mimicking the APIs) are better names than the very specific "fan" and "leds" for the categories?

> Some of the values that would conflict such as `fan rpm` and `fan duty` can be set to be mutually exclusive in the validator.

Yes, but perhaps those could just be made into warnings, and we would just pick the last one set.

With a `global` section if feels to me like we're going to explore overrides, and at that point having something in the global section that slightly conflicts with, and is overridden by a later device section, feels like a common use case.

> The `config_version` is a string that is the version number for the configuration file. This will allow the config file format to have breaking changes without requiring liquidctl getting a major version update.

That's very good to have.

> This file format will make some fields mandatory at each level while also allowing any extra values to be accepted for use as device or driver specific configuration options.

Makes sense, although I wouldn't make too many things mandatory.  For now, I'd say that the only thing that's essential is, once a "dev1" is present, specifying what actual device it should correspond to.

tbaumann, 2021-04-08:
> @tbaumann,
>
>
> That's essentially orthogonal to this issue (although it would presumably benefit from it), but, no, we are not currently looking into integrating dynamic (e.g. depending on CPU [...]

Yea it's a mess. I see why you would not want it in the cli.
Maybe if it's a very loose (plugin style) dependency...

It's a pretty deep feature gap for non-windows users. And all solutions I found have issues playing nice with the rest of the ecosystem.
Anyway, not to be discussed in this issue.

> By definition, yes, model names collide if you have multiples of the same device. That's where `--pick` (depends on OS enumeration order) and other filters come in.
>
> `--address` (usually, but not [...]

> 
> On Linux it may also be possible to use an udev to use other information (like the USB port for a USB HID device) to pin the device to some particular name, which a script could then use to figure out the correct `--address` at runtime.
> 
Yes I suppose the device enumeration/selection issue will always be a problem. I myself don't really see anything that would solve that.
Maybe bus topology.

The ordering of the devices probably won't randomly change. So the current match methods will be stable. But using that in scripts or configfiles is a bit awkward. Anyway, can't be helped I guess. I was just fishing for ideas.

I am interested to see how you guys implement it here. I'm ready to steal your code. :)

jonasmalacofilho, 2021-04-08:
> It's a pretty deep feature gap for non-windows users. And all solutions I found have issues playing nice with the rest of the ecosystem.

I suppose the better solution on Linux may be, when possible, to use hwmon drivers (with pwm support).

I'm writing some for the devices I can personally test: [liquidtux](https://github.com/liquidctl/liquidtux).

tbaumann, 2021-04-08:
> @tbaumann,
> 
> > It's a pretty deep feature gap for non-windows users. And all solutions I found have issues playing nice with the rest of the ecosystem.
> 
> I suppose the better solution on Linux may be, when possible, to use hwmon drivers (with pwm support).
> 
> I'm writing some for the devices I can personally test: [liquidtux](https://github.com/liquidctl/liquidtux).

Oh, that's an interesting approach. It's a bit hardcore to put the driver in the kernel. But the benefits are clear because I can use fancontrol out of the box.

Your solution works whereas with the userspace drivers I need to add a ton of work to implement something that already exists.
Thanks man