liquidctl/liquidctl #200: Support for H100i original from 2012?

Jawk1995, 2020-10-19:
I've recently brought my old desktop back to life and its got an ax1500i and h100i I'd love to be able to control on ubuntu and found when I setup OpenCorsairLink they both were recognised, but don't seem to be with liquidctl.

Please let me know if there's some branch I'm missing, or if these should work at the moment or if I can assist in getting these working!

Thanks
Julius

jonasmalacofilho, 2020-10-19:
That would probably be covered by #142.  What are the USB vendor and product IDs?

jonasmalacofilho, 2020-10-22:
It's probably `1b1c:0c04`,<sup>[[1]](https://github.com/audiohacked/OpenCorsairLink/blob/61d336a61b85705a5e128762430dc136460b110e/device.c#L120-L132)</sup> which is indeed covered by the issue I mentioned before.

closed 2020-10-22 by jonasmalacofilho