liquidctl/liquidctl #201: Fan settings affects Fan 1 only and disables fan2

baldyman01, 2020-10-19:
Linux MONSTER 5.4.0-51-generic #56-Ubuntu SMP Mon Oct 5 14:28:49 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
liquidctl 1.4.1  +  Corsair H100i RGB Platinum - Bus 001 Device 004: ID 1b1c:0c18 Corsair H100i Platinum

Setting fan1 appears to do nothing, setting fan or fan2 sets fan1 to the selected speed and disabled fan2. It's physically stopped.
Nothing will then start fan2 again apart from a power down.

pete@MONSTER:~$ liquidctl initialize
Corsair H100i Platinum (experimental)
└── Firmware version    1.0.32  

pete@MONSTER:~$ liquidctl list
Device ID 0: Corsair H100i Platinum (experimental)

pete@MONSTER:~$ liquidctl status
Corsair H100i Platinum (experimental)
├── Liquid temperature    29.9  °C
├── Fan 1 speed           2658  rpm
├── Fan 2 speed             rpm
└── Pump speed            1895  rpm

pete@MONSTER:~$ liquidctl set fan2 speed 50
pete@MONSTER:~$ liquidctl status
Corsair H100i Platinum (experimental)
├── Liquid temperature    31.5  °C
├── Fan 1 speed           1408  rpm
├── Fan 2 speed              0  rpm
└── Pump speed            1895  rpm

pete@MONSTER:~$
pete@MONSTER:~$ liquidctl set fan1 speed 80
pete@MONSTER:~$ liquidctl status
Corsair H100i Platinum (experimental)
├── Liquid temperature    31.8  °C
├── Fan 1 speed           1408  rpm
├── Fan 2 speed              0  rpm
└── Pump speed            1895  rpm

pete@MONSTER:~$ liquidctl status
Corsair H100i Platinum (experimental)
├── Liquid temperature    31.9  °C
├── Fan 1 speed           1408  rpm
├── Fan 2 speed              0  rpm
└── Pump speed            1894  rpm

pete@MONSTER:~$ ^C
pete@MONSTER:~$ liquidctl set fan speed 80

pete@MONSTER:~$ liquidctl status
Corsair H100i Platinum (experimental)
├── Liquid temperature    32.4  °C
├── Fan 1 speed           2183  rpm
├── Fan 2 speed              0  rpm
└── Pump speed            1893  rpm

pete@MONSTER:~$ liquidctl initialize
Corsair H100i Platinum (experimental)
└── Firmware version    1.0.32  

pete@MONSTER:~$ liquidctl status
Corsair H100i Platinum (experimental)
├── Liquid temperature    30.7  °C
├── Fan 1 speed           2662  rpm
├── Fan 2 speed              0  rpm
└── Pump speed            1893  rpm

jonasmalacofilho, 2020-10-19:
@MarshallAsch 

Can you reproduce this with your cooler?

MarshallAsch, 2020-10-19:
I have not experienced this behaviour or been able to replicate it on my machine, (I also cant get my fan speed to go to 0 using liquidctl). 

- Fedora 31
- kernel version 5.8.31
- liquidctl 1.4.1
- Corsair H100i Platinum
- firmware version 1.1.15

@baldyman01 are you able to update the firmware version on the cooler to see if that resolves the problem?

@jonasmalacofilho is it possible that the something is different with that firmware version?

baldyman01, 2020-10-19:
Gents, 

Well that was fun. After utterly failing to get the corsair software running under wine or VirtualBox, I ended up taking the side off and cross connecting the cooler to a Windows laptop. They really did not make that easy....

Update to 1.1.5 now and can confirm this fixes it. 

Thanks all.

Pete

jonasmalacofilho, 2020-10-19:
@baldyman01, ok, thanks for letting us know.

@MarshallAsch thanks for you help here too.

closed 2020-10-19 by jonasmalacofilho