liquidctl/liquidctl #202: NZXT X53 not showing up

thibaultcha, 2020-10-20:
Hello,

liquidctl seems to not be able to locate my installed Kraken X53, even after the suggested usbhid workaround.

```
$ liquidctl --version
liquidctl v1.3.3 (Fedora 32; liquidctl-1.3.3-1.fc32.noarch)
```

Output of `liquidctl initialize`:
```
$ liquidctl --hid hid --debug initialize all
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (Fedora 32; liquidctl-1.3.3-1.fc32.noarch)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1702
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1702
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x4d9:0x141
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x4d9:0x141
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x4d9:0x141
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x48d:0x8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2007
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1702
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x4d9:0x141
[DEBUG] liquidctl.driver.usb: probing drivers for device 0xbda:0x5411
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1a40:0x101
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x8087:0xaaa
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x48d:0x8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2006
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.cli: device: NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 60 02 01 e8 03 01 e8 03 (and 56 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 60 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 10 01 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: write 20 03 (and 62 padding bytes)
[DEBUG] liquidctl.driver.nzxt_smart_device: received 67 04 44 00 28 00 06 57 42 48 35 37 31 20 03 00 02 01 01 00 00 00 00 00 04 2f 8c 1f f9 1e 00 00 00 00 00 00 00 00 00 00 3e 00 3c 00 42 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 11 01 44 00 28 00 06 57 42 48 35 37 31 20 06 20 01 01 0c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 61 03 44 00 28 00 06 57 42 48 35 37 31 20 03 00 02 01 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[DEBUG] liquidctl.driver.nzxt_smart_device: received 21 03 44 00 28 00 06 57 42 48 35 37 31 20 02 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
NZXT Smart Device V2 (experimental)
├── Firmware version                     1.12.0  
└── LED 1 accessory 1    HUE 2 LED Strip 300 mm  
```

Output of `lsusb -tv`:
```
$ lsusb -tv
[...]
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    [...]
    |__ Port 11: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0608 Genesys Logic, Inc. Hub
        |__ Port 3: Dev 9, If 0, Class=Human Interface Device, Driver=, 12M
            ID 1e71:2006 NZXT Smart Device V2
        |__ Port 2: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1e71:2007 NZXT 
[...]
```

After running `echo -n '1-11.3:1.0' > /sys/bus/usb/drivers/usbhid/bind`, the output slightly changed to:
```
$ echo -n '1-11.3:1.0' > /sys/bus/usb/drivers/usbhid/bind
$ lsusb -tv
[...]
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    [...]
    |__ Port 11: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0608 Genesys Logic, Inc. Hub
        |__ Port 3: Dev 9, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1e71:2006 NZXT Smart Device V2
        |__ Port 2: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1e71:2007 NZXT 
[...]
```

Is it an issue that the Kraken is connected to an internal USB hub? Am I not executing the workaround properly? Thanks in advance.

jonasmalacofilho, 2020-10-20:
Hi there!

The issue is that you're on liquidctl 1.3.3, but 1.4.x (currently 1.4.1) is necessary for support to this device.

thibaultcha, 2020-10-20:
@jonasmalacofilho Gotcha! I'll just wait for the package to be updated in this case. Thank you :)

closed 2020-10-20 by thibaultcha

jonasmalacofilho, 2020-10-20:
That may take a while, one of the dependencies is partially broken in the version that is shipped in Fedora 31/32.  Take a look at the discussion in #195.

thibaultcha, 2020-10-20:
I see. Well no big deal, this can wait. Thanks for the prompt response!

jonasmalacofilho, 2020-10-20:
In the mean time you can install liquidctl 1.4.1in a Python virtual environment:

```
$ python -m venv <path: to use for the virtual environment>
$ <path>/bin/pip install liquidctl
$ <path>/bin/liquidctl --version
```