liquidctl/liquidctl #203: Kraken x62 Fans are tracking liquid rather than CPU temps

Tanhe, 2020-10-20:
Can't seem to find a method in the docs to toggle between fan speed based on CPU vs Liquid temps.

Thanks.

jonasmalacofilho, 2020-10-20:
The CLI only exposes the native capabilities of the devices, and the CPU temperature is not available to them. But the liquidctl APIs allow you to do this, as long as you have some way to get the CPU temperature from the OS.

You can find a couple of scripts in `extra/` that showcase how this works and, if you're on Linux or Mac OS, they are even usable as is: just check `--help` for the necessary dependencies.  Of the two scripts that do this, [`extra/yoda`](https://github.com/jonasmalacofilho/liquidctl/blob/master/extra/yoda) is more capable and recommended (the other one would be `extra/krakencurve-poc`).

closed 2020-10-20 by jonasmalacofilho

Tanhe, 2020-10-20:
Thanks for the info. That makes total sense. Do you know if the Kraken cares if it has fans connected at all? I can use your awesome software to configure the pump and rgb and just control the fans from the MB otherwise. Thanks again for your hard work on this project.

jonasmalacofilho, 2020-10-21:
It doesn't care at all, you should be able to use the MB without any issues.

And thanks! : )

commit e1cdcb68 (2019-11-17): Patch PyUSB and upgrade libusb to 1.0.22 on Windows builds

The patch for PyUSB successfully prevents access violation errors during
cleanup with libusb 1.0.22, but "some libusb_devices were leaked"
warnings can still be seen with LIBUSB_DEBUG=3.

Patch: 8041ec11b2c4 ("Fix #203: libusb sometimes cleaned up too early.")
Related: pyusb/pyusb#227 ("Fix #203: libusb sometimes cleaned up too early.")
Related: pyusb/pyusb#203 ("OSError with wxPython #203")