liquidctl/liquidctl #204: Thoughts on a bash completion script

MarshallAsch, 2020-10-22:
I have recently been looking at bash tab completion and I was wondering if this would be something your interested in having added to this project before I work on that much more. 

Personally I really like tab completion because I don't have to remember the entire command I want to type and it will give hints. 

Im not sure how great my implementation will be, but I think it will be better than no completions.

jonasmalacofilho, 2020-10-22:
Absolutely!

---

On a related note, in my `.zshrc` I'm using `compdef _gnu_generic liquidctl` to have some completion be automatically generated from the output of `--help`.  But it's limited to options, so neither commands nor device specific details are suggested.

```
$ liquidctl --<TAB>
--address          -- Filter devices by address in bus
--alert-color      -- Color used by the visual high temperature alert
--alert-threshold  -- Threshold temperature for a visual alert (°C)
--bus              -- Filter devices by bus
--debug            -- Show debug information on stderr
--device           -- Select device by listing id
--help             -- Show this message
--hid              -- Deprecated
--legacy-690lc     -- Use Asetek 690LC in legacy mode (old Krakens)
--match            -- Filter devices by description substring
--pick             -- Pick among many results for a given filter
--product          -- Filter devices by product id
--pump-mode        -- Set the pump mode (certain Corsair coolers)
--release          -- Filter devices by release number
--serial           -- Filter devices by serial number
--single-12v-ocp   -- Enable single rail +12V OCP
--speed            -- Abstract animation speed (device/mode specific)
--time-off         -- Time to wait with the LED turned off (seconds)
--time-per-color   -- Time to wait on each color (seconds)
--unsafe           -- Comma-separated bleeding-edge features to enable
--usb-port         -- Filter devices by USB port in bus
--vendor           -- Filter devices by vendor id
--verbose          -- Output additional information
--version          -- Display the version number
```

MarshallAsch, 2020-10-22:
Thats cool I didn't know it did it automatically, I know that zsh has a far more powerful autocomplete functionality, but I use bash and haven't used it very much. 

In #205, I did not add any device specific completions, but I did go through all of the files in `/doc` to pull out the different channels for each of the devices.  Although I did add the commands in there are a couple things that I did not implement yet.

PR #205 (2020-11-24): Bash completion
Closes: #204 
Added a basic bash completion script. 

Tasks that this does / should have:

- [X] all of the flags
- [X] the arguments for most of the flags that take arguments for specifying a device
- [X] each of the commands and the sub commands that are currently implemented
- [X] all of the channels under the set command (note that these are not device specific suggestions rather just all the currently accepted options)
- [x] Remove flags that have already been specified from the suggestions
- [x] allow flags to be given after a command, might not actually want this tab com [... cut]
closed 2020-11-24 by jonasmalacofilho