liquidctl/liquidctl #208: NZXT x63 not showing Up

Divadi, 2020-10-30:
When I first set up my PC and used liquidctl yesterday, it found the x63 and I was able to set up fan curves for it.
Today, however, liquidctl cannot seem to find it. What could be the issue?
Also, even if I can't find the x63 anymore, is the fan setup I created yesterday still in use?

`[DEBUG] liquidctl.cli: running liquidctl v1.4.1
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: instanced driver for Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 8087:0029
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 048d:8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.cli: device: Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
[DEBUG] liquidctl.driver.usb: sending feature report 0xcc with 63 bytes: 60:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[DEBUG] liquidctl.driver.usb: got feature report 0x00 with 63 bytes: 01:00:01:01:00:06:00:00:00:00:00:49:54:38:32:39:37:42:58:2d:47:42:58:35:37:30:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02:00:01:00:02:01:00:00:00:01:02:00:00:01:97:82:00:00:00:00
Gigabyte RGB Fusion 2.0 8297 Controller (experimental)
├── Hardware name       IT8297BX-GBX570  
└── Firmware version            1.0.6.0 `

jonasmalacofilho, 2020-10-30:
> Today, however, liquidctl cannot seem to find it. What could be the issue?

Most commonly something else is unbinding the normal `usbhid` driver from the device.  Can you post the output of `lsusb -tv`?

> Also, even if I can't find the x63 anymore, is the fan setup I created yesterday still in use?

Perhaps if the PSU was kept on, but it also depends on how your motherboard is configured.  If the cooler was powered off, it's generally is better to assume it also reset.

That said, if your RGB settings persisted (and you're not using other software for that), then the pump profile should have persisted as well.

---

> When I first set up my PC and used liquidctl yesterday, it found the x63 and I was able to set up fan curves for it.

There's no fan controller (or header) on the X63.  Did you mean a pump curve, or do you have a X62?

Divadi, 2020-10-30:
Here is my lsusb -tv output:
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 480M
    |__ Port 5: Dev 2, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 5: Dev 2, If 1, Class=Wireless, Driver=btusb, 12M
    |__ Port 6: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 480M
    |__ Port 1: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 6: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M

> Also, even if I can't find the x63 anymore, is the fan setup I created yesterday still in use?
I hadn't set any RGB settings, so I'm not sure. PSU was kept on. Does this mean I have to re-set the setup every boot?

> There's no fan controller (or header) on the X63. Did you mean a pump curve, or do you have a X62?
Ah, I meant the pump curve. I just copy-pasted
liquidctl set pump speed 20 30 30 50 34 80 40 90 50 100

jonasmalacofilho, 2020-10-30:
> Here is my `lsusb -tv` output:

You're missing the verbose output from the `v` flag; try adding it twice then (`-tvv`).

> I hadn't set any RGB settings, so I'm not sure. PSU was kept on. Does this mean I have to re-set the setup every boot?

That's generally better, yes.  You may want to take a look at [_Automation and running at boot_](https://github.com/jonasmalacofilho/liquidctl#automation-and-running-at-boot).

> Ah, I meant the pump curve.

Oh, ok, that makes sense.

Divadi, 2020-10-30:
Does the x63 have its own default pump curves? I have 3900x under load right now and it's hovering 61 - 71. I wonder if the x63 is doing work already

Here is lsusb -vv: (I also tried lsusb -tvv, but it seems the same as -tv?)
(base) divadi@divadi-linux:~$ lsusb -vv

Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            5.04
  iManufacturer           3 
  iProduct                2 
  iSerial                 1 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           31
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0

Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
  bcdDevice            5.04
  iManufacturer           3 
  iProduct                2 
[... 643 lines cut ...]
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12

damienbr, 2020-11-02:
Hello,

I'm also facing issues with X63 not showing up. I already tried to bind/unbind but it doesn't solve the issue.
The NZXT Smart Device is properly listed but not the X63. Even though, I'm able to control the x63 RGB through openrgb.

```console
# liquidctl list --debug
[DEBUG] liquidctl.cli: running liquidctl v1.3.3 (Fedora 32; liquidctl-1.3.3-1.fc32.noarch)
[DEBUG] liquidctl.driver.usb: searching GenericHidBus (api=hid, drivers=[CommonSmartDeviceDriver, CorsairHidPsuDriver, KrakenTwoDriver, SeasonicEDriver, SmartDeviceDriver, SmartDeviceV2Driver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1369
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1369
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1369
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1369
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x41e:0x324d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x835:0x8502
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x4d9:0x169
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x4d9:0x169
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x835:0x8501
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2006
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Smart Device V2 (experimental)
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x48d:0x8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2007
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[AsetekDriver, CommonAsetekDriver, CorsairAsetekDriver, LegacyAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1038:0x1369
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x41e:0x324d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x835:0x8502
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x835:0x8500
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x72f:0x9000
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x4d9:0x169
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x46d:0x82d
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x835:0x8501
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x835:0x8500
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2006
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x3
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x48d:0x8297
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1e71:0x2007
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x5e3:0x608
[DEBUG] liquidctl.driver.usb: probing drivers for device 0x1d6b:0x2
Device ID 0: NZXT Smart Device V2 (experimental)
├── Vendor ID: 0x1e71
├── Product ID: 0x2006
├── Release number: 0x0200
├── Serial number: 00000000001A
├── Bus: hid
├── Address: 0003:0003:00
└── Driver: SmartDeviceV2Driver using module hid
[DEBUG] liquidctl.cli: hierarchy: CommonSmartDeviceDriver, UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice

```

```console
# lsusb -tv
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 6: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0608 Genesys Logic, Inc. Hub
        |__ Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=, 12M
            ID 1e71:2006 NZXT Smart Device V2
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 10000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 05e3:0608 Genesys Logic, Inc. Hub
        |__ Port 1: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
            ID 1e71:2007 NZXT 
    |__ Port 4: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        ID 048d:8297 Integrated Technology Express, Inc. IT8297 RGB LED Controller
[... more devices ...]
```
[image]

jonasmalacofilho, 2020-11-02:
@damienbr 

The X63 is only supported by liquidctl 1.4.0 or later.

damienbr, 2020-11-02:
My bad, thanks

jonasmalacofilho, 2020-11-07:
@Divadi,

Were you able to find anything in the kernel logs?

Divadi, 2020-11-07:
Not sure...
It somehow appeared again in the list. I will debug again if something goes wrong

jonasmalacofilho, 2020-11-07:
Ok, thanks for letting me know.

closed 2020-11-07 by jonasmalacofilho