liquidctl/liquidctl #209: cannot detect h100i platinium on ubuntu 20.04

digrouz, 2020-11-02:
Hello,

It seems liquidctl does not detect my h100i platinium while it's listed by lsusb. Am I missing something ?

```
~$ liquidctl list -g
[DEBUG] liquidctl.cli: running liquidctl v1.4.2
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hidraw, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 0463:ffff
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:4082
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:408a
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
[DEBUG] liquidctl.driver.usb: probing drivers for device 0a5c:21e8
[DEBUG] liquidctl.driver.usb: probing drivers for device 046d:c52b
[DEBUG] liquidctl.driver.usb: probing drivers for device 0463:ffff
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0003
**[DEBUG] liquidctl.driver.usb: probing drivers for device 1b1c:0c15**
[DEBUG] liquidctl.driver.usb: probing drivers for device 1d6b:0002
~$ lsusb
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0
Bus 003 Device 063: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 084: ID 0463:ffff MGE UPS Systems UPS
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
**Bus 001 Device 002: ID 1b1c:0c15 Corsair H100i Platinum**
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

jonasmalacofilho, 2020-11-02:
This device has a misleading product descriptor.  While there is a later H100 Platinum, `1b1c:0c15` was sold by Corsair as the H100 PRO.  We don't yet support it (see #137).

closed 2020-11-02 by jonasmalacofilho

digrouz, 2020-11-02:
Oh ok, weird one.