liquidctl/liquidctl #210: Kraken X63 shows 1% duty all the time and doesn't react on changes

Mettkrieger, 2020-11-06:
Hey,

just installed liquidctl via pip (running Version 1.4.2) and I can't control the pump of my kraken x63. It always shows pump duty 1% with varying speeds.
Is there any debug mode I could enablt to help investigate this?

Mettkrieger, 2020-11-06:
Works after restarting

closed 2020-11-06 by Mettkrieger

jonasmalacofilho, 2020-11-06:
Before restarting, did you run `liquidctl initialize`?

Mettkrieger, 2020-11-06:
Yes I did

jonasmalacofilho, 2020-11-07:
Ok, thanks for clarifying that.

If the problem comes back (and answering your original question), try to gather some information with the `--debug` flag (e.g. `liquidctl status --debug`).

Additionally, can you tell me which firmware version your Kraken is running?

Mettkrieger, 2020-11-07:
Hey, there seems to be no issues anymore. I can't even replicate them.

Can't tell you whihc firmware I'm using since I never installed kraken. Or is there a liquidctl command for that? I could of course run that one ;)

On a side note: If I use task scheduler to start the batch file will this not show the batch file command prompt?

jonasmalacofilho, 2020-11-07:
With the X63, the firmware version is part of the output of `liquidctl initialize`.

---

Task Scheduler has an option to log the output of tasks, otherwise it discards them.

You can also do your own logging by redirecting stdout (normal output) and stderr (errors, warnings, logging) to append to a file.  I'm not sure about the precise syntax in Windows Command Prompt/batch files (you should look for some official documentation), but it probably looks something like:

```
liquidctl <command> [options] >>liquidctl.log 2>>liquidctl.log
```

For example:

```
liquidctl set pump speed 90 >>liquidctl.log 2>>liquidctl.log
liquidctl set fan speed  20 30  30 50  34 80  40 90  50 100 >>liquidctl.log 2>>liquidctl.log
liquidctl set ring color fading 350017 ff2608 >>liquidctl.log 2>>liquidctl.log
liquidctl set logo color spectrum-wave >>liquidctl.log 2>>liquidctl.log
```

You could also use some other command to output the current datetime at the beginning of the run, or use different log files for each boot...