liquidctl/liquidctl #211: liquidctl doesn't detect my x72

wqkl, 2020-11-08:
Running a 5600x on x72, liquidctl doesn't detect the x72 usb
[image]

wqkl, 2020-11-08:
Here is another screenshot I took earlier
https://imgur.com/a/1zY2kVh

jonasmalacofilho, 2020-11-08:
You need to restore the original driver (Microsoft Generic HID Driver).

You can find more information about when to use custom drivers, and when not to, in the README, specifically in the list of supported devices and in the Windows and Troubleshooting sections.

wqkl, 2020-11-08:
I have changed it to WinUSB, yet it still doesn't detect it, sorry still learning python in campus
[image]

jonasmalacofilho, 2020-11-08:
You need to uninstall the custom driver, and let Windows restore a default one.  The X72 is a USB HID, and liquidctl uses [HIDAPI](https://github.com/libusb/hidapi) to communicate with HIDs; but this only works with the (original) Microsoft Generic HID Driver.

closed 2020-11-08 by jonasmalacofilho