liquidctl/liquidctl #214: XPG Levante 240 AIO

radianttap, 2020-11-09:
[This AIO](https://www.xpg.com/en/feature/644/) seems to be using Asetek 6th gen pump. It seems pretty high-end and I just took delivery on it since it's 27mm radiators which is important point for SFF builds. Waiting on CPU now so I can try it out but I am keen to add support for it into liquidctl, especially since I'll use this as part of a Hackintosh.

I'll try to provide more details as I figure out how (after reading the dev docs).

jonasmalacofilho, 2020-11-09:
It's nice that XPG includes an exploded view in their marketing material.  It does look like a sixth gen. design, but it could also be a 7th gen. one, both are quite similar.

Anyway, the main issue I see is with supporting this device in liquidctl is that it doesn't appear to be controlled through USB: in both the marketing material and the manual I can only find references to standard PWM and ARGB connectors.

If that's indeed the case, then what you'll need support for is your motherboard (or external fan and/or RGB controllers, if you're using those).

radianttap, 2020-11-10:
Huh...thanks for letting me know. The mobo does have support (I think) but was hoping I could control or at least monitor the pump speed. We'll see, soon (5900X is proving harder to find around here than I anticipated).

jonasmalacofilho, 2020-11-11:
I see.  Since you already have the cooler, can you check if it has a USB interface that's not mentioned in the manual?

radianttap, 2020-11-11:
Sadly no, only what's listed there already: power cable and ARGB cable.

jonasmalacofilho, 2020-11-11:
Oh : \

What board are you going to use?

radianttap, 2020-11-11:
Plan for now is ASRock X570 ITX/TB3.

jonasmalacofilho, 2020-11-11:
Well, once you have the board, we can see if it's possible to support it here.

In the meantime I'm going to close the issue since there's nothing that can be done for the cooler (specifically).

closed 2020-11-11 by jonasmalacofilho