liquidctl/liquidctl #215: usb.core.NoBackendError - MacOS

JakubKnezinek, 2020-11-10:
Hello,

I have problem with connection to my NZXT kraken x72 in MacOS. Do you have any ideas why it is not function?
I have dual boot system. NZXT Cam in Windows is working perfect with Kraken.

Thank you very much in advance for your answer.

```
jakub@192 ~ % pip3 list -v
Package Version Location Installer
docopt 0.6.2 /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages
hidapi 0.10.0.post1 /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages
libusb 1.0.23b6 /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages
liquidctl 1.4.2 /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages
pip 20.2.4 /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages pip
pyusb 1.1.0 /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages pip
setuptools 49.2.1 /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages pip
jakub@192 ~ % liquidctl list -v
Traceback (most recent call last):
File "/Library/Frameworks/Python.framework/Versions/3.9/bin/liquidctl", line 33, in <module>
sys.exit(load_entry_point('liquidctl==1.4.2', 'console_scripts', 'liquidctl')())
File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/liquidctl/cli.py", line 270, in main
selected = list(find_liquidctl_devices(**opts))
File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/liquidctl/driver/__init__.py", line 51, in find_liquidctl_devices
for dev in  bus_cls().find_devices(**kwargs):
File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/liquidctl/driver/usb.py", line 521, in find_devices
for handle in PyUsbDevice.enumerate(vendor, product):
File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/liquidctl/driver/usb.py", line 299, in enumerate
for handle in usb.core.find(find_all=True, **args):
File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/usb/core.py", line 1297, in find
raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
jakub@192 ~ % liquidctl list   
usb.core.NoBackendError: No backend available
jakub@192 ~ % liquidctl status 
usb.core.NoBackendError: No backend available
jakub@192 ~ %

jonasmalacofilho, 2020-11-10:
You missing the `libsub` dynamic library.  Have you installed liquidctl with homebrew or pip?

JakubKnezinek, 2020-11-10:
Hello and thank you

How can I install `libsub` dynamic library?
I already install `libusb` with `pip3 install libsub` command in homebrew.
I install liquidctl with command `pip3 install /Users/jakub/Applications/libusb` in homebrew.

Is it correct way?

Thank you for your help

jonasmalacofilho, 2020-11-11:
There are more complete instructions in the readme<sup>[1](https://github.com/jonasmalacofilho/liquidctl#installing-on-macos)</sup>, but the main problem here is that you need the system library, not the PyPI wrapper with the same name: `brew install libusb`.

jonasmalacofilho, 2020-11-11:
I'm closing this, since you'll probably get it to work once you `brew install libusb`, but please feel free to ask more questions.

closed 2020-11-11 by jonasmalacofilho

JakubKnezinek, 2020-11-12:
Yes, it work's :)
Thank you for you help!