liquidctl/liquidctl #216: MacOS (Catalina) LaunchDaemon fails with segmentation fault

glassworks-projects, 2020-11-11:
Hi, I tried setting up a LaunchDaemon using [this shell script](https://github.com/icedterminal/ga-z270x-ug/blob/master/post_install/pump_control/system-setup.sh) as a starting point. It failed, so I added some lines to the .plist to log stderr and stdout to files. My stderr log file showed that every liquidctl command resulted in a segmentation fault: 11: 

`/usr/local/share/LiquidCfg.sh: line 8:   240 Segmentation fault: 11  /usr/local/bin/liquidctl initialize`
`/usr/local/share/LiquidCfg.sh: line 9:   250 Segmentation fault: 11  /usr/local/bin/liquidctl set ring color fading ea5c1f ffda1f 0e34a0`
`/usr/local/share/LiquidCfg.sh: line 10:   251 Segmentation fault: 11  /usr/local/bin/liquidctl set pump speed 30 30 35 50 40 80 50 100`

I tried modifying my Python path to `usr/local/bin/python3`, then removing it entirely, per #196. I also tried changing the chmod command for LiquidCfg.sh from `chmod 777` to `chmod +x` in my setup script. None of these fixed the issue. 

I'm on a hackintosh, running Catalina 10.15.7. Setting the script as a [login item](https://www.tonymacx86.com/threads/gigabyte-z490-vision-d-thunderbolt-3-i5-10400-amd-rx-580.298642/page-24#post-2138475) works fine, but ideally I'd like this to launch at startup, not login, and be user-independent.

jonasmalacofilho, 2020-11-11:
Can you add `--debug` to all liquidctl invocations?  This will give us a general ideal on where the problem may lie.

glassworks-projects, 2020-11-11:
Done, here's the stderr output now:

[DEBUG] liquidctl.cli: running liquidctl v1.4.2 (homebrew; liquidctl 1.4.2)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
/usr/local/share/LiquidCfg.sh: line 8:   240 Segmentation fault: 11  /usr/local/bin/liquidctl initialize --debug
[DEBUG] liquidctl.cli: running liquidctl v1.4.2 (homebrew; liquidctl 1.4.2)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
/usr/local/share/LiquidCfg.sh: line 9:   254 Segmentation fault: 11  /usr/local/bin/liquidctl set ring color fading ea5c1f ffda1f 0e34a0 --debug
[DEBUG] liquidctl.cli: running liquidctl v1.4.2 (homebrew; liquidctl 1.4.2)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
/usr/local/share/LiquidCfg.sh: line 10:   255 Segmentation fault: 11  /usr/local/bin/liquidctl set pump speed 30 30 35 50 40 80 50 100 --debug

jonasmalacofilho, 2020-11-11:
These are surprisingly short.  Just to double check: have you omitted any lines?  Assuming you haven't, we have a problem in cython-hidapi (either directly there, or in its underlying dependency, hidapi).

For comparison, can you show me the output of the same script (with `--debug`) when running as a login item, or manually from a terminal shell?

glassworks-projects, 2020-11-11:
I was surprised too, but that's everything! Here's the output when run manually from a terminal shell: 
[output.txt](https://github.com/jonasmalacofilho/liquidctl/files/5525545/output.txt)

jonasmalacofilho, 2020-11-11:
Thanks!

The fact that the bug only reproduces in the hard-to-reason-about system context of a global agent could suggest a simple incompatibility between the version of cython-hidapi that was compiled/installed with liquidctl, and the actual Python runtime being picked in that context.

As you've seen in #196, we don't really know what's going on there.  If you can find any documentation related to this, it would be a great help.

Still, another possibility is a memory management bug in cython-hidapi/hidapi.  The global agent runs a bit earlier in the boot process, and the device could not be ready yet; this shouldn't be a problem in itself, but maybe there's something else that fails to consider this scenario and ends doing an illegal memory access.

Do you know how to get a stack trace of that segfault on mac OS? (Unfortunately I don't)

glassworks-projects, 2020-11-11:
Hmm, I don't. I tried [this](https://stackoverflow.com/a/21308843), but nothing got written to /cores. I did find the .crash files, though. Here's the latest one, in .txt format: 

[crashfile.txt](https://github.com/jonasmalacofilho/liquidctl/files/5526760/crashfile.txt)

jonasmalacofilho, 2020-11-11:
That helps!

It does seem like we're in the latter case... I'll report this upstream (cython-hidapi) and maybe with their help we can figure out what's causing this.

jonasmalacofilho, 2020-11-12:
Can you try the suggestion mentioned in the upstream issue?

To do that you'll need to:

- uninstall homebrew's liquidctl
- install cython-hidapi from git first: `pip3 install git+https://github.com/trezor/cython-hidapi.git`
- and only then: `pip3 install liquidctl`

Note that this will normally pollute your global Python environment.  You could use a virtual environment, but then you'll also need to execute liquidctl from that environment.

```
# assuming <path> is where you want to the create the virtual environment

# create the virtual environment
python -m venv <path>

# install cython-hidapi and liquidctl
<path>/bin/pip3 install git+https://github.com/trezor/cython-hidapi.git
<path>/bin/pip3 install liquidctl

# use liquidctl
<path>/bin/liquidctl --version

# note: in you boot script you'll need to use an absolute path to the virtual environment
```

glassworks-projects, 2020-11-12:
I'll try this now. I'm going the virtual environment route, but I'm still a little unclear on how to manage python virtual envs. In the launch daemon, what should I set my PATH variable to? Or does it not matter?

jonasmalacofilho, 2020-11-12:
You'll need to set it to the path to the virtual environment.

glassworks-projects, 2020-11-12:
Same behavior; script works in terminal but fails at launch. 
[crash2.txt](https://github.com/jonasmalacofilho/liquidctl/files/5533910/crash2.txt)

jonasmalacofilho, 2020-11-13:
I honestly expected that, but we had to give it a try.

I'm thinking the issue may be in the set returned by `IOHIDManagerCopyDevices`, in [`libusb/hidapi/mac/hid.c#L585-L588`](
https://github.com/libusb/hidapi/blob/8ed36853ba08d82ebce3bd27f7c510db87fa7b0a/mac/hid.c#L585-L588).  While not clearly documented behavior, an obvious thing to check would be if it isn't a `NULL`.

However, if that's the case, it also means that we simply wont be able to enumerate HIDs at that time... for some unknown reason.

_(will copy past this into the upstream cython-hidapi issue)_

jonasmalacofilho, 2020-11-24:
@glassworks-projects 

Hi again,

I'm not sure if you were following the upstream discussion, but here's the gist:

- the segmentation fault should be fixed (pending cython-hidapi updating their submodule),
- but no device will be found.

Unfortunately none of us (liquidctl, cython-hidapi, hidapi) knows why no HIDs are enumerated when running from a boot agent.  One idea is that the agent might _not_ have run as root, although I thought that's the default for global agents.

Alternatively, the agent may be starting before the kernel HID subsystem is ready (which could explain the never seen before segfault), so perhaps it's just a matter of adjusting the agent's dependencies.  Since you have a mac, can you look into this?

closed 2020-12-10 by jonasmalacofilho