liquidctl/liquidctl #217: Confusion about Kraken 4th Gen X Series fan control...

Invertex, 2020-11-12:
The docs state that these coolers no longer provide fan control, but that's confusing to me since these coolers require you to plug a cable into the CPU fan header and have male fan header cable to control the radiator fans... So how exactly is that not "fan control"?  

If I use this software, does that mean I shouldn't even use those connections then and just control rad fans from main fan controller? It's got me a bit confused caused it feels contradictory to NZXT's setup instructions.

MarshallAsch, 2020-11-12:
Which document are you talking about and which specific device? (I'm not personally all that familiar with those specific devices I have mostly been working with the Corsair portion)

jonasmalacofilho, 2020-11-12:
@Invertex 

On the latest generation of Kraken X coolers, the fans can only be connected directly to the motherboard (or external fan controller).

This is different to the previous generation, as well as the current Kraken Z coolers, which have an additional fan header receptacle coming out of the cooler's CPU block, and allow the cooler itself to monitor and set fan speeds, as well as expose those capabilities to CAM (and liquidctl) through the USB interface.

To clarify: on the current X(3) models the fans aren't connected to the cooler.  The ability to use either liquidctl or CAM to control them will depend on support for whatever motherboard or external fan controller you connect them to.

Invertex, 2020-11-12:
> Which document are you talking about and which specific device? (I'm not personally all that familiar with those specific devices I have mostly been working with the Corsair portion)
This guide:
https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/kraken-x3-z3-guide.md

> @Invertex
>
> On the latest generation of Kraken X coolers, the fans can only be connected directly to the motherboard (or external fan controller).
>
> This is different to the previous generation, as [...]

Ahh okay, I took the page to imply that the X3 were 4th gen and so X2 would be "3rd gen". But what you are saying is that there have been different revisions of the X3 model, not so much generations? Because I just bought my X73 a month ago and it has the CPU header connection and fan control cable.

jonasmalacofilho, 2020-11-12:
No, not at all.  There are two revisions of the X2 coolers, but (AFAIK) only the one revision of the X3 coolers.

Can you show me some pictures of the headers and cables your cooler has?

Can you also list where each component is connected to?  Something like:

```
fan 1 -> fan spltiter receptacle 1
fan 2 -> fan splitter receptacle 2
fan 3 -> fan splitter receptacle 3
fan splitter connector -> motherboard SYSFAN header

pump fan-style connector -> motherboard CPUFAN header

pump USB port -> motherboard USB 2.0 header
```

Invertex, 2020-11-12:
Yeah I definitely misremembered my build process. I guess there is no fan splitter. 
It's:
1xUSB header cable
1xCPU fan header cable for receiving CPU
1xSATA power connection
1xARGB connector

So no fan control indeed. Sorry for the misunderstanding.

jonasmalacofilho, 2020-11-12:
Don't worry about it.

Oh, by the way, I'm pretty sure the CPU fan connector is only used for the pump tachometer (monitoring by motherboard), not control.

At least that was the case with the previous generation, and AFAIK that hasn't changed.  You can confirm it by checking which pins are populated: probably only ground (pin 1) and sense (pin 3).

closed 2020-11-12 by jonasmalacofilho