liquidctl/liquidctl #218: Corsair H150 Elite Cappalex

billygilbrech, 2020-11-12:
Can you add support for this AIO?

billygilbrech, 2020-11-12:
If you need anything also please add me on Discord billyg#2475

billygilbrech, 2020-11-12:
This devices connected to the Corsair Commander Core.

MarshallAsch, 2020-11-13:
Are you able to give the output from `lsusb -v` for that device?

And would you be able to capture some USB traffic between iCue and the device?

jonasmalacofilho, 2020-11-13:
Hopefully the Commander CORE, like the Lighting Node CORE, is also based on the Commander Pro.

billygilbrech, 2020-11-13:
I have no idea how to capture traffic.

billygilbrech, 2020-11-13:
Also on Windows, what's a softtware to grab USB Ids

jonasmalacofilho, 2020-11-14:
@billygilbrech, please try start with [this short guide](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/developer/capturing-usb-traffic.md#capturing-usb-traffic-on-a-native-windows-host).

We can probably infer which is the correct device by ourselves, and the traffic capture should also give us the USB IDs and descriptors.

billygilbrech, 2020-11-14:
[capture.zip](https://github.com/jonasmalacofilho/liquidctl/files/5539989/capture.zip)

billygilbrech, 2020-11-14:
I changed all settings from the AIO in iCue. Here are the provided captures I got.

billygilbrech, 2020-11-14:
1b1c:0c1c

MarshallAsch, 2020-11-14:
This seems to use the `usb.capdata` field for almost all of the communications. and there are an awful lot of really long messages that are almost entirely zeros. You didn't make any notes about what you changed when do you? @jonasmalacofilho I'm not quite sure how to figure out what the captured data actually means without a description of what happened when.

@billygilbrech can you please also give the output of `lsusb -d 1b1c:0c1c -v` (that gives us some information such as the request and responce lengths to help filter out some of the other usb traffic that is on that usb bus)

In that capture there are what appears to be traffic with 13 different devices. on that bus.

jonasmalacofilho, 2020-11-14:
@MarshallAsch 

The best way I found to filter these captures is by searching for the right `GET DESCRIPTOR Response DEVICE`: in this case this leads to filter with the expression `usb.device_address == 9`.
_(Note: based on @billygilbrech comment that we're interested in VID:PID `1b1c:0c1c`)_

Most stuff `lsusb` would report, apart from the complete HID _report_ descriptors, is in frame no. 34.  Unfortunately the HID report descriptors are nowhere to be found, but that's because the they are probably only fetched by the Windows kernel when it first sees the device.

billygilbrech, 2020-11-14:
https://hastebin.com/viriyunoxe.properties

billygilbrech, 2020-11-14:
It does say arctis 7 for some reason, anyways its the corsair aio, just usb groups are borked

jonasmalacofilho, 2020-11-14:
@billygilbrech 

To clarify: from the point of view of the OS, there is no AIO... it's just the Commander CORE controller, right?

Or are you directly connecting the AIO to the system, through USB or another bus?

jonasmalacofilho, 2020-11-14:
> It does say arctis 7 for some reason, anyways its the corsair aio, just usb groups are borked

Can you `sudo lsusb -d 1b1c:0c1c -v` so the string descriptors are fetched too?

00firestar00, 2020-12-31:
Here is the descriptor for this device:

```
Bus 001 Device 007: ID 1b1c:0c1c Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c1c 
  bcdDevice            1.00
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR iCUE Commander CORE
  iSerial                 3 010b00e08405c6bae4cf67e52091005f
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      34
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0300  1x 768 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      21
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0300  1x 768 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      0
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

MarshallAsch, 2021-01-01:
@00firestar00 would you be able to capture some of the traffic between the device and the icue using wireshark with some comments about what buttons in icue were clicked I dont have one of those devices available to test things with.

00firestar00, 2021-01-01:
I'll attach 2 files of me clicking around in it. Mostly changing the colors.
I ran wireshark from within windows, though not sure if that was the best option.

The problem is there seems to be a lot of garbage traffic, so I'm not actually sure what traffic was an action I took, or just corsair spam. I can try again, but i'm not really familiar with wireshark or how to potentially filter out garbage requests. (I did filter ONLY corsair, but to me the data looks like the craziest game of ping pong)

[corsair.zip](https://github.com/liquidctl/liquidctl/files/5758622/corsair.zip)

Just some extra info:
I just installed the icue H150i today, came across this post looking for something to control it from linux.
Host machine is Fedora 32, kernel 5.8, but I run Windows 10 in a KVM so I passed through the usb device.

billygilbrech, 2021-01-20:
I'm still here, just went inactive

MarshallAsch, 2021-01-21:
I was talking with @ZeldaZach last night going over how to reverse eng the protocol and he got an idea how to figure out the protocol. He was going to work on a basic protocol doc, but it might have to be implemented by someone else. 

For now Im going to assign this issue to you.

billygilbrech, 2021-01-21:
Excellent. Thank you! I can help if needed. I do have access to both Linux and Windows on my system. :)

chasingmaxwell, 2021-01-21:
I also have the Commander Core and H150i Elite Capellix with access to both linux and windows. If there's any way I can help with this effort I'd be glad to.

ZeldaZach, 2021-01-21:
Yeah feel free to assign this my way. I'll do some digging over the next week or so and see what I can uncover. @chasingmaxwell if you'd like to team up and help with the cause, feel free to reach out directly on Discord Zach#1064

jostack, 2021-01-21:
Hello everyone, I also have this Aio in case you need help.

`root@brain:/home/neuron/aio/liquidctl# lsusb -d 1b1c:0c1c -v

Bus 003 Device 006: ID 1b1c:0c1c Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c1c 
  bcdDevice            1.00
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR iCUE Commander CORE
  iSerial                 3 d00630307a8cc6ba2a47d6e57091005f
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      34
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0300  1x 768 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      21
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0300  1x 768 bytes
        bInterval               1
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      0
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
root@brain:/home/neuron/aio/liquidctl# `

I also have a comander pro on the computer which does recognize me as installed

`root@brain:/home/neuron/aio/liquidctl# liquidctl list
Device ID 0: Corsair Commander Pro (experimental)
root@brain:/home/neuron/aio/liquidctl# `

Congratulations on the project, I just installed it and it is fantastic !!!!

mschabhuettl, 2021-03-08:
@jonasmalacofilho I've also got the Commander Core on my system (Arch Linux), in case you need any help.

jonasmalacofilho, 2021-03-08:
Thanks!  By the way, you're (all) welcome to join our developer's [Discord server](https://discord.gg/f5HKCAadsr), where you'll find a `#wg-commander-core` channel.

Boothin, 2021-04-19:
The capellix requires you send 2 packets to put it into software mode before it still start accepting the color packets:
`08 01 03 00 02`
`08 0d 00 22`
After these 2 packets are sent, you can start sending the color packets that start with `08 06 a1 00 00 00 00 12`

To set a hardware mode, you need to send 3 packets first: 
`08 01 03 00 02`
`08 0d 01 64 6d`
`08 09 01`
then send a packet with the hardware mode you want to set.

aleksarias, 2021-07-24:
Is the H150i ELITE CAPELLIX not supported by this software?

jonasmalacofilho, 2021-07-24:
@aleksarias

As of 1.7.0, it is _partially_ supported (emphasis on "partially"), as stated in the "Supported devices" table and shown by the two PRs that have already been merged (linked just before your comment).

aleksarias, 2021-07-24:
[image]
I can't seem to get it to detect it. I tried the bind command but that was unsuccesful either.

jonasmalacofilho, 2021-07-24:
@aleksarias, can you show me the output of the commands bellow?

```
$ liquidctl list -g
$ liquidctl initialize -g
$ lsusb -tvv
```

aleksarias, 2021-07-24:
[image]
[image]
[image]

jonasmalacofilho, 2021-07-24:
@aleksarias , you appear to have two different versions of liqudictl installed.

And when you try to run it with `sudo`, you end up getting the older (1.6.1) one, probably due to how/where they are installed.

aleksarias, 2021-07-24:
@jonasmalacofilho that was it! thanks!

totaskin, 2021-12-01:
I have just installed H150i ELITE CAPELLIX and using dual boot (ubuntu/windows) and if you need help I could try contribute?

moutasem1989, 2022-02-11:
I am having trouble loading the driver for  _Crosair icue Commander Core  XT_
I suppose it has the same drivers as the _Crosair icue Commander Core_ ...
Using `lsusb` I can see the the device is connected: `Bus 002 Device 003: ID 1b1c:0c2a Corsair`
but using `liquidctl` initialize all I get: `Error: no devices matches available drivers and selection criteria`.
I only need the fan control function to work and nothing else. can anyone help?

ParkerMc, 2022-02-11:
At this point, we have no clue if the Commander Core XT uses the same protocol as the Commander Core. I've heard of it but I've never see any captures for it.
I would recommend opening a separate issue for it as there is not one for it yet. If you can use get captures using wireshark([guide here](https://github.com/liquidctl/liquidctl/blob/main/docs/developer/capturing-usb-traffic.md)). Then I can take a look to see if it's the same protocol.

jonasmalacofilho, 2022-02-11:
@moutasem1989,

> I suppose it has the same drivers as the _Crosair icue Commander Core_ ...

That's unknown at this time, even though I'm told there appear to be some similarities.

Take a look at PR #410, and maybe help with the unanswered questions and/or with some traffic captures.

debian-user-france1, 2022-03-08:
It's currently possible to get the speed of the fans and the liquid temp, but it's not possible to set the speed and change the LED behaviour, am I correct?

https://gitlab.com/codifryed/coolero/activity

jonasmalacofilho, 2022-03-08:
@debian-user-france1,

Not in liquidctl 1.8.1; but the code in our main branch [should already support it](https://github.com/liquidctl/liquidctl/blob/main/docs/corsair-commander-core-guide.md#programming-the-pump-and-fan-speeds) (this was added by PR #405).

---

P.S. Nothing against Coolero, quite the opposite!  But we don't know that code base... next time, try posting the output of the equivalent liquidctl command (e.g. `liquidctl set fan1 speed 75 --debug`), instead of a screenshot from Coolero.

jonasmalacofilho, 2025-11-15:
We'll no longer taking new device requests. Instead, please see [how](https://github.com/liquidctl/liquidctl/blob/main/CONTRIBUTING.md) you can help add support for that device and contribute your changes back to the project.

closed 2025-11-15 by jonasmalacofilho