liquidctl/liquidctl #219: Gigabyte Board not found

Erik262, 2020-11-13:
Today I updated my hackintosh to macOS 11 and realize that liqudctl only recognizes my NZXT Kraken X pump but not my Gigabyte board for changing the colors. It had worked on Catalina flawlessly.

Erik262, 2020-11-13:
This is what I got after typing "liquidctl list --debug"

[DEBUG] liquidctl.cli: running liquidctl v1.4.2 (homebrew; liquidctl 1.4.2)
[DEBUG] liquidctl.driver.usb: searching HidapiBus (api=hid, drivers=[CorsairHidPsu, HydroPlatinum, Kraken2, KrakenX3, KrakenZ3, NzxtEPsu, RgbFusion2, SmartDevice, SmartDevice2, _CommonSmartDeviceDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: instanced driver for NZXT Kraken X (X42, X52, X62 or X72)
[DEBUG] liquidctl.driver.usb: probing drivers for device 1395:0025
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0267
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0267
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0267
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0267
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0267
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0267
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0267
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: probing drivers for device 004c:0269
[DEBUG] liquidctl.driver.usb: searching PyUsbBus (drivers=[Hydro690Lc, Legacy690Lc, Modern690Lc, _CommonAsetekDriver])
[DEBUG] liquidctl.driver.usb: probing drivers for device 045e:076f
[DEBUG] liquidctl.driver.usb: probing drivers for device 1395:0025
[DEBUG] liquidctl.driver.usb: probing drivers for device 1e71:170e
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
[DEBUG] liquidctl.driver.usb: probing drivers for device 05e3:0608
Device ID 0: NZXT Kraken X (X42, X52, X62 or X72)
├── Vendor ID: 0x1e71
├── Product ID: 0x170e
├── Release number: 0x0200
├── Serial number: 6D8842AD495
├── Bus: hid
├── Address: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC@14/XHC@14000000/HS11@14500000/USB2.0 Hub@14500000/AppleUSB20Hub@14500000/AppleUSB20HubPort@14540000/NZXT USB Device@14540000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice
└── Driver: Kraken2 using module hid
[DEBUG] liquidctl.cli: hierarchy: UsbHidDriver, BaseUsbDriver, BaseDriver; HidapiDevice

jonasmalacofilho, 2020-11-14:
Yeah, the board's RGB controller is nowhere to be seen; there is a problem at the OS+USB level.

closed 2020-12-10 by jonasmalacofilho