liquidctl/liquidctl #222: Add Support for the Asus ROG STRIX RTX2080TI O11G

MarshallAsch, 2020-11-14:
This device does not use a `usb` interface, but it uses a `i2c/smbus` interface. 
As per discussions with @jonasmalacofilho this is a good starting point for adding in `i2c/smbus` devices.
The protocol for this device is described in the [openRGB project](https://gitlab.com/CalcProgrammer1/OpenRGB/-/wikis/ASUS-Aura-GPU) with a c++ implementation [here](https://gitlab.com/CalcProgrammer1/OpenRGB/-/blob/master/Controllers/AuraGPUController).

txf-, 2020-11-16:
I'll just throw out my 2 cents here, and ask if it wouldn't make more sense just to support the openRGB SDK? If you're going to be adding smbus support just to modify RGB values on graphics cards, isn't that expanding the scope of liquidctl, replicating functionality present in an existing well maintained project.

Or is the goal to have the same functionality implemented purely in python?

MarshallAsch, 2020-11-16:
@txf- I hadn't considered using the openRGB SDK to do some of the work for this project. (Although we are aware of that project). My main reasons for originally selecting this project over that project, from a user perspective, was wanting a cli to do the things rather then a GUI (as far as I know openRGB is just an SDK and a GUI) and because openRGB does not run on any of my systems for some reason, and after a quick attempt with running the python SDK I got the same unsuccessful result. Those are my reasons, Im not sure about @jonasmalacofilho's reasons though.

txf-, 2020-11-16:
Yeah, makes sense. I hope I didn't come across as being too pushy.

MarshallAsch, 2020-11-16:
All good, asking why we are "reinventing the wheel" is a perfectly valid question.

txf-, 2020-11-16:
I'm liable to need both projects installed, one to control cooling and another to do the lighting of various systems. I don't want to end up in a position where I need x number of different programs running to handle all that stuff, something ostensibly both programs exist to remedy.

Maybe I'm only worrying about a hypothetical.

MarshallAsch, 2020-11-16:
> I don't want to end up in a position where I need x number of different programs running to handle all that stuff, something ostensibly both programs exist to remedy.

That makes a lot of sense Its why I have been adding support for my devices to this project

What devices do you currently have / which program do you use to control them?

If you prefer to use the liquidctl program you can add an issue for a new device on here

txf-, 2020-11-16:
will do. Thanks.

jonasmalacofilho, 2020-11-17:
@txf- 

Your questions were valid...  And, honestly, I don't have a final answer to them.

For now, I'm interested in exploring how to do I²C/SMBus with Python, figuring out which devices¹ we can support across all platforms, and how to handle device detection in the safest possible way.

I think the experience will be valuable regardless of whether we decide to use the OpenRGB SDK, and even if we end up dropping this effort altogether.

¹ Not necessarily limited to RGB controllers.

PR #233 (2020-11-28): ASUS Strix 2080 TI OC
Closes: #222 

Added the ASUS Strix 2080 Ti OC driver that is currently working for me along with documentation on how to use the device.
- added the ASUS 2080ti rev A. and the EGVA 1080 FTW to the udev rule list

This is the same as #228 that accidentally got closed when the destination branch got deleted.

Steps to do before this can be merged:

- [x] Move driver over to the `nvidia.py` driver
- [x] Create automated tests.
closed 2020-11-28 by MarshallAsch