liquidctl/liquidctl #225: Add support for GSkill Trident Z Neo RAM

txf-, 2020-11-16:
I believe that this RAM uses the Asus AuraSync protocol over SMBus. I might tackle this myself once the liquidctl SMBus protocol takes shape.

MarshallAsch, 2020-11-24:
I believe that this also uses the ASUS smbus protocol: [here](https://gitlab.com/CalcProgrammer1/OpenRGB/-/wikis/ASUS-Aura-Overview).

Parts of a reference probe implementation from openRGB [here](https://gitlab.com/CalcProgrammer1/OpenRGB/-/blob/master/Controllers/AsusAuraSMBusController/AsusAuraSMBusControllerDetect.cpp).

@txf- would you be able to provide output for `sudo i2cdetect -l` then `udevadm info --name=/dev/i2c-<num> --attribute-walk` where `<num>` is the i2c device number for one of the ram sticks that you got from the first command. This will provide information to confirm that only the correct i2c bus gets probed when trying to detect the controller.

MarshallAsch, 2020-11-28:
@txf- The SMBus base classes have now been created and merged into master if you want to take a look at that if your still interested in adding a driver for this.

txf-, 2020-11-29:
finally got ubuntu installed on my new pc. For reference the results of i2cdetect:
```
i2c-3	i2c       	nvkm-0000:09:00.0-aux-0004      	I2C adapter
i2c-10	i2c       	nvkm-0000:09:00.0-bus-0008      	I2C adapter
i2c-1	i2c       	nvkm-0000:09:00.0-bus-0002      	I2C adapter
i2c-8	i2c       	nvkm-0000:09:00.0-bus-0007      	I2C adapter
i2c-15	i2c       	sor-0006-0f44                   	I2C adapter
i2c-6	i2c       	nvkm-0000:09:00.0-bus-0006      	I2C adapter
i2c-13	i2c       	nvkm-0000:09:00.0-aux-0009      	I2C adapter
i2c-4	i2c       	nvkm-0000:09:00.0-bus-0005      	I2C adapter
i2c-11	i2c       	nvkm-0000:09:00.0-aux-0008      	I2C adapter
i2c-2	i2c       	nvkm-0000:09:00.0-bus-0004      	I2C adapter
i2c-0	i2c       	nvkm-0000:09:00.0-bus-0001      	I2C adapter
i2c-9	i2c       	nvkm-0000:09:00.0-aux-0007      	I2C adapter
i2c-16	i2c       	sor-0006-0f82                   	I2C adapter
i2c-7	i2c       	nvkm-0000:09:00.0-aux-0006      	I2C adapter
i2c-14	i2c       	sor-0006-0f84                   	I2C adapter
i2c-5	i2c       	nvkm-0000:09:00.0-aux-0005      	I2C adapter
i2c-12	i2c       	nvkm-0000:09:00.0-bus-0009      	I2C adapter
```
this doesn't seem to be particularly clear.  This is on an x570 motherboard.

jonasmalacofilho, 2020-11-29:
I just learned that nvkm stands for NVIDIA Kernel Module, aka nouveau, and you might be able to gather more information about the `sor*` buses by inspecting `/sys/bus/i2c/devices/i2c-<number>/device`.

Can you also post the output of `ls -l /sys/bus/i2c/devices/`?

txf-, 2020-11-29:
output:
l```
s -l /sys/bus/i2c/devices/
total 0
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-0 -> ../../../devices/pci0000:00/0000:00:03.1/0000:09:00.0/i2c-0
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-1 -> ../../../devices/pci0000:00/0000:00:03.1/0000:09:00.0/i2c-1
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-10 -> ../../../devices/pci0000:00/0000:00:03.1/0000:09:00.0/i2c-10
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-11 -> ../../../devices/pci0000:00/0000:00:03.1/0000:09:00.0/i2c-11
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-12 -> ../../../devices/pci0000:00/0000:00:03.1/0000:09:00.0/i2c-12
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-13 -> ../../../devices/pci0000:00/0000:00:03.1/0000:09:00.0/i2c-13
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-14 -> ../../../devices/i2c-14
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-15 -> ../../../devices/i2c-15
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-16 -> ../../../devices/i2c-16
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-2 -> ../../../devices/pci0000:00/0000:00:03.1/0000:09:00.0/i2c-2
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-3 -> ../../../devices/pci0000:00/0000:00:03.1/0000:09:00.0/i2c-3
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-4 -> ../../../devices/pci0000:00/0000:00:03.1/0000:09:00.0/i2c-4
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-5 -> ../../../devices/pci0000:00/0000:00:03.1/0000:09:00.0/i2c-5
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-6 -> ../../../devices/pci0000:00/0000:00:03.1/0000:09:00.0/i2c-6
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-7 -> ../../../devices/pci0000:00/0000:00:03.1/0000:09:00.0/i2c-7
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-8 -> ../../../devices/pci0000:00/0000:00:03.1/0000:09:00.0/i2c-8
lrwxrwxrwx 1 root root 0 nov 29 15:54 i2c-9 -> ../../../devices/pci0000:00/0000:00:03.1/0000:09:00.0/i2c-9
```
doesn't seem like this helps much. Navigating the sor device structures, doesn't yield anything helpful. Just circular references

jonasmalacofilho, 2020-11-29:
> I can't really even find the bus where the SPD EEPROMs are located.

That's exactly what I was hoping to see, since their location is defined by the DDR4 spec.

What driver are the parents of `i2c-1[4-6]` using?  
(You should be able to find this with `ls -l /sys/bus/i2c/devices/<dev>/device/driver`).

---

How about `lspci`, can you see a SMBus controller there?

And what you get with `lspci -vvv -s <address of that smbus controller>`?

txf-, 2020-11-29:
> (You should be able to find this with `ls -l /sys/bus/i2c/devices/<dev>/device/driver`).

this is what I get:
```
ls -l /sys/bus/i2c/devices/i2c-15/
delete_device  i2c-dev/       name           new_device     power/         subsystem/     uevent 
```

lscpi:
```
lspci -vvv -s 00:14.0
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 61)
	Subsystem: Gigabyte Technology Co., Ltd FCH SMBus Controller
	Control: I/O+ Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap- 66MHz+ UDF- FastB2B- ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	IOMMU group: 12
	Kernel modules: i2c_piix4, sp5100_tco
```
it says the kernel module is i2c_piix4 as expected, but i2cdetect doesn't list it as such.

jonasmalacofilho, 2020-11-29:
> The SPD addresses are 0x52 and 0x53.
> The LED controllers must be at 0x70 and 0x71.

Possibly, but I would also check the other "ports" (i2c-1 and i2c-2).  Based on what I've been looking into for the Vengeance RAM, I expected the device address part (least significant 3 bits) to be same between SPD and LED.

txf-, 2020-11-30:
IIRC I probed i2c 1 and 2 and no devices enumerated on those busses. I'll check again tomorrow, though.

jonasmalacofilho, 2020-11-30:
If they are really different buses, absolutely; but if these ports are just some abstraction at the data layer, with a single bus at the physical level, it shouldn't matter.

In the DIMM module the SPD EEPROM and the LED controller are almost certainly different chips.  The only limitation is the DIMM connector, which has pins for a single 2-wire SMBus.

That said, the DIMM connector also only has pins for one 3-bit device address (SA0 to SA2), which is why it's weird to have the LEDs at 0x7[01], not 0x7[23].  But maybe they "flipped" some SAs into the LED controller for some reason.

txf-, 2020-11-30:
Running OpenRGB on windows detects one of the sticks at 0x74. I don't know how pertinent this is, I believe SMBus (as opposed to I2C) has the ability to remap addresses at runtime?

jonasmalacofilho, 2020-11-30:
SMBus 2.0 and later that have that ability in the address resolution protocol.

The addresses could change from boot to boot:

> Assigned addresses remain constant while device power is applied; address retention
> through device power loss is also allowed.

jonasmalacofilho, 2020-11-30:
I forgot to mention, but the upside of SMBus 2.0 and ARP is that it's also possible to enumerate the bus.

I'm not sure if we can use this from user-space (or even kernel space, the board firmware could be in control)... and we don't want to really "enumerate" the entire bus (this would be slow and could also confuse the board firmware).  But it might be possible to query the unique ID of a particular address, and that could tell us a lot about which device it is.

jonasmalacofilho, 2020-12-11:
@txf-

Some DDR4 stuff I was working on has finally reached the main branch: [`liquidctl.driver.ddr4`](https://github.com/jonasmalacofilho/liquidctl/blob/master/liquidctl/driver/ddr4.py).

The blind probing of addresses is significantly limited by first inspecting the detected DIMMs and their corresponding SPD EEPROM data.  Unfortunately this is fairly specific to the kernel support for a given SMBus, and there are significant differences between the Intel (`i801_smbus`) and modern AMD (`piix4_smbus`) drivers.  And I only implemented the Intel/i801 part...

We still need to assume a DTIC (upper 4-bits of the 7-bit I²C/SMBus address) for the RGB controller in the DIMMs.  On the other hand, since we will know which modules are present (and their lower 3-bit select addresses) we should be able to work around the inconsistent behavior you reported.

Also, feel free to join our developer's Discord server: <https://discord.gg/eJ5MswXSf9>.

jonasmalacofilho, 2021-08-06:
@txf- is this something you still want/plan to work on?  Do you need some help with the next steps?

jonasmalacofilho, 2022-04-05:
Closing the issue for now as interest seems to have died down, and this type of device also isn't the main focus of liquidctl.  That said, we'd probably still take a pull request... if someone is working on it, please feel free to get in touch.

closed 2022-04-05 by jonasmalacofilho